Koenkk/zigbee2mqtt #1402: Efficient way to do auto response?

he902, 2019-04-10:
My Yunmi Door Lock reports 'genPollCtrl' message every 20mins to the coordinator, and the coordinator needs to write attrId 16653 with attrData 0x0 when 'genPollCtrl' received.
Currently I'm doing it via HA Automation with MQTT to trigger the write action.
But I think it is not efficient and reliable by going through outside system (HA & MQTT). Any way inside zigbee2mqtt can do such task?

Here is the HA Automation I used:
```
- alias: 'Yunmi Lock Poll'
  trigger:
    platform: mqtt
    topic: zigbee2mqtt/yunmi_lock
  condition:
    condition: template
    value_template: "{{ 'poll' in trigger.payload_json }}"
  action:
    - service: mqtt.publish
      data:
        payload: "Active"
        topic: "zigbee2mqtt/yunmi_lock/set"
        retain: false
```
Here is the zigbee2mqtt debug log:
```
  zigbee2mqtt:debug 4/10/2019, 5:08:08 PM Received zigbee message of type 'attReport' with data '{"cid":"genPollCtrl","data":{"20480":0}}' of device 'FNB56-DOR23FB1.3' (0x00158d0002ac2222)
  zigbee2mqtt:info 4/10/2019, 5:08:08 PM MQTT publish: topic 'zigbee2mqtt/yunmi_lock', payload '{"date_poll":"4/10/2019, 5:08:08 PM","poll":0,"linkquality":68}'
  zigbee2mqtt:debug 4/10/2019, 5:08:08 PM Received MQTT message on 'zigbee2mqtt/yunmi_lock/set' with data 'Active'
  zigbee2mqtt:info 4/10/2019, 5:08:08 PM Zigbee publish to device '0x00158d0002ac2222', closuresDoorLock - write - [{"attrId":16653,"dataType":32,"attrData":0}] - {"manufSpec":0,"disDefaultRsp":0} - null
```

Koenkk, 2019-04-10:
You can add an `onZigbeeMessage(message, device, mappedDevice)` here https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/xiaomi.js to capture the message and respond to it. (example: https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/deviceReceive.js#L39). 

Note: when you add the `onZigbeeMessage` to `xiaomi.js` it will get called automatically (you don't have to register it somewhere).

he902, 2019-04-10:
Thanks Koenkk  a lot.
I need to investigate how to construct & send a new message inside onZigbeeMessage().

Koenkk, 2019-04-11:
@he902 please review https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/devicePublish.js#L184

he902, 2019-04-21:
@Koenkk I added `onZigbeeMessage()` followed your guidelines.
It is OK to send out a message **with** response from the device, but timeout (30000ms) to send out a message **without** response from the device.
Such as '`defaultRsp`', it has no further response from the device (just to response the message previously sent from the device):
```
    onZigbeeMessage(message, device, mappedDevice) {
        logger.info(`onZigbeeMessage(): 
                message.type = ${message.type},
                message.data.cid = ${message.data.cid},
                device.ieeeAddr = ${device.ieeeAddr}`);

        const callback = (error, rsp) => {
            logger.info(`onZigbeeMessage(): callback(): error = ${error}, rsp = ${rsp}`);
        };
    
        if (message.type === 'attReport' && message.data.cid === 'genPollCtrl') {
            this.zigbee.publish(
                device.ieeeAddr,
                'device',
                'genPollCtrl', //cid
                'defaultRsp', //cmd
                'foundation',
                { //zclData
                    cmdId: 0x0a,
                    statusCode: 0x0,
                },
                { //cfg
                     manufSpec: 0,
                     disDefaultRsp: 1,
                },
                null, //default endpoint
                callback
            );
        }
    }
```
  zigbee2mqtt:debug 4/21/2019, 12:57:21 PM Received zigbee message of type 'attReport' with data '{"cid":"genPollCtrl","data":{"20480":0}}' of device 'FNB56-DOR23FB1.3' (0x00158d0002acba74) of endpoint 1
  zigbee2mqtt:info 4/21/2019, 12:57:21 PM onZigbeeMessage(): 
                message.type = attReport,
                message.data.cid = genPollCtrl,
                device.ieeeAddr = 0x00158d0002acba74
  zigbee2mqtt:info 4/21/2019, 12:57:21 PM Zigbee publish to device '0x00158d0002acba74', genPollCtrl - defaultRsp - {"cmdId":10,"statusCode":0} - {"manufSpec":0,"disDefaultRsp":1} - null
  zigbee2mqtt:error 4/21/2019, 12:57:51 PM Zigbee publish to device '0x00158d0002acba74', genPollCtrl - defaultRsp - {"cmdId":10,"statusCode":0} - {"manufSpec":0,"disDefaultRsp":1} - null failed with error Error: Timed out after 30000 ms
  zigbee2mqtt:info 4/21/2019, 12:57:51 PM onZigbeeMessage(): callback(): error = Error: Timed out after 30000 ms, rsp = undefined
```

Koenkk, 2019-04-22:
`zigbee.publish` is indeed the right function. I think some changes need to be made to not provide a callback (just `null`). Can you check if you can provide `null` for the `callback` parameter?

he902, 2019-04-22:
Tried to provide `null` for the `callback` parameter, but still getting 30000ms timeout.
It always waits for 30000ms before execute the next command in the queue.

Koenkk, 2019-04-23:
The timeout is probably somewhere in zigbee-shepherd, perhaps you can do a search for that number to find out where it is coming from.

he902, 2019-04-23:
Searched "30000" in all the `.js` source code:
```
pi@raspberrypi:/opt/zigbee2mqtt $ find -name "*.js" -exec grep "30000" {} +
./node_modules/cc-znp/node_modules/chai/karma.sauce.js:  config.captureTimeout = 300000;
./node_modules/cc-znp/lib/ccznp.js:        }, 30000);
./node_modules/es5-ext/string/#/normalize/_data.js:    12133: [[30000], 256],
./node_modules/areq/index.js:    this._areqTimeout = areqTimeout || 30000;
./node_modules/zigbee-shepherd-converters/converters/toZigbee.js:                    'high': 0x04030000,
./node_modules/lodash/throttle.js: * var throttled = _.throttle(renewToken, 300000, { 'trailing': false });
./node_modules/lodash/lodash.js:     * var throttled = _.throttle(renewToken, 300000, { 'trailing': false });
./node_modules/zigbee-shepherd/lib/components/controller.js:    }, 30000);
./node_modules/chai/karma.sauce.js:  config.captureTimeout = 300000;
```
And looked into them one by one, still no clue.

he902, 2019-04-23:
I found the 30000ms timeout is handled  at `/opt/zigbee2mqtt/node_modules/q/q.js`:
```
Promise.prototype.timeout = function (ms, error) {
    var deferred = defer();
    var timeoutId = setTimeout(function () {
        if (!error || "string" === typeof error) {
            error = new Error(error || "Timed out after " + ms + " ms");
            error.code = "ETIMEDOUT";
        }
        deferred.reject(error);
    }, ms);
```
I tried to hardcode it with `ms = 2000`, and it can execute the consequent commands in the queue after waiting for 2000ms if no response from the device (such as  `defaultRsp`).

Is there any API to send command directly, instead of sending to the queue via `zigbee.publish`?
The commands sent by `zigbee.publish` are always waiting for response from device side with 30000ms timeout, which is not expected for `defaultRsp`.

Koenkk, 2019-04-24:
Can you make a PR of your current code? In this way I can help debugging.

he902, 2019-04-24:
PR made:
[https://github.com/Koenkk/zigbee2mqtt/pull/1467](https://github.com/Koenkk/zigbee2mqtt/pull/1467)

he902, 2019-04-29:
Done by PR #1480
This issue can be close.
Thanks for @Koenkk

closed 2019-04-29 by he902