Koenkk/zigbee2mqtt #1403: heineman water leak (HS1-WL-E) not supportet?

salopette, 2019-04-10:
my .db

https://pastebin.com/8xfJQ76j

Can you please add the Heiman HS1-WL-E?

Thank you

Koenkk, 2019-04-11:
Please follow http://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html, I expect that you can re-use: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L3043

salopette, 2019-04-11:
I do not quite understand what you mean.
I thought all the devices listed [here ](https://www.zigbee2mqtt.io/information/supported_devices.html)work out of the box?

Koenkk, 2019-04-11:
Yes, but that one is not listed (however one with a similar model is: HS1WL)

salopette, 2019-04-11:
which file or files do I have to edit?
How is it when I update? will they be overdressing again?
I am Totaller Noob what programming and writing concerns :-(

salopette, 2019-04-17:
@Koenkk 

can you fix that please?

Koenkk, 2019-04-19:
Do you see any `No converter available..` messages of this device in your zigbee2mqtt log? (try to trigger it by putting it in water).

salopette, 2019-04-19:
{"ieeeAddr":"0x000d6f0011d5a8bb","type":"EndDevice","model":"WaterSensor-EM","                                                                                               friendly_name":"0x000d6f0011d5a8bb","nwkAddr":14465,"manufId":4619,"manufName":"                                                                                               HEIMAN","powerSource":"Battery","modelId":"WaterSensor-EM"}]}'
Apr 19 21:15:00 raspberrypi npm[592]:   zigbee2mqtt:warn 2019-4-19 21:15:00 Devi                                                                                               ce with modelID 'WaterSensor-EM' is not supported.
Apr 19 21:15:00 raspberrypi npm[592]:   zigbee2mqtt:warn 2019-4-19 21:15:00 Plea                                                                                               se see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Apr 19 21:21:08 raspberrypi npm[592]:   zigbee2mqtt:warn 2019-4-19 21:21:08 Device with modelID 'WaterSensor-EM' is not supported.
Apr 19 21:21:08 raspberrypi npm[592]:   zigbee2mqtt:warn 2019-4-19 21:21:08 Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Koenkk, 2019-04-19:
Can you set `log_level: debug` (http://www.zigbee2mqtt.io/configuration/configuration.html) and provide the same log?

salopette, 2019-04-19:
2019-4-19 21:41:16 - info: New device 'WaterSensor-EM' with address 0x000d6f0011d5a8bb connected!
2019-4-19 21:41:17 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x000d6f0011d5a8bb","meta":{"modelID":"WaterSensor-EM"}}'
2019-4-19 21:41:17 - warn: Device with modelID 'WaterSensor-EM' is not supported.
2019-4-19 21:41:17 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

salopette, 2019-04-19:
.db

https://pastebin.com/myuAuNq3

Koenkk, 2019-04-19:
I don't see any debug logging, please verify that 

```yaml
advanced:
   log_level: debug
``` 

is in your configuration.yaml

salopette, 2019-04-20:
I'm sorry, did not take over the first time
https://pastebin.com/CEJ5RJdU

Koenkk, 2019-04-22:
During this log, did you also try to trigger the sensor? (by putting it in water)

salopette, 2019-04-22:
2019-4-22 19:51:23 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"HEIMAN","modelId":"WaterSensor-EM","powerSource":3}}' of device 'WaterSensor-EM' (0x000d6f0011d5a8bb) of endpoint 1
2019-4-22 19:51:23 - warn: Device with modelID 'WaterSensor-EM' is not supported.
2019-4-22 19:51:23 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
2019-4-22 19:51:24 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":18,"stackVersion":2,"hwVersion":16,"dateCode":"2018.3.10","swBuildId":"1.1.1"}}' of device 'WaterSensor-EM' (0x000d6f0011d5a8bb) of endpoint 1
2019-4-22 19:51:24 - warn: Device with modelID 'WaterSensor-EM' is not supported.
2019-4-22 19:51:24 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
2019-4-22 19:51:26 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f0011d5a8bb"'
2019-4-22 19:51:28 - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genPowerCfg","data":{"65533":1,"batteryVoltage":32,"batteryPercentageRemaining":200,"batteryRatedVoltage":32}}' of device 'WaterSensor-EM' (0x000d6f0011d5a8bb) of endpoint 1
2019-4-22 19:51:28 - warn: Device with modelID 'WaterSensor-EM' is not supported.
2019-4-22 19:51:28 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
2019-4-22 19:51:28 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f0011d5a8bb"'

Koenkk, 2019-04-23:
Is this really what is send when putting it in water?

salopette, 2019-04-23:
Yes, more is not coming

Koenkk, 2019-04-24:
Can you add the following to devices.js:

```js
    {
        zigbeeModel: ['WaterSensor-EM'],
        model: 'HS1WL-test',
        vendor: 'HEIMAN',
        description: 'Water leakage sensor',
        supports: 'water leak',
        fromZigbee: [fz.heiman_water_leak],
        toZigbee: [],
         configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
            ];

            execute(device, actions, callback, 1000);
        },
    },
```

salopette, 2019-04-25:
https://pastebin.com/VtxMa8rm

Koenkk, 2019-04-26:
When starting zigbee2mqtt, make sure that the sensor is awake by pressing some button on it. 

Now you see `Failed to configure 0x000d6f0011d5a8bb (0x000d6f0011d5a8bb)`. But you should see something like `Succesfully configured 0x000d6f0011d5a8bb......`

salopette, 2019-05-01:
I have now removed all devices from Stick, only Watermelder and Mi Button are online

Log:

https://pastebin.com/6eCur4Fb

db.

https://pastebin.com/vtf7bxXi

Koenkk, 2019-05-02:
In the log I see `Device with modelID 'WaterSensor-EM' is not supported.` which indicates that 
https://github.com/Koenkk/zigbee2mqtt/issues/1403#issuecomment-486388565 is in `devices.js` anymore.

salopette, 2019-05-02:
Was probably overwritten during the update.

Here again the LOG.

https://pastebin.com/hp5aNTES

Koenkk, 2019-05-02:
Is it working? I see water leak false and true messages.

salopette, 2019-05-02:
Yes it seems to work, if I put it in water, the message true and false!

Koenkk, 2019-05-02:
Great, support in latest dev branch.

closed 2019-05-02 by Koenkk

salopette, 2019-05-03:
Hi, the sensor works, but I have strange behavior in the log.

https://pastebin.com/Gf0QcN9R

Koenkk, 2019-05-03:
If this is only right after starting you can ignore it.