Koenkk/zigbee2mqtt #1404: Linking Tradfri bulb and dimmer in HA

eflemingtra, 2019-04-11:
Hi,

This will probably have an obvious answer. I've gotten both the Tradfri bulb and Dimmer discovered by Home Assistant using Zigbee2mqtt, no problem. But how do I control the bulb with the dimmer once it is in HA? Do i pair them normally as per IKEA's instructions (that didn't seem to work, but perhaps the order I do it in his important). 

If there's any documentation in HA I need to read, I'd be gratefully appreciative if I can be pointed in that direction.

andreasbrett, 2019-04-11:
Zigbee2MQTT won't know about those changes as both devices are talking directly to each other. There are 2 ways to get what you want:

1) enable reporting so that the IKEA bulb will report to the coordinator that its value(s) changed. Not sure though if reporting works with IKEA bulbs. 

2) don't bind the dimmer to the bulb itself but rather pair it with the coordinator. You would then need to create a rule in HA that would react upon changes in the dimmer and send the corresponding changes to the bulb. I'm an openHAB user so I don't know HA very well. Not sure how you would do this in HA but there should be some logic to react upon state changes in one device and apply some settings on another device.

andreasbrett, 2019-04-11:
Check those 2 FAQ entries for binding and reporting. Binding is what you manually did (=pairing 2 devices directly without coordinator). Reporting is the feature where devices report state changes to the coordinator.

https://www.zigbee2mqtt.io/information/binding.html
https://www.zigbee2mqtt.io/information/report.html

eflemingtra, 2019-04-11:
Thanks for that, makes perfect sense. I'll look into it and see what the best (easiest) solution will be for me.

eflemingtra, 2019-04-11:
Just an update if anyone else is in my position, yes, reporting does work with the TRADFRI bulb. Follow the guide from the link https://www.zigbee2mqtt.io/information/report.html, restart Zigbee2MQTT and it works to report the state of the bulb to Home Assistant whenever you use the physical dimmer.

closed 2019-06-17 by stale[bot]