Koenkk/zigbee2mqtt #1406: Osram Smart+ Motion Sensor still does not work?

salopette, 2019-04-11:
I got an Osram motion sensor, but it still does not work?
He is listed as supported in the list of supported devices?
I have the last DEV, but the sensor is not recognized

https://github.com/Koenkk/zigbee2mqtt/issues/507#issuecomment-481718610

Koenkk, 2019-04-12:
can you post your zigbee2mqtt startup log?

salopette, 2019-04-12:
https://pastebin.com/gzMpMbMu

.db

https://pastebin.com/0TTSBA0w

Koenkk, 2019-04-12:
The device was not paired correctly it seems (undefined model). Please remove it and re-pair.

salopette, 2019-04-12:
no matter what I do, it connects again and again!

https://pastebin.com/ZTEVHtSU

salopette, 2019-04-14:
Yesterday I completely deleted the zigbee2mqtt folder and set everything up again. And the motion detector was immediately recognized correctly

closed 2019-04-14 by salopette