Koenkk/zigbee2mqtt #1407: Automations not triggering

petro49er, 2019-04-11:
I've got a weird problem with automations sometimes not triggering.  Right now I am simply testing with xiaomi motion sensors triggering a switch (Lutron Caseta) to come on.  Sometimes it triggers, sometimes it doesn't.  I can see the motion in the zigbee2mqtt log and occupancy shows as true, but no automation kicks off.  I'm not sure where the breakdown is occurring.  Any idea where I can view logs to see where the breakdown might be happening?  I am running the following:
HASSIO 0.91.2 
zigbee2mqtt 1.2.1
Mosquito broker 4.1
Nod-RED 2.0.2

Thanks for any suggestions.

22MarioZ, 2019-05-14:
I had similar issue. It start working when I enable the automation..  Check in HA the state of the automation.

petro49er, 2019-05-14:
Mario,
This is probably a dumb question but where do I check the status of an automation?  In my configuration.yaml file, the only thing I have that says anything about automation is:

automation: !include automations.yaml

Thank you for your help on this!

petro49er, 2019-05-14:
> I had similar issue. It start working when I enable the automation.. Check in HA the state of the automation.

Mario,
This is probably a dumb question but where do I check the status of an automation? In my configuration.yaml file, the only thing I have that says anything about automation is:

automation: !include automations.yaml

Thank you for your help on this!

22MarioZ, 2019-05-14:
Open in the HA client (web) left side panel and at the bottom you have Developer tools. Click on <> and you will get new window "States". In the table you can see all HA entities and their states. Check the State column for specific automation if is on or off. If you click the icon in-front of the entity name an window will open where you can switch the automation on or off.

petro49er, 2019-05-14:
> Open in the HA client (web) left side panel and at the bottom you have Developer tools. Click on <> and you will get new window "States". In the table you can see all HA entities and their states. [...]

Mario,

When I check that page (<>) I can't seem to find an automation.  It's a simple automation created in Node-RED that says if motion is detected, turn on light.  If I look on the developer page, I see entries for the motion sensor and Caseta light switch.  I can't seem to turn the occupancy sensor on as there is no option for on when I click the link.  Am I not looking at the correct area?

Thanks,

Brian

22MarioZ, 2019-05-14:
Sorry. I thought you are using native HA automations. I cannot help you with Node-RED.

petro49er, 2019-05-14:
> Sorry. I thought you are using native HA automations. I cannot help you with Node-RED.

No problem.  I appreciate your help!

closed 2019-07-20 by stale[bot]