Koenkk/zigbee2mqtt #1410: Compatibility with Yale YRD226 lock

poma, 2019-04-11:
I have YRD226 lock and want to change my August network module (utter garbage btw) to a ZigBee module. Looks like I'll need model [AYR202-ZB-HA](https://shopyalehome.com/Products/Accessories/AYR202-ZB-HA-USA.aspx). I don't see it in supported devices but I've seen some recent issues on some other Yale locks.

Will it be compatible with existing YRD426 lock converter? If not, will I be able to figure out how to send open/close command and get lock status without yale hub?

Koenkk, 2019-04-12:
It may be compatible, there is also another yale lock being added right now: https://github.com/Koenkk/zigbee-shepherd-converters/pull/399/files

poma, 2019-04-12:
Great! How do authors of YRD426 and YMF40 converters figure out which commands do they need to send to lock/unlock it? Will I be able to do the same if mine will be different and I don't have yale hub?

Koenkk, 2019-04-12:
@poma with a bit of luck, we can use the same command.

poma, 2019-04-12:
ok got it, thanks

closed 2019-04-12 by poma

Koenkk, 2019-04-12:
I think it's your lucky day, somebody just made this pr: https://github.com/Koenkk/zigbee-shepherd-converters/pull/402