Koenkk/zigbee2mqtt #1412: Not able to install on OrangePi

olili, 2019-04-12:
I'm not able to install ZigBee2MQTT. 
Following error is logged.  Any hints ?

15323 verbose stack Error: @serialport/bindings@2.0.7 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
15323 verbose stack Exit status 1
15323 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
15323 verbose stack     at EventEmitter.emit (events.js:189:13)
15323 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
15323 verbose stack     at ChildProcess.emit (events.js:189:13)
15323 verbose stack     at maybeClose (internal/child_process.js:970:16)
15323 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
15324 verbose pkgid @serialport/bindings@2.0.7
15325 verbose cwd /opt/zigbee2mqtt
15326 verbose Linux 4.19.20-sunxi
15327 verbose argv "/usr/bin/node" "/usr/bin/npm" "install"
15328 verbose node v10.15.3
15329 verbose npm  v6.4.1
15330 error code ELIFECYCLE
15331 error errno 1
15332 error @serialport/bindings@2.0.7 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
15332 error Exit status 1
15333 error Failed at the @serialport/bindings@2.0.7 install script.
15333 error This is probably not a problem with npm. There is likely additional logging output above.
15334 verbose exit [ 1, true ]

andreasbrett, 2019-04-12:
Could you check your NPM and NODEJS versions?

`node --version  # Should output v10.X`
`npm --version  # Should output 6.X`

olili, 2019-04-12:
some additional informations to my above request: 
node v10.15.5   npm 6.4.1   OS: DietPi  V6.22.3

olili, 2019-04-12:
some additional informations to my above request:
node v10.15.5 npm 6.4.1 OS: DietPi V6.22.3

Can you help?

Best regards, O.

aktentasche, 2019-04-14:
there is some information when you google for "error code ELIFECYCLE", maybe this will help you. one option is to run "npm cache clean".

olili, 2019-04-14:
danke.
Hab's jetzt selber hinbekommen.
1 Tag Googeln hat dann schließlich geholfen; ich musste perl-dev updaten.

Gruß, O:

henkiejan1, 2019-05-19:
> 
> danke. Hab's jetzt selber hinbekommen. 1 Tag Googeln hat dann schließlich geholfen; ich musste perl-dev updaten. Gruß, O: Am So., 14. Apr. 2019 um 09:48 Uhr schrieb aktentasche < notifications@github.com>:
> […](#)

Hi, i have the same problem. Can you explain what was your solution? I running Armbian on a Orange Pi One Plus H6 and can't get fixed...

henkiejan1, 2019-05-19:
I got the following error when i run npm install:
`gyp ERR! configure error
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:345:16)
gyp ERR! stack     at ChildProcess.emit (events.js:189:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System Linux 5.1.3-sunxi64
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /opt/zigbee2mqtt/node_modules/cc-znp/node_modules/@serialport/bindings
gyp ERR! node -v v10.15.3
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm WARN bl@0.7.0 requires a peer of stream-browserify@* but none is installed. You must install peer dependencies yourself.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm64"})

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@2.0.7 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the @serialport/bindings@2.0.7 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.`

olili, 2019-05-19:
in my case "apt-get install perl-dev" helped. 
Good luck

henkiejan1, 2019-06-23:
When i has run ´npm install node-gyp´ in the opt/zigbee2mqtt diretory zigbee2mqtt installs ok. With the standard warnings but that should not a problem.

rolandow, 2019-06-25:
On armbian I installed node-gyp with `sudo apt install node-gyp`. This will also install some dependencies (like python2, while I alread had 3.. hope this won't clash). Now npm install seems to run 'normally'. Just a bunch of warnings.

closed 2019-08-31 by stale[bot]