Koenkk/zigbee2mqtt #1413: Xiaomi MiJia Honeywell smoke detector (JTYJ-GD-01LM/BW) test button

SargonofAssyria, 2019-04-12:
zigbee2mqtt version: 1.3.0 (commit #da94252)
CC2531 firmware version: '20190223'

Pushing the test button give this message:
`4/12/2019, 9:18:48 AM - info: MQTT publish: topic 'zigbee2mqtt/Smoke', payload '{"smoke":false,"linkquality":13,"device.....
4/12/2019, 9:18:48 AM - info: MQTT publish: topic 'zigbee2mqtt/Smoke', payload '{"smoke":false,"linkquality":13,"device....
4`
The message is transfered via Domoticz plugin to Domoticz, but there is no action in Domoticz.
I suppose that is because is the '"smoke":false'
Is this correct? How can this device be tested so that a action in Domoticz is generated.

tb-killa, 2019-04-12:
https://github.com/Koenkk/zigbee2mqtt/issues/391

SargonofAssyria, 2019-04-14:
So I set debug on, and this is what I get. So that 'statusChange' should be handled by The plugin and then by Domoticz? Is there any logic in the message 'devChange'?
`4/14/2019, 9:53:51 AM - debug: Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":2}' of device 'lumi.sensor_smoke' (0x00158d0001d64498) of endpoint 1
4/14/2019, 9:53:51 AM - info: MQTT publish: topic 'zigbee2mqtt/Smoke', payload '{"smoke":false,"linkquality":57,"device":{"ieeeAddr":"0x00158d0001d64498","friendlyName":"Smoke","type":"EndDevice","nwkAddr":37960,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_smoke","hwVersion":"unknown","swBuildId":"unknown","dateCode":"unknown","status":"online"},"battery":100,"voltage":3065}'
4/14/2019, 9:53:51 AM - debug: Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"1":3065,"3":27,"4":5032,"5":37,"6":[0,2],"8":4100,"10":0,"100":0,"150":536870912}}}' of device 'lumi.sensor_smoke' (0x00158d0001d64498) of endpoint 1
4/14/2019, 9:53:51 AM - info: MQTT publish: topic 'zigbee2mqtt/Smoke', payload '{"smoke":false,"linkquality":57,"device":{"ieeeAddr":"0x00158d0001d64498","friendlyName":"Smoke","type":"EndDevice","nwkAddr":37960,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_smoke","hwVersion":"unknown","swBuildId":"unknown","dateCode":"unknown","status":"online"},"battery":100,"voltage":3065}'
4/14/2019, 9:53:51 AM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"65281":[null,3065,null,27,null,null,[null,2],null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,536870912]}}' of device 'lumi.sensor_smoke' (0x00158d0001d64498) of endpoint 1
4/14/2019, 9:53:58 AM - debug: Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":0}' of device 'lumi.sensor_smoke' (0x00158d0001d64498) of endpoint 1
`

closed 2019-06-20 by stale[bot]