Koenkk/zigbee2mqtt #1415: [VENV] Trying to install server on Synology NAS (ARM)

schumi2004, 2019-04-12:
Hi,

I tried to install zigbee2mqtt server on my Synology NAS (DS715) in a virtual python3 environment since the NAS itself is unable to run Docker.,
Followed the guide and started installation.

It returned a few errors:
(zigbee2mqtt) root@DS715:/volume1/@appstore/virtualenv/zigbee2mqtt# sudo npm install
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)

> @serialport/bindings@2.0.7 install /volume1/@appstore/virtualenv/zigbee2mqtt/node_modules/cc-znp/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
prebuild-install WARN install No prebuilt binaries found (target=10.15.1 runtime=node arch=arm libc= platform=linux)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
gyp: Call to 'node -e "require('nan')"' returned exit status 0 while in binding.gyp. while trying to load binding.gyp
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/volume1/@appstore/virtualenv/zigbee2mqtt/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:345:16)
gyp ERR! stack     at ChildProcess.emit (events.js:189:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System Linux 3.10.105
gyp ERR! command "/volume1/@appstore/virtualenv/zigbee2mqtt/bin/node" "/volume1/@appstore/virtualenv/zigbee2mqtt/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /volume1/@appstore/virtualenv/zigbee2mqtt/node_modules/cc-znp/node_modules/@serialport/bindings
gyp ERR! node -v v10.15.1
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.7 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.7: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@2.0.7 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @serialport/bindings@2.0.7 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-04-12T15_49_31_953Z-debug.log
https://pastebin.com/uk23eNDL

And the debug the install refers to 2019-04-12T15_49_31_953Z-debug.log
http://batman.gyptis.org/zerobin/?4d5382137d6fc5b7#HLPHWNzneKRysjlSLC07fp1xNPJ6KATYkpdG6CX43Cw=

Then although it gave errors tried to start it.
(zigbee2mqtt) root@DS715:/volume1/@appstore/virtualenv/zigbee2mqtt# npm start
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)

> zigbee2mqtt@1.3.0 start /volume1/@appstore/virtualenv/zigbee2mqtt
> node index.js

node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
node: /lib/libstdc++.so.6: no version information available (required by node)
  zigbee2mqtt:info 4/12/2019, 5:51:06 PM Logging to directory: '/volume1/@appstore/virtualenv/zigbee2mqtt/data/log/2019-04-12.17-51-06'
internal/modules/cjs/loader.js:583
    throw err;
    ^

Error: Cannot find module 'zigbee-shepherd'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:581:15)
    at Function.Module._load (internal/modules/cjs/loader.js:507:25)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:22:18)
    at Object.<anonymous> (/volume1/@appstore/virtualenv/zigbee2mqtt/lib/zigbee.js:1:81)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.3.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.3.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-04-12T15_51_06_602Z-debug.log
https://pastebin.com/4pK1TqH4

And the debug log it refers to 2019-04-12T15_51_06_602Z-debug.log
https://pastebin.com/pi2WYMP3

I really hope it can be done otherwise i need to rethink what i want.

Koenkk, 2019-04-13:
You need to install libstdc++ 6 (required by one of zigbee2mqtt dependencies) https://github.com/sgerrand/alpine-pkg-glibc/issues/80

schumi2004, 2019-06-12:
I didn't had time yet to dive deeper in to this issue but i also asked for assistance from the SynCommunity getting this fixed.

closed 2019-08-18 by stale[bot]