Koenkk/zigbee2mqtt #1416: HEIMAN HS2WD-E Smart Siren

opebayle, 2019-04-12:
# Bug Report
## What happened
I'm trying to add a new device, Heiman Smart siren.
Pairing with the device seems to be ok, I've got this message
```
4/12/2019, 6:12:15 PM - warn: Device with modelID 'WarningDevice' is not supported.
4/12/2019, 6:12:15 PM - warn: Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
```
So I've added this to "zigbee-shepherd-converters/devices.js"
```
    {
        zigbeeModel: ['WarningDevice'],
        model: 'HS2WD-E',
        vendor: 'HEIMAN',
        description: 'Siren Device',
        supports: 'Siren',
        fromZigbee: [],
        toZigbee: [],
    }
```
But nothing append, no report from the device.
I haven't found exemple of IAS Warning Devices. So I don't know how to configure the device to have a report or to control it ...

thanx for your help

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: version 1.0.1 (commit #43fc66c)
CC253X firmware version:  Coordinator firmware version: '20181024'

Koenkk, 2019-04-15:
Can you add the following?

**devices.js**
```js
    {
        zigbeeModel: ['WarningDevice'],
        model: 'HS2WD-E',
        vendor: 'HEIMAN',
        description: 'Siren Device',
        supports: 'Siren',
        fromZigbee: [],
        toZigbee: [tz.warning],
    }
```

**toZigbee.js**
```js
    warning: {
        key: ['warning'],
        convert: (key, value, message, type, postfix) => {
            if (type === 'set') {
                return {
                    cid: 'ssIasWd',
                    cmd: 'startWarning',
                    cmdType: 'functional',
                    zclData: {startwarninginfo: 1, warningduration: 10},
                    cfg: cfg.default,
                };
            }
        },
    },
````

You should be able to control the device by sending to `zigbee2mqtt/[DEVICE_FRIENDLY_NAME]/set` payload `{"warning": ""}`

opebayle, 2019-04-15:
With your code nothing append.
I've tryed with the squawk command instead, and it works :
```
return {
   cid: 'ssIasWd',
   cmd: 'squawk',
   cmdType: 'functional',
   zclData: {squawkinfo: 1},
   cfg: cfg.default,
};
```
But it's only a squawk (a few seconds) not a warning during "warningduration" :(

Any idea ?

Koenkk, 2019-04-17:
Can you try with

**toZigbee.js**
```js
    warning: {
        key: ['warning'],
        convert: (key, value, message, type, postfix) => {
            if (type === 'set') {
                return {
                    cid: 'ssIasWd',
                    cmd: 'startWarning',
                    cmdType: 'functional',
                    zclData: {startwarninginfo: 103, warningduration: 10},
                    cfg: cfg.default,
                };
            }
        },
    },
````
Documentation: https://www.nxp.com/docs/en/user-guide/JN-UG-3077.pdf (28.6.2)

mccoy88f, 2019-05-27:
I'm trying the same code for my heiman warning device, were i can find locally con my Hass.io the "toZigbee" file to modify it?
Thanks

Koenkk, 2019-05-28:
http://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices_on_hassio.html and add `converters/fromZigbee.js` relative from `devices.js`, start by copying https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js

mccoy88f, 2019-05-28:
I cannot make it works i recive this answer from the log:
  zigbee2mqtt:debug 5/28/2019, 6:31:40 PM Received MQTT message on 'zigbee2mqtt/0x000d6f000fadc498/set' with data '{"warning": ""}'
/zigbee2mqtt-1.4.0/lib/extension/devicePublish.js:158
            const converter = converters.find((c) => c.key.includes(key));
                                                       ^
TypeError: Cannot read property 'key' of undefined
    at converters.find (/zigbee2mqtt-1.4.0/lib/extension/devicePublish.js:158:56)
    at Array.find (<anonymous>)
    at keys.forEach (/zigbee2mqtt-1.4.0/lib/extension/devicePublish.js:158:42)
    at Array.forEach (<anonymous>)
    at DevicePublish.onMQTTMessage (/zigbee2mqtt-1.4.0/lib/extension/devicePublish.js:157:14)
    at results.extensions.filter.map (/zigbee2mqtt-1.4.0/lib/controller.js:152:27)
    at Array.map (<anonymous>)
    at Controller.onMQTTMessage (/zigbee2mqtt-1.4.0/lib/controller.js:152:14)
    at MQTT.onMessage (/zigbee2mqtt-1.4.0/lib/mqtt.js:81:18)
    at MqttClient.emit (events.js:182:13)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.4.0 start: `node index.js`
npm ERR! Exit status 1
npm 
ERR! 
npm ERR! Failed at the zigbee2mqtt@1.4.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-05-28T16_31_40_672Z-debug.log

I ADD A PICTURE TO SEE WHERE I PUT toZigbee.js and fromZigBee.js
https://ibb.co/x8R57Yy

Koenkk, 2019-05-28:
Can you add to https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L6 `console.log(fz, tz)` and post the log here (on pastebin.com)

mccoy88f, 2019-05-29:
I add the line you tell me but i cannot find where is the log...

Koenkk, 2019-05-29:
Can you add `console.log('I AM HERE', fz, tz)` and search for `I AM HERE` in your log?

mccoy88f, 2019-05-29:
In log.txt file or in the log screen of zigbee2mqtt there isn't

Koenkk, 2019-05-29:
I'm not sure what's happening, perhaps it's is easier to setup a bare metal installation (http://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html) and follow http://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

mccoy88f, 2019-06-05:
Someone solved?

mccoy88f, 2019-06-07:
Is possible to use the siren as a switch (on siren on and off siren off)
Coping a generic ON/OFF converter and adding in the set request the
return {
                    cid: 'ssIasWd',
                    cmd: 'startWarning',
                    cmdType: 'functional',
                    zclData: {startwarninginfo: 1, warningduration: 10},
                    cfg: cfg.default,
}

we also need a return for the off
What do you thing @Koenkk

Koenkk, 2019-06-08:
Can you turn it off with:

```js
return {
cid: 'ssIasWd',
cmd: 'startWarning',
cmdType: 'functional',
zclData: {startwarninginfo: 0, warningduration: 0},
cfg: cfg.default,
}
```

JLFN, 2019-06-21:
I have also buy this device iam want to get help out also.
It found this info.
[https://github.com/gabriele-v/hubitat/blob/master/Drivers/Heiman_HS2WD-E_Driver.groovy](https://github.com/gabriele-v/hubitat/blob/master/Drivers/Heiman_HS2WD-E_Driver.groovy)

JLFN, 2019-06-21:
@Koenkk 
Do you mean like this n toZigbee.js
```js
    },
    warning: {
        key: ['warning'],
        convert: (key, value, message, type, postfix) => {
            if (type === 'set') {
                return {
                    cid: 'ssIasWd',
                    cmd: 'startWarning',
                    cmdType: 'functional',
                    zclData: {startwarninginfo: 103, warningduration: 10},
                    cfg: cfg.default,
                };
            }
        },
    },
    warning_on_off: {
        key: ['state'],
        convert: (key, value, message, type, postfix) => {
            const cid = 'genOnOff';
            const attrId = 'onOff';

            if (type === 'set') {
                if (typeof value !== 'string') {
                    return;
                }

                return [{
                    cid: cid,
                    cmd: startWarning',
                    cmdType: 'functional',
                    zclData: {startwarninginfo: 1, warningduration: 10},
                    cfg: cfg.default,
                    },
                }];
            } else if (type === 'get') {
                return [{
                    cid: cid,
                    cmd: startWarning',
                    cmdType: 'functional',
                    zclData: {startwarninginfo: 0, warningduration: 0},
                    cfg: cfg.default,
                }];
            }
        },
    },
```

Koenkk, 2019-06-23:
Looks good to me and this also appear to be whats in https://github.com/gabriele-v/hubitat/blob/master/Drivers/Heiman_HS2WD-E_Driver.groovy#L132, what's the error? Can you post debug logs when the command  is send?

JLFN, 2019-06-24:
@Koenkk iam away on abroad on vacation right, what happened i can't start zigbee2mqtt it crashes direct thats why i think there might be something wrong code i tried. I Haven't paired the device yet maybe its because max devices maybe have been reach. I will try to sniff when iam back. Do you have the possibility to use this code and see if you can start you zigbee2mqtt. thank you in advance. @mccoy88f can you try the code?

JLFN, 2019-07-03:
@Koenkk 
I got it working better now with lastest dev.
i used on_off and edited it.
Whats the command for mqtt to try it?

##device.js
```.js
//WarningDevice
    {
        zigbeeModel: ['WarningDevice'],
        model: 'HS2WD-E',
        vendor: 'HEIMAN',
        description: 'Siren Device',
        supports: 'Siren',
        fromZigbee: [],
        toZigbee: [tz.warning, tz.warning_on_off],
},
```
##toZigbee.js
```.js
//
    warning: {
        key: ['warning'],
        convert: (key, value, message, type, postfix) => {
            if (type === 'set') {
                return {
                    cid: 'ssIasWd',
                    cmd: 'startWarning',
                    cmdType: 'functional',
                    zclData: {startwarninginfo: 103, warningduration: 10},
                    cfg: cfg.default,
                };
            }
        },
    },
//
    warning_on_off: {
        key: ['warning_on_off'],
        convert: (key, value, message, type, postfix, options) => {
            const cid = 'genOnOff';
            const attrId = 'onOff';

            if (type === 'set') {
                if (typeof value !== 'string') {
                    return;
                }

                return [{
                    cid: ssIasWd,
                    cmd: 'startWarning',
                    cmdType: 'functional',
                    zclData: {startwarninginfo: 1, warningduration: 10},
                    cfg: options.disFeedbackRsp ? cfg.defaultdisFeedbackRsp : cfg.default,
                    newState: {state: value.toUpperCase()},
                }];
            } else if (type === 'get') {
                return [{
                    cid: ssIasWd,
                    cmd: 'startWarning',
                    cmdType: 'functional',
                    zclData: {startwarninginfo: 0, warningduration: 0},
                    cfg: cfg.default,
                }];
            }
        },
    },
```
##database.db
```
{"id":2,"type":"Router","ieeeAddr":"0x000d6f0012061fe8","nwkAddr":31591,"manufId":4619,"manufName":"Heiman","modelId":"WarningDevice","epList":[1],"zclVersion":1,"appVersion":18,"stackVersion":2,"hwVersion":16,"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":1027,"inClusterList":[0,1,3,4,9,1280,1282],"outClusterList":[3,25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genAlarms":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{}},"ssIasWd":{"dir":{"value":1},"attrs":{}}}}},"dateCode":"2018.03.21","_id":"LCumeQAjfc5bU4hQ"}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b001bb859d0","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,11],"status":"online","joinTime":1562147960,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}}}}},"_id":"W7ixccc4mPTlWfcI"}
{"$$indexCreated":{"fieldName":"id","unique":true,"sparse":false}}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b001bb859d0","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1562147988,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}}},"_id":"W7ixccc4mPTlWfcI"}
{"id":2,"type":"Router","ieeeAddr":"0x000d6f0012061fe8","nwkAddr":31591,"manufId":4619,"manufName":"Heiman","modelId":"WarningDevice","epList":[1],"zclVersion":1,"appVersion":18,"stackVersion":2,"hwVersion":16,"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":1027,"inClusterList":[0,1,3,4,9,1280,1282],"outClusterList":[3,25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genAlarms":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{}},"ssIasWd":{"dir":{"value":1},"attrs":{}}}}},"dateCode":"2018.03.21","_id":"LCumeQAjfc5bU4hQ"}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b001bb859d0","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,11],"status":"online","joinTime":1562147988,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}}, [... cut ...] Id":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}}}}},"_id":"W7ixccc4mPTlWfcI"}
```

Koenkk, 2019-07-04:
Together with @JLFN we added support for this device. This device is now supported in the dev branch, documentation on how to trigger the alarm can be found here: https://github.com/Koenkk/zigbee2mqtt.io/blob/dev/devices/HS2WD-E.md#triggering-the-alarm

closed 2019-07-04 by Koenkk

JLFN, 2019-07-05:
> Together with @JLFN we added support for this device. This device is now supported in the dev branch, documentation on how to trigger the alarm can be found here: https://github.com/Koenkk/zigbee2mqtt.io/blob/dev/devices/HS2WD-E.md#triggering-the-alarm
Hey @Koenkk  t
Can you please make this look like state for on and off . Thank you in advance

Jul 05 20:14:32 bluebox npm[10501]:   zigbee2mqtt:info 7/5/2019, 8:14:32 PM Zigbee publish to device '0x000d6f001f', ssIasWd - startWarning - {"startwarninginfo":53,"warningduration":1800} - {"manufSpec":0,"disDefaultRsp":0} - null
Jul 05 20:14:33 bluebox npm[10501]:   zigbee2mqtt:info 7/5/2019, 8:14:33 PM Zigbee publish to device '0x000d6f001f', ssIasWd - startWarning - {"startwarninginfo":1,"warningduration":0} - {"manufSpec":0,"disDefaultRsp":0} - null

Koenkk, 2019-07-05:
@JLFN please try to setup reporting for the startwarninginfo attribute, we cannot do this in the converter directly.

JLFN, 2019-07-05:
@Koenkk can you please write the line then i can try thank you in advance

Koenkk, 2019-07-06:
I've checked the cluster, but it does not have an attribute which indicates that the alarm is on