Koenkk/zigbee2mqtt #1417: Ikea Tradfri repeater no quality link value reported

tiagofreire-pt, 2019-04-12:
# Bug Report
## What happened
No link quality reported on devices connected to this repeater

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:  1.3.0
CC253X firmware version: March 29, 2019

abmantis, 2019-05-26:
Related question for devs (so I can try and fix this): how is the linkquality value obtained? Looking at the code they seem to come from a zigbee message, but I have not yet found any zigbee message with that by looking at debug logs.

dnaphreak, 2019-07-20:
I am seeing the same behavior. Coordinator firmware: 20190425 and zigbee2mqtt: 1.4.0 I have three repeaters that show as joined devices but never report linkquality values.

Koenkk, 2019-07-21:
The reason for this is that the TRADFRI repeater doesn't send any messages.

Perhaps we could configure this so that it does, please try with the following configuration:

**devices.js**
```js
{
        zigbeeModel: ['TRADFRI signal repeater'],
        model: 'E1746',
        description: 'TRADFRI signal repeater',
        supports: 'linkquality',
        vendor: 'IKEA',
        fromZigbee: [fz.ignore_basic_change, fz.ignore_diagnostic_change],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);

            const actions = [
                (cb) => device.report('genBasic', 'modelId', 0, 3, 0, cb),
            ];

            execute(device, actions, callback);
        },
    },
```

**fromZigbee.js**
```js
E1746_linkquality: {
        cid: 'genBasic',
        type: ['attReport', 'readRsp'],
        convert: (model, msg, publish, options) => {
            return {linkquality: msg.linkquality};
        },
    },
```

dnaphreak, 2019-07-21:
I may have not done this correctly. I am getting this warning:

`No converter available for 'E1746' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"modelId":"TRADFRI signal repeater"}}'`

Koenkk, 2019-07-22:
I forgot something, please try with:

**devices.js**
```js
{
        zigbeeModel: ['TRADFRI signal repeater'],
        model: 'E1746',
        description: 'TRADFRI signal repeater',
        supports: 'linkquality',
        vendor: 'IKEA',
        fromZigbee: [fz.ignore_basic_change, fz.ignore_diagnostic_change, fz.E1746_linkquality],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);

            const actions = [
                (cb) => device.report('genBasic', 'modelId', 0, 3, 0, cb),
            ];

            execute(device, actions, callback);
        },
    },
```

**fromZigbee.js**
```js
E1746_linkquality: {
        cid: 'genBasic',
        type: ['attReport', 'readRsp'],
        convert: (model, msg, publish, options) => {
            return {linkquality: msg.linkquality};
        },
    },
```

3DIRK, 2019-07-22:
@Koenkk 

I tried the modification as well - my three IKEA repeaters are reporting their linkquality now. 
Unforturnately they report each 3 seconds, which increases the load on the network substantially.

Koenkk, 2019-07-22:
@3DIRK the code above was for testing purposes, I would recommend changing to

```js
(cb) => device.report('genBasic', 'modelId', 3600, 14400, 0, cb),
```

which would result in linkquality reported somewhere between every 1 till 4 hours.

dnaphreak, 2019-07-22:
Okay, I have made the changes and am now receiving linkquality reports from each repeater! Thank you! Do you need anything from my end?

Koenkk, 2019-07-24:
Nope, will be available in the dev branch soon, thanks!

closed 2019-07-24 by Koenkk

joseal, 2020-12-07:
Hi @Koenkk, 
I'm facing this issue on an home assistant that I've just installed, so I would like to check with you if this change is present in the current release?
thank you!

Koenkk, 2020-12-07:
@joseal it's already in for a long time (+ 1 year), if it doesn't work you can try to reconfigure: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure

joseal, 2020-12-07:
Hi @Koenkk, thanks for the quick reply!
I tried to reconfigure, but it was give me an exception, so I decided to rebuild the container and it looks much better now.
Also, the upgrade from raspberry 3B to raspberry 4 seems to cause a negative impacting on the signal quality between the coordenator and the ikea repeater... must be the noise caused by the PI4 (it's a bit better with another USB extension, but not the same as with PI3B).
thank you

soenkegissel, 2021-03-08:
@joseal I have the same problem. What do you mean with "rebuild the container"?

joseal, 2021-03-09:
> @joseal I have the same problem. What do you mean with "rebuild the container"?

@soenkegissel, in my case I had to rebuild the docker container (you can do it by simple removed the addon and added it again). Also, validate if your repeater is running the latest firmware (check for OTA updates) and do a reconfirgure after detected (mine became more stable after the reconfigure... don't ask me why :) ).

What are you using as coordinator?

joseal, 2021-03-09:
I never tested with that device, but I've 3 CC2531 and one with the RF Amp running without any issues.

Are you using Raspberry Pi 4?

soenkegissel, 2021-03-09:
> I never tested with that device, but I've 3 CC2531 and one with the RF Amp running without any issues.
> 
> Are you using Raspberry Pi 4?
Yes

joseal, 2021-03-09:
Can you bring one device plus the repeater close to the coordinator, just for testing?