Koenkk/zigbee2mqtt #1418: CC2530/CC2531 Router always appears as disconnected

glentakahashi, 2019-04-13:
# Bug Report
## What happened
The CC2530 router always shows as disconnected in home assistant, even though the device is pinging alive every 60 seconds.

I see this in logs:
```
zigbee2mqtt      |   zigbee2mqtt:info 4/13/2019, 2:55:36 AM MQTT publish: topic 'zigbee2mqtt/0x00124b0014d9d4fa', payload '{"state":false,"linkquality":47,"last_seen":"2019-04-13T02:55:36.539Z","device":{"ieeeAddr":"0x00124b0014d9d4fa","friendlyName":"0x00124b0014d9d4fa","type":"Router","nwkAddr":61243,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.router","status":"online"},"elapsed":59991}'
```

The `status` is `online`, yet the `state` variable says it's `false`, which is what homeassistant goes off of.

I also have `availability_timeout` set to 60, and am still seeing the issue.
## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: latest as of 2019-04-12
CC253X firmware version:  latest as of 2019-04-12

Koenkk, 2019-04-13:
The state does represent the LED state, not the disconnected/connected state.

glentakahashi, 2019-04-14:
Hmm, should this be set to `status` then? And change the payload_on and payload_off? https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/homeassistant.js#L84

This is what i see in HA.

[image]

Koenkk, 2019-04-14:
This is indeed highly misleading, could you update the HA integration?

Woodpeckercz, 2019-04-28:
I have the same problem, I use two CC2530 routers and they are constantly shown as offline.

dr-apple, 2019-05-25:
Hello ich have the same with all my routers :-(

MQTT publish: topic 'zigbee2mqtt/0x00124b0018e207bc', payload '{"state":false,"linkquality":170}'

Filipe-Figueiredo, 2019-05-31:
Hello,
I have the same problem:

[image]

Link Quality is ok, but the device appears to be offline.

Koenkk, 2019-06-01:
This should already be fixed in the latest dev branch.

stepanov1975, 2019-06-11:
Sorry, but I still don't understand. When router status is offline is it a bug? If not, is this mean that router not working? And what should I do to fix this?

Koenkk, 2019-06-11:
The state means the led state, not the online/offline state, your router is probably working.

closed 2019-08-17 by stale[bot]