Koenkk/zigbee2mqtt #1419: Failed at the zigbee2mqtt@1.3.0 start script

valentin38315, 2019-04-13:
I tried to start but it comes this error :( :

TypeError: Cannot read property 'base_topic' of null
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/extension/devicePublish.js:7:54)
    at Module._compile (internal/modules/cjs/loader.js:701:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:712:10)
    at Module.load (internal/modules/cjs/loader.js:600:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:539:12)
    at Function.Module._load (internal/modules/cjs/loader.js:531:3)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:22:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/controller.js:14:32)
    at Module._compile (internal/modules/cjs/loader.js:701:30)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.3.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.3.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-04-13T08_19_28_517Z-debug.log

Koenkk, 2019-04-13:
You should add the base topic to your configuration.yaml like shown in http://www.zigbee2mqtt.io/configuration/configuration.html

valentin38315, 2019-04-13:
yes I already had

pcarmo, 2019-05-09:
I'm also having trouble in starting zigbee2mqtt, but due to npm not finding 'isarray' module.
This is the respective log text:

Error: Cannot find module 'isarray'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:582:15)
    at Function.Module._load (internal/modules/cjs/loader.js:508:25)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:22:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/readable-stream/lib/_stream_readable.js:32:15)
    at Module._compile (internal/modules/cjs/loader.js:701:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:712:10)
    at Module.load (internal/modules/cjs/loader.js:600:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:539:12)
    at Function.Module._load (internal/modules/cjs/loader.js:531:3)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.3.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.3.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-05-09T21_45_33_513Z-debug.log

closed 2019-07-09 by Koenkk