Koenkk/zigbee2mqtt #1421: cannot get values from Xiaomi Aqara temperature sensor, value "unavailable"

net47, 2019-04-13:
# Bug Report
## What happened
I'm trying to get temperature (and other) values from im sum three Xiaomi Aqara temperature sensor (see https://www.banggood.com/New-Arrival-Original-Xiaomi-Aqara-Intelligent-Smart-Home-Temperature-Humidity-Sensor-Set-White-p-1148666.html). I'm using HA and zigbee2mqtt as docker containers on my Raspberry Pi 3 (model B+). Pairing at zigbee2mqtt was successful. 

My ```configuration.yaml```:
```
sensor:
  - platform: "mqtt"
    name: "Wohnzimmer Temperatur"
    state_topic: "zigbee2mqtt/temp_01_126a5"
    availability_topic: "zigbee2mqtt/bridge/state"
    unit_of_measurement: '°C'
    value_template: "{{ value_json.temperature | round(1) }}"
    force_update: true
    device_class: "temperature"
mqtt:
  discovery: true
  discovery_prefix: zigbee2mqtt
  #broker: localhost
  birth_message:
    topic: 'hass/status'
    payload: 'online'
  will_message:
    topic: 'hass/status'
    payload: 'offline'
```
Output of zigbee2mqtt container log: 
```
zigbee2mqtt:info 4/13/2019, 9:43:17 AM Logging to directory: '/app/data/log/2019-04-13.09-43-17'
  zigbee2mqtt:info 4/13/2019, 9:43:18 AM Starting zigbee2mqtt version 1.3.0 (commit #unknown)
  zigbee2mqtt:info 4/13/2019, 9:43:18 AM Starting zigbee-shepherd
  zigbee2mqtt:info 4/13/2019, 9:43:20 AM zigbee-shepherd started
  zigbee2mqtt:info 4/13/2019, 9:43:20 AM Coordinator firmware version: '20190223'
  zigbee2mqtt:info 4/13/2019, 9:43:20 AM Currently 3 devices are joined:
  zigbee2mqtt:info 4/13/2019, 9:43:20 AM temp_01_126a5 (0x00158d0002c126a5): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
  zigbee2mqtt:info 4/13/2019, 9:43:20 AM temp_02_7765d (0x00158d0002c7765d): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
  zigbee2mqtt:info 4/13/2019, 9:43:20 AM temp_03_12856 (0x00158d0002c12856): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
  zigbee2mqtt:info 4/13/2019, 9:43:20 AM Zigbee: disabling joining new devices.
  zigbee2mqtt:info 4/13/2019, 9:43:20 AM Connecting to MQTT server at mqtt://localhost
  zigbee2mqtt:info 4/13/2019, 9:43:20 AM zigbee-shepherd ready
  zigbee2mqtt:info 4/13/2019, 9:43:20 AM Connected to MQTT server
  zigbee2mqtt:info 4/13/2019, 9:43:20 AM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 4/13/2019, 9:43:20 AM MQTT publish: topic 'zigbee2mqtt/temp_01_126a5', payload '{"temperature":24.64,"linkquality":55,"humidity":27.65,"pressure":993.7,"battery":100,"voltage":3015}'
  zigbee2mqtt:info 4/13/2019, 9:43:20 AM MQTT publish: topic 'zigbee2mqtt/temp_02_7765d', payload '{"temperature":24.65,"linkquality":78,"humidity":28.16,"pressure":993.8,"battery":100,"voltage":3065}'
  zigbee2mqtt:info 4/13/2019, 9:43:20 AM MQTT publish: topic 'zigbee2mqtt/temp_03_12856', payload '{"temperature":24.51,"linkquality":115,"humidity":27.98,"pressure":994.3,"battery":100,"voltage":3075}'
  zigbee2mqtt:info 4/13/2019, 9:43:20 AM MQTT publish: topic 'homeassistant/sensor/0x00158d0002c126a5/temperature/config', payload '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","state_topic":"zigbee2mqtt/temp_01_126a5","json_attributes_topic":"zigbee2mqtt/temp_01_126a5","name":"temp_01_126a5_temperature","unique_id":"0x00158d0002c126a5_temperature_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0002c126a5","name":"temp_01_126a5","sw_version":"Zigbee2mqtt 1.3.0","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
...
zigbee2mqtt:info 4/13/2019, 9:43:34 AM MQTT publish: topic 'zigbee2mqtt/temp_01_126a5', payload '{"temperature":24.57,"linkquality":76,"humidity":29.97,"pressure":993.7,"battery":100,"voltage":3015}'
```
Tcpdump on HA container:
```
/config # tcpdump -ni any -s0 port 1883
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on any, link-type LINUX_SLL (Linux cooked), capture size 262144 bytes
11:47:36.977473 IP 127.0.0.1.57966 > 127.0.0.1.1883: Flags [P.], seq 3802436226:3802436357, ack 239816240, win 342, options [nop,nop,TS val 1633446660 ecr 1633424254], length 131
11:47:37.022476 IP 127.0.0.1.1883 > 127.0.0.1.57966: Flags [.], ack 131, win 1897, options [nop,nop,TS val 1633446705 ecr 1633446660], length 0
```
## What did you expect to happen
Temperature sensor values, unfortunately I only see "unavailable". 
## How to reproduce it (minimal and precise)
Two docker containers using the above config, a Rasperry Pi 3 Model B+ and the following zigbee2mqtt stick:
https://www.aliexpress.com/item/Wireless-Zigbee-CC2531-Sniffer-Bare-Board-Packet-Protocol-Analyzer-Module-USB-Interface-Dongle-Capture-Packet-Module/32852226435.html?spm=a2g0s.9042311.0.0.248d4c4dVMYYSa
## Debug Info
HA docker container version: ```homeassistant/raspberrypi3-homeassistant 0.91.3```
zigbee2mqtt version: ```1.3.0, container: koenkk/zigbee2mqtt:arm32v6```
CC253X firmware version: ```CC2531_DEFAULT_20190223```

CodeFinder2, 2019-04-13:
I am having the same issue. However, the question is: is this related to zigbee2mqtt or HA? 🙈

hawaltie, 2019-04-14:
I am having issues that could be related to this proble. All my version 2 Xiaomi sensors (WSDCGQ11LM: Xiaomi Aqara temperature, humidity and pressure sensor, RTCGQ11LM: Xiaomi Aqara human body movement and illuminance sensor (occupancy and illuminance) didn't report their values correctly. I de-paired the devices, and when I tried to repair the devices, I recieved the message unknown device. Debug info is comparable with the first post.

CodeFinder2, 2019-04-14:
Surprisingly, one of my sensor now displays correct values (not "unavailable" anymore). I've just rebooted the entire system (in HA, side panel -> "Hass.io" -> "SYSTEM" -> "Host system", "REBOOT").

But other devices still don't work as expected. For example, my Tradfi remote does not even generate messages in zigbee2mqtt's (addon) log output when pushing ANY button on the remote. However, the device appears in my Zigbee map so I assume its paired correctly ...

Two Xiaomi door sensors don't even pair... (Maybe #1401 will help?)

Koenkk, 2019-04-15:
@CodeFinder2 are you running the latest zigbee2mqtt CC253X firmware? this is required for the TRADFRI remote control (https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator)

CodeFinder2, 2019-04-15:
Thank you very much! That did the trick:  i needed to upgrade from 20190109 to 20190223. After repairing, the remote is working fine now. :-)

net47, 2019-04-16:
@Koenkk any ideas about my problem? Can this be something with zigbee2mqtt or should I open an issue at HA side?

Koenkk, 2019-04-16:
@net47 if you see that the values are published by zigbee2mqtt, it's probably not a zigbee2mqtt issue

net47, 2019-04-16:
OK thanks for the info, then I will contact the HA guys.

closed 2019-04-16 by net47

valna, 2019-08-24:
Hey, having the same issue, did u figure it out?

net47, 2019-08-24:
> Hey, having the same issue, did u figure it out?

no, unfortunately not... I went away from HA after that and use the temp sensors as standalone sensors without integration.