Koenkk/zigbee2mqtt #1422: # Bug Report: Osram Smart+ plug autopower on at addon restart

fabioxp7980, 2019-04-13:
## What happened

1 of 2 Osram smart+ plugs connected with zigbee dongle, zigbee2mqtt addon in Hassio installed on raspberry pi 3B+, auto power on at every zigbee2mqtt restart.
The other plug is switched off.
I checked and removed all automations involving this plug so there is no automation that power on the plug, I'm investigating if is the plug itself  but this behaviour is very strange so I decided to open this issue. 

## What did you expect to happen

I expect that a plug doesn't switch on automatically if there isn't an automation that states that, is not a good thing having a plug that power on without reason, if the zigbee2mqtt  addon is restarted

## How to reproduce it (minimal and precise)

In my case evry time I restart zigbee2mqtt addon the plug auto switch on

## Debug Info

zigbee2mqtt version: 1.3.0
CC253X firmware version: coordinator 20180815, router not known, is showed offline in the Zigbee map but I can manually switch on and off the plug, so the connection with coordinator is working, even if the router is not online, anyway the plug auto power on and this is not right.

https://pastebin.com/ACiEuPMB

https://imgur.com/a/eeA7qnM

Koenkk, 2019-04-15:
On line 52 you see that zigbee2mqtt receives an MQTT message to turn of the device:

```
4/13/2019, 4:41:07 PM - debug: Received MQTT message on 'zigbee2mqtt/0x7cb03eaa00b0c7bd/set' with data 'ON'
```

That means it's not a zigbee2mqtt problem.

I expect that it's either:
- Still some automation is running
- There are some retained MQTT messages

fabioxp7980, 2019-04-17:
Ok I will use tools to remove retained MQTT messages and as soon as possible report if I have solved my issue.

fabioxp7980, 2019-04-18:
The issue can be closed, I successfully removed the retained message using MQTT Explorer in Windows, connecting to the mosquitto broker on raspberry and finally don't have any message about that problematic topic.
Thank you so much for the help and patience.

closed 2019-04-19 by Koenkk