Koenkk/zigbee2mqtt #1423: Add Home Decorators Zigbee Fan

dwest576, 2019-04-13:
Would like to add Home Decorators Fan to this system.  I have created a zigbee2mqtt device and am attempting to add 7 ceiling fans to the system.

The ModelID is HDC52EastwindFan

I have tried working through another topic of an addon module that is very close to this:  https://github.com/Koenkk/zigbee2mqtt/issues/115

But whenever I add the HBUniversalCFRemote git, rebuild and restart Z2M it errors out and loops, I must be doing something wrong.  

I have also pulled the device for the git out and changed to my own model:

    // Hampton Bay
    {
        zigbeeModel: ['HDC52EastwindFan'],
        model: 'HDC52EastwindFan',
        vendor: 'Hampton Bay',
        description: 'Universal wink enabled white ceiling fan premier remote control',
        supports: 'on/off',
        fromZigbee: [
            fz.generic_state, fz.light_brightness, fz.ignore_fan_change, fz.generic_fan_mode, fz.ignore_onoff_change,
            fz.ignore_level_report,
        ],
        toZigbee: [tz.onoff, tz.light_brightness_onoff, tz.fan_mode],
        meta: {
            fan_mode: {
                'off': 0,
                'low': 1,
                'medium': 2,
                'medium-high': 3,
                'high': 4,
                'breeze': 6,
            },
        },
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.report('genOnOff', 'onOff', 0, 1000, 0, cb),
                (cb) => device.bind('genLevelCtrl', coordinator, cb),
                (cb) => device.report('genLevelCtrl', 'currentLevel', 0, 1000, 0, cb),
                (cb) => device.bind('hvacFanCtrl', coordinator, cb),
                (cb) => device.report('hvacFanCtrl', 'fanMode', 0, 1000, 0, cb),
            ];

            execute(device, actions, callback);
        },
    },

Whenever Z2M receives any zigbee signals, the system restarts.   I think I probably need to start to add my own device from scratch, but not sure where to start.   Any help would be appreciated.

zigbee2mqtt version: 1.3.0
CC253X firmware version:  '20190223'

Koenkk, 2019-04-15:
Can you provide the log when it crashes?

dwest576, 2019-04-15:
When I add the Hampton Bay fan per Issue #115,

cd node_modules
rm -rf zigbee-shepherd-converters
git clone https://github.com/Koenkk/zigbee-shepherd-converters.git -b HBUniversalCFRemote
cd zigbee-shepherd-converters
npm install

I get:
Apr 15 14:03:59 931-HUB-LOFT npm[611]: npm ERR!     /home/pi/.npm/_logs/2019-04-15T21_03_59_494Z-debug.log
Apr 15 14:03:59 931-HUB-LOFT systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Apr 15 14:03:59 931-HUB-LOFT systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Apr 15 14:03:59 931-HUB-LOFT systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Apr 15 14:03:59 931-HUB-LOFT systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Apr 15 14:03:59 931-HUB-LOFT systemd[1]: Stopped zigbee2mqtt.
Apr 15 14:03:59 931-HUB-LOFT systemd[1]: Started zigbee2mqtt.
Apr 15 14:04:01 931-HUB-LOFT npm[643]: > zigbee2mqtt@1.3.0 start /opt/zigbee2mqtt
Apr 15 14:04:01 931-HUB-LOFT npm[643]: > node index.js
Apr 15 14:04:02 931-HUB-LOFT npm[643]:   zigbee2mqtt:info 4/15/2019, 2:04:02 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-04-15.14-04-02'
Apr 15 14:04:02 931-HUB-LOFT npm[643]: /opt/zigbee2mqtt/lib/extension/devicePublish.js:13
Apr 15 14:04:02 931-HUB-LOFT npm[643]:     zigbeeShepherdConverters.toZigbeeConverters.light_onoff_brightness,
Apr 15 14:04:02 931-HUB-LOFT npm[643]:                                                 ^
Apr 15 14:04:02 931-HUB-LOFT npm[643]: TypeError: Cannot read property 'light_onoff_brightness' of undefined
Apr 15 14:04:02 931-HUB-LOFT npm[643]:     at Object.<anonymous> (/opt/zigbee2mqtt/lib/extension/devicePublish.js:13:49)
Apr 15 14:04:02 931-HUB-LOFT npm[643]:     at Module._compile (internal/modules/cjs/loader.js:701:30)
Apr 15 14:04:02 931-HUB-LOFT npm[643]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:712:10)
Apr 15 14:04:02 931-HUB-LOFT npm[643]:     at Module.load (internal/modules/cjs/loader.js:600:32)
Apr 15 14:04:02 931-HUB-LOFT npm[643]:     at tryModuleLoad (internal/modules/cjs/loader.js:539:12)
Apr 15 14:04:02 931-HUB-LOFT npm[643]:     at Function.Module._load (internal/modules/cjs/loader.js:531:3)
Apr 15 14:04:02 931-HUB-LOFT npm[643]:     at Module.require (internal/modules/cjs/loader.js:637:17)
Apr 15 14:04:02 931-HUB-LOFT npm[643]:     at require (internal/modules/cjs/helpers.js:22:18)
Apr 15 14:04:02 931-HUB-LOFT npm[643]:     at Object.<anonymous> (/opt/zigbee2mqtt/lib/controller.js:14:32)
Apr 15 14:04:02 931-HUB-LOFT npm[643]:     at Module._compile (internal/modules/cjs/loader.js:701:30)
Apr 15 14:04:02 931-HUB-LOFT npm[643]: npm ERR! code ELIFECYCLE
Apr 15 14:04:02 931-HUB-LOFT npm[643]: npm ERR! errno 1
Apr 15 14:04:02 931-HUB-LOFT npm[643]: npm ERR! zigbee2mqtt@1.3.0 start: `node index.js`
Apr 15 14:04:02 931-HUB-LOFT npm[643]: npm ERR! Exit status 1
Apr 15 14:04:02 931-HUB-LOFT npm[643]: npm ERR!
Apr 15 14:04:02 931-HUB-LOFT npm[643]: npm ERR! Failed at the zigbee2mqtt@1.3.0 start script.
Apr 15 14:04:02 931-HUB-LOFT npm[643]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Apr 15 14:04:02 931-HUB-LOFT npm[643]: npm ERR! A complete log of this run can be found in:
Apr 15 14:04:02 931-HUB-LOFT npm[643]: npm ERR!     /home/pi/.npm/_logs/2019-04-15T21_04_02_869Z-debug.log
Apr 15 14:04:02 931-HUB-LOFT systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Apr 15 14:04:02 931-HUB-LOFT systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Apr 15 14:04:02 931-HUB-LOFT systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Apr 15 14:04:03 931-HUB-LOFT systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.

Only way for me to get it to work again is to delete and reload Z2M

dwest576, 2019-04-15:
For the other error, when getting a signal from the zigbee fan, I will need to reproduce this as I have to reload my Z2M controller again and setup Z2M the way I had it.

I was hoping hoping I could simply convert the HB fan code over to the Home Decorator.   I am using a in wall HB remote to talk to mine, so I was thinking the controls are very similar.

Koenkk, 2019-04-17:
#1357 tries to integrate exactly the same device. The required changes are available there. Let's continue there.

closed 2019-04-17 by Koenkk

dwest576, 2019-04-17:
Will 1357 fix my issue, as the error being thrown in the logs is:  TypeError: Cannot read property 'light_onoff_brightness' of undefined

It appears that something is happening to the devices.js file when I git pull HBUniversalCFRemote and recompile

Koenkk, 2019-04-17:
yes, the HBUniversalCFRemote is a very old out-of-date  branch which doesn't work

dwest576, 2019-04-17:
Got it.   I'll try resetting everything today and adding the above to see how it goes.   Appreciate your help.