Koenkk/zigbee2mqtt #1424: Network stuck at ping to Xiaomi Wall Switch

lenisko, 2019-04-14:
# Bug Report
## What happened
Whole network got blocked/stuck after unsuccessful ping to one of devices (Xiaomi Wall Switch - which is working just fine manually).
It has happened after few hours of non-problematic work.

## What did you expect to happen
Not block whole network? :-)

## How to reproduce it (minimal and precise)
No idea.

## Debug Info
zigbee2mqtt version: 1.3.0 / 00ebd441a0f804e1bbcae4c2a03cbe9698b58010
CC253X firmware version: 20190223 default
log: https://hastebin.com/ginogokupa.log (starting at `4/13/2019, 9:58:53 PM`)
configuration:
```
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://192.168.1.1'
serial:
  port: /dev/zigbee
advanced:
  report: false
  rtscts: false
  channel: 11
  baudrate: 115200
  log_level: debug
  log_directory: data/log/
  cache_state: true
  soft_reset_timeout: 0
  network_key: ...
devices: devices.yaml
groups: groups.yaml
```

Koenkk, 2019-04-15:
It doesn't seem to block the network, as you still see zigbee messages being received, are you still able to control devices?

lenisko, 2019-04-15:
@Koenkk only
```
4/13/2019, 10:44:53 PM - debug: Ping 0x00158d00029c2ce5 (basic)
4/13/2019, 10:44:55 PM - error: Failed to ping 0x00158d00029c2ce5
4/13/2019, 10:45:53 PM - debug: Ping 0x00158d00029c2ce5 (basic)
4/13/2019, 10:45:55 PM - error: Failed to ping 0x00158d00029c2ce5
4/13/2019, 10:46:53 PM - debug: Ping 0x00158d00029c2ce5 (basic)
4/13/2019, 10:46:55 PM - error: Failed to ping 0x00158d00029c2ce5
4/13/2019, 10:47:53 PM - debug: Ping 0x00158d00029c2ce5 (basic)
4/13/2019, 10:47:55 PM - error: Failed to ping 0x00158d00029c2ce5
4/13/2019, 10:48:53 PM - debug: Ping 0x00158d00029c2ce5 (basic)
4/13/2019, 10:48:55 PM - error: Failed to ping 0x00158d00029c2ce5
4/13/2019, 10:49:51 PM - debug: Saving state to file /home/hass/zigbee2mqtt/data/state.json
```
been present in log, I wasn't able to control anything nor get status of sensors.

EDIT
But.. I'm not sure if that was reason of problem.
I found some weird device?? in my `devices.conf` file:
```
4/11/2019, 2:33:00 AM - info: 0x00158d000211353b (0x00158d000211353b): unkown - undefined unknown (EndDevice)
```
after removing it via mqtt and from config, looks like things are working again. 
I will monitor it for 2-3 more days and let you know.

lenisko, 2019-04-15:
@Koenkk problem showed itself again, so it's not caused by unknown device
https://hastebin.com/ivurojacuf.txt

Network is not reacting to anything, no reports from sensors as well till I will replug & restart it.

Koenkk, 2019-04-17:
Can you try with the max stability firmware? https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator

lenisko, 2019-05-23:
Helped, thanks

closed 2019-05-23 by lenisko