Koenkk/zigbee2mqtt #1425: No leak payload on aqara water leak SJCGQ11LM

skjnldsv, 2019-04-14:
- [x] First make sure you are on the latest version of zigbee2mqtt.
  `zigbee2mqtt@1.3.0 start /zigbee2mqtt-1.3.0`
- [x] Make sure you are running the latest CC253X firmware from https://github.com/koenkk/z-stack-firmware.
  `017a592a042b09f8a348717627a25196  CC2531ZNP-Prod.hex`
  from https://github.com/Koenkk/Z-Stack-firmware/commit/e98e0e8c2b5826ad4d29a1046bfeaf78c08b4c53
- [x] Provide a clear description of the problem.
- [x] If applicable, provide steps how to reproduce the problem.
- [x] Provide a log with `log_level` set to `debug`, see https://www.zigbee2mqtt.io/configuration/configuration.html
  https://bin.skjnldsv.com/?97631df8c3524446#y+xOEoJMWiiUQnBNpoO/ElQyQWpRXSlTYO7cuHmC/yY=

# Bug Report
## What happened
After pairing my aqara water leak sensor, I only get one payload on zigbeemqtt start and then nothing unless I press the sensor. And if I do, I only get a payload with battery, link quality and voltage, not the sensor state:
`zigbee2mqtt:info 4/14/2019, 9:48:06 AM MQTT publish: topic 'zigbee2mqtt/aqara_water_leak', payload '{"battery":97,"voltage":2995,"linkquality":42}'`
If I put the sensor in water, I get no notification or state change.

## What did you expect to happen
Water leak sensor should warn if in water (or send more frequent payload?)

## How to reproduce it (minimal and precise)
1. Install hassio
2. Install mqtt mosquito
3. install zigbeemqtt@latest
4. permit_join true: pair device
5. permit_join false and restart zigbeemqtt: see initial device payload

## Debug Info
zigbee2mqtt version: 1.3.0
CC253X firmware version: CC2531_DEFAULT_20190223

Koenkk, 2019-04-18:
Can you provide the `log_level: debug` when putting the sensor in the water? http://www.zigbee2mqtt.io/configuration/configuration.html

skjnldsv, 2019-04-18:
@Koenkk Hi, it does not trigger. No payload is sent when in water :woman_shrugging:

skjnldsv, 2019-04-18:
Okay, I tried pressing the button while in water and when coming out I received this which I never saw before:
```
  zigbee2mqtt:debug 4/18/2019, 6:13:38 PM Received zigbee message of type 'endDeviceAnnce' with data '"0x00158d000281f58a"' of device 'lumi.sensor_wleak.aq1' (0x00158d000281f58a) of endpoint 1
  zigbee2mqtt:debug 4/18/2019, 6:13:43 PM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"1":3005,"3":26,"4":17320,"5":54,"6":[0,1],"8":516,"10":0,"100":0}}}' of device 'lumi.sensor_wleak.aq1' (0x00158d000281f58a) of endpoint 1
  zigbee2mqtt:info 4/18/2019, 6:13:43 PM MQTT publish: topic 'zigbee2mqtt/aqara_water_leak', payload '{"battery":100,"voltage":3005,"linkquality":28}'
  zigbee2mqtt:debug 4/18/2019, 6:13:43 PM Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"65281":[null,3005,null,26,17320,54,[0,1],null,516,null,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,0]}}' of device 'lumi.sensor_wleak.aq1' (0x00158d000281f58a) of endpoint 1
  zigbee2mqtt:debug 4/18/2019, 6:14:30 PM Received zigbee message of type 'endDeviceAnnce' with data '"0x00158d000281f58a"' of device 'lumi.sensor_wleak.aq1' (0x00158d000281f58a) of endpoint 1
  zigbee2mqtt:debug 4/18/2019, 6:14:35 PM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"1":3005,"3":25,"4":13224,"5":54,"6":[0,1],"8":516,"10":0,"100":0}}}' of device 'lumi.sensor_wleak.aq1' (0x00158d000281f58a) of endpoint 1
  zigbee2mqtt:info 4/18/2019, 6:14:35 PM MQTT publish: topic 'zigbee2mqtt/aqara_water_leak', payload '{"battery":100,"voltage":3005,"linkquality":39}'
  zigbee2mqtt:debug 4/18/2019, 6:14:35 PM Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"65281":[null,null,null,25,13224]}}' of device 'lumi.sensor_wleak.aq1' (0x00158d000281f58a) of endpoint 1
```

Koenkk, 2019-04-19:
Can you post a few of these message? When putting it in and out of the water (make sure to also dry the sensor after getting it out of the water), and indicate which messages came from putting it in the water and which messages came from drying it.

ccorderod, 2019-04-20:
I do have the same issue with the leak sensors. When imput them in contact with water, the blue light blinks one time, but no msg is logged in z2m. Looks like the sensor is not paired but I can’t manage to reset and pair them again (pressing button 5sg then releasing and keeping pressing the button). Any advise?
Thank you.

ccorderod, 2019-04-21:
Hello. Fixed.
In my case, the leak sensors were not binded with the cc, hence no changestatus frame was received by z2m. I kept my system running all the night with  "log_level": "debug", and "zigbee_shepherd_debug": true on the config. This morning, I had a look at the log files (there were a few), looking at ping frames from the leak sensors. There were NO such frames getting in (I read somewhere those sensors send each hour (+/-) a status frame with batt level). 
I deleted in database.db the entries for those sensors, and restarted the join process. This time:
- push the drop (and keep it pushing for a few seconds till the blue light flash 3 times) AND
- push the drop each 5 seconds for a while (running an mqtt spky you will receive at some point a homeassistant/bridge/log topic with {"type":"pairing","message":"device incoming"} payload; after a few seconds, the sensor is joined.
They work like a charm now. @Koenkk, thank you again for a wonderfull piece of software.
C.

skjnldsv, 2019-04-21:
@Koenkk sure!
### Boot 
https://bin.skjnldsv.com/?56df534ecbe1f071#P3uY8a9fwNA5aL1IK8xwpmJ5hdl7A/CO5HgpUmTB8z0=
### Pressing button once out of water
https://bin.skjnldsv.com/?923f5494dd56cfb7#ZFpw9Uy4+4rVWuEGOEstDmROm0qLXGdayW7SEwpuTWg=
### Putting in water for 5 min
---> no log aside from
`zigbee2mqtt:debug 4/21/2019, 9:32:33 AM Saving state to file /share/zigbee2mqtt/state.json`
### Pressing while in water
---> no logs (blue light flash once)
### Drying it and pressing out of water:
`zigbee2mqtt:debug 4/21/2019, 9:36:32 AM Received zigbee message of type 'endDeviceAnnce' with data '"0x00158d000281f58a"' of device 'lumi.sensor_wleak.aq1' (0x00158d000281f58a) of endpoint 1`
`zigbee2mqtt:debug 4/21/2019, 9:36:37 AM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"1":3005,"3":22,"4":17320,"5":54,"6":[0,1],"8":516,"10":0,"100":0}}}' of device 'lumi.sensor_wleak.aq1' (0x00158d000281f58a) of endpoint 1`
`zigbee2mqtt:info 4/21/2019, 9:36:37 AM MQTT publish: topic 'zigbee2mqtt/aqara_water_leak', payload '{"battery":100,"voltage":3005,"linkquality":42}'`
`zigbee2mqtt:debug 4/21/2019, 9:36:37 AM Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"65281":[null,null,null,null,null,null,[null,1]]}}' of device 'lumi.sensor_wleak.aq1' (0x00158d000281f58a) of endpoint 1
`
### Pressing again
`zigbee2mqtt:debug 4/21/2019, 9:37:22 AM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"1":3005,"3":22,"4":17320,"5":54,"6":[0,0],"8":516,"10":0,"100":0},"modelId":"lumi.sensor_wleak.aq1"}}' of device 'lumi.sensor_wleak.aq1' (0x00158d000281f58a) of endpoint 1`
`zigbee2mqtt:info 4/21/2019, 9:37:22 AM MQTT publish: topic 'zigbee2mqtt/aqara_water_leak', payload '{"battery":100,"voltage":3005,"linkquality":39}'`
`zigbee2mqtt:debug 4/21/2019, 9:37:22 AM Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"65281":[null,3005,null,22,17320,54,[0,0],null,516,null,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,0]}}' of device 'lumi.sensor_wleak.aq1' (0x00158d000281f58a) of endpoint 1`

@ccorderod did you received messages like me before? When you put it in water, does it trigger instantly?
I'll try re-pairing again.

skjnldsv, 2019-04-21:
If I try to re-pair, I get this: `zigbee2mqtt:error 4/21/2019, 9:51:22 AM Cannot get the Node Descriptor of the Device: 0x00158d000281f58a (Error: Timed out after 10000 ms)`

skjnldsv, 2019-04-21:
If I try to re-pair, I get this: `zigbee2mqtt:error 4/21/2019, 9:51:22 AM Cannot get the Node Descriptor of the Device: 0x00158d000281f58a (Error: Timed out after 10000 ms)`

I cannot pair the device again. Battery have been changed, issue still occurred :woman_shrugging: 
EDIT: removed entries and settings in home assistant and other configs, and was able to pair again.

@Koenkk here the full pairing log, there are some odd lines in it: https://bin.skjnldsv.com/?b3f113923f19cd4c#FksbzlgAxehR3Sd/oFSAp7BIDER0Oby7eipdJdVut1A=

@ccorderod could you paste the sensor entry from your database.db ?
Mine looks like this: `{"id":2,"type":"EndDevice","ieeeAddr":"0x00158d000281f58a","nwkAddr":30408,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_wleak.aq1","epList":[1],"zclVersion":1,"appVersion":4,"stackVersion":2,"hwVersion":2,"swBuildId":"3000-0001","status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}}}}},"dateCode":"20170721","_id":"fA5SZFdWwF0DD7aF"}`

ccorderod, 2019-04-21:
2 entries. here you go:
{"id":20,"type":"EndDevice","ieeeAddr":"0xxxxxxxxxxxxxxxxxx","nwkAddr":58395,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_wleak.aq1","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}}}}},"_id":"zVZzJcf5ySay2MC1"}
{"id":20,"type":"EndDevice","ieeeAddr":"0xxxxxxxxxxxxxxxxxx","nwkAddr":58395,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_wleak.aq1","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}}}}},"_id":"zVZzJcf5ySay2MC1"}

skjnldsv, 2019-04-21:
@ccorderod And it works when you put it in the water?

ccorderod, 2019-04-22:
yes John, all of them and at speed light!. I have 9 of them joined.

skjnldsv, 2019-04-22:
Damn!
@Koenkk Any clue? :disappointed:

Koenkk, 2019-04-23:
@skjnldsv besides re-pairing and maybe testing with the original gateway (to check if the device itself is the issue), no.

skjnldsv, 2019-06-22:
Yep, I gave up.
Nowhere close to have it working :(

KaiboshOz, 2019-08-06:
Not a fix, but a workaround that worked for me on the 4 water leak sensors that I have - after pairing and setting "retain: true" in my devices.yaml file for zigbee2mqtt (and restarting zigbee2mqtt), I manually triggered each sensor (just touching both terminals with two fingers on the same hand - no need to wet it to test).

That turned the payload from '{"battery":100,"voltage":3015,"linkquality":31}' to '{"battery":100,"voltage":3015,"linkquality":31,"water_leak":true}', and then back to false for water_leak.
The water_leak field is now present in all messages from the sensors, and remains there after HA & zigbee2mqtt restarts, updating correctly when sensor is triggered.

zigbee2mqtt (hassio addon) version: 1.5.1.1
CC253X firmware version: CC2531_20190425  (default)

CASATAJO, 2019-09-30:
Info have same problem with 3 leak sensors same type. Did you touch the two terminals while adding the sensor?

KaiboshOz, 2019-09-30:
@CASATAJO - nope, just normal pairing using the "button" in the middle of the device under the water symbol.  Nothing extra/unusual while pairing beyond the usual slightly finicky Xiaomi device pairing and having to press the button every couple of seconds during pairing.

Touching the sensors to set it off is only after it's paired, and the zigbee2mqtt device config is updated to "retain: true" (and restarting zigbee2mqtt to make it effective; and then restarting HA after that, which I'm pretty sure isn't necessary but since I'm restarting stuff anyway).

closed 2019-12-06 by stale[bot]