Koenkk/zigbee2mqtt #1426: Pairing xiaomi aqara motion RTCGQ11LM device unsupported

jlo88, 2019-04-14:
# Bug Report
## What happened
I'm trying to pair a xiaomi aqara RTCGQ11LM, in the zigbee2mqtt log I see device incoming and pairing but it reports as undefined. It also does not show up under integrations in home assistant. It does show up in devices.yaml:

'0x00158d0003665269':
  friendly_name: '0x00158d0003665269'
  retain: false

 and in the database: 

{"id":38,"type":"EndDevice","ieeeAddr":"0x00158d0003665269","nwkAddr":37537,"manufId":4151,"epList":[1],"status":"online","joinTime":1555232692,"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,1,3,1024,1030,1280,65535],"outClusterList":[0,25],"clusters":{}}},"_id":"ukotEDsNL7iZ8aKk"}
{"id":38,"type":"EndDevice","ieeeAddr":"0x00158d0003665269","nwkAddr":37537,"manufId":4151,"epList":[1],"status":"online","joinTime":1555232692,"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,1,3,1024,1030,1280,65535],"outClusterList":[0,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"msIlluminanceMeasurement":{"dir":{"value":1},"attrs":{}},"msOccupancySensing":{"dir":{"value":1},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"ukotEDsNL7iZ8aKk"}

## What did you expect to happen
Pairing and showing up of the device in home assistant.

## How to reproduce it (minimal and precise)
Stop zigbee2mqtt
remove the device from devices.yaml and database.db 
start zigbee2mqtt
pair the device 

## Debug Info
zigbee2mqtt version: Edge 1.3.0
CC253X firmware version: 20190315

log: https://pastebin.com/zWWjiMVs

Koenkk, 2019-04-17:
Should be fixed in the latest dev branch (please checkout latest edge in +- 2 hours). 

You should be easy to simply fix this by pressing the reset button on the device once.

skjnldsv, 2019-04-18:
Similar to #1425 ?

esas, 2019-04-25:
I can confirm that it now works..

skjnldsv, 2019-04-26:
@esas any specific procedure you followed?

esas, 2019-04-26:
Just used the latest-dev docker image and that worked. I couldn't find the changes on git though.

skjnldsv, 2019-04-26:
@esas and clean install, long press, allow pairing , wait for the paired notification in log and that's it?

esas, 2019-04-26:
Actually I kept the old config files. Didn't have to repair. The device was already in the configuration.yaml. It just worked as soon as the now container (tag: latest-dev) started. Before this I had the same problems as you described running ver. 1.3.0.

jlo88, 2019-04-28:
I can also confirm that this is working now with 1.3.1. I paired a sensor yesterday without any problems.

Koenkk, 2019-04-29:
Great, looks that this has been solved, feel free to comment in case you are still having problems.

closed 2019-04-29 by Koenkk