Koenkk/zigbee2mqtt #1427: CC2530 startup failing

gadolithium, 2019-04-14:
I have 2 coordinator devices to test this: the CC2531 works fine, the CC2530 worked fine until a couple of days ago.
I reflashed the firmware to be sure : revison is 20190223 (cc-debugger works just fine, i can read, write and verify from the device so it probably is OK)
Starting zigbee2mqtt with debug log doesn't help a lot unfortunately. I checked the serial parameters and rtscts (which did not changed). Any ideas how to debug this ?

Debug log attached:

0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.4.1
3 info using node@v10.14.2
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.3.0~prestart: zigbee2mqtt@1.3.0
6 info lifecycle zigbee2mqtt@1.3.0~start: zigbee2mqtt@1.3.0
7 verbose lifecycle zigbee2mqtt@1.3.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.3.0~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/app/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/u
sr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle zigbee2mqtt@1.3.0~start: CWD: /app
10 silly lifecycle zigbee2mqtt@1.3.0~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.3.0~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.3.0~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.3.0 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
13 verbose stack     at EventEmitter.emit (events.js:182:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:182:13)
13 verbose stack     at maybeClose (internal/child_process.js:962:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
14 verbose pkgid zigbee2mqtt@1.3.0
15 verbose cwd /app
16 verbose Linux 4.9.0-8-amd64
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v10.14.2
19 verbose npm  v6.4.1
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.3.0 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.3.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

bartplessers, 2019-04-14:
Probably the same problem here:
Hardware:
- RPI 2
- lrwxrwxrwx 1 root root 13 Apr 15 00:03 usb-Texas_Instruments_TI_CC2531_USB_CDC___0XC8DF8435CB67FFFF-if00 -> ../../ttyACM0

Starting zigbee2mqtt:
====================================
```
pi@rpi-83:/opt/zigbee2mqtt $ sudo npm start

> zigbee2mqtt@1.3.0 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 4/15/2019, 12:09:34 AM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-04-15.00-09-34'
  zigbee2mqtt:info 4/15/2019, 12:09:36 AM Starting zigbee2mqtt version 1.3.0 (commit #00ebd44)
  zigbee2mqtt:info 4/15/2019, 12:09:36 AM Starting zigbee-shepherd
  zigbee2mqtt:info 4/15/2019, 12:09:43 AM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
  zigbee2mqtt:info 4/15/2019, 12:10:43 AM Starting zigbee-shepherd
  zigbee2mqtt:error 4/15/2019, 12:10:51 AM Error while starting zigbee-shepherd!
  zigbee2mqtt:error 4/15/2019, 12:10:51 AM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 4/15/2019, 12:10:51 AM Failed to start
        {"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:268:22)\n    at Object.onceWrapper (events.js:277:13)\n    at CcZnp.emit (events.js:189:13)\n    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:247:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
  zigbee2mqtt:error 4/15/2019, 12:10:51 AM Exiting...
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.3.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.3.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-04-14T22_11_26_685Z-debug.log
pi@rpi-83:/opt/zigbee2mqtt $
```
====================================

In the logfile:
====================================
```
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.4.1
3 info using node@v10.15.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.3.0~prestart: zigbee2mqtt@1.3.0
6 info lifecycle zigbee2mqtt@1.3.0~start: zigbee2mqtt@1.3.0
7 verbose lifecycle zigbee2mqtt@1.3.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.3.0~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle zigbee2mqtt@1.3.0~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.3.0~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.3.0~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.3.0~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.3.0 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
13 verbose stack     at EventEmitter.emit (events.js:189:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:189:13)
13 verbose stack     at maybeClose (internal/child_process.js:970:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid zigbee2mqtt@1.3.0
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.14.98-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v10.15.3
19 verbose npm  v6.4.1
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.3.0 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.3.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
```
====================================

Only: I'm not sure if I really have a CC2531.
on the chip, following is marked
```
CC2540
F256
TI 7AI
A3DC G4

```

And on the flash software (see attachment), also
**CC2540**
[image]

However, flashing seems okay and device is also announced as 
Texas_Instruments_TI_**CC2531**_USB_CDC___0XC8DF8435CB67FFFF-if00 

Any help is appreciated!

Kind regards,
Bart Plessers

Koenkk, 2019-04-15:
@bartplessers you have a CC2540, the serial ID probably comes from the firmware.

closed 2019-06-21 by stale[bot]

bartplessers, 2019-08-21:
@Koenkk : ok, sorry for this inconvenience. Received a CC2531 and things are more ore less working now. Thanx for answer!