Koenkk/zigbee2mqtt #1429: Z-Stack 3 on CC2652R adventures

Koenkk, 2019-04-14:
Texas Instruments has recently released a new high performance Zigbee chip: the CC2652R. This chip is much faster and contains much more memory than the CC2530/CC2531. 

This issue is created in order to gather feedback of users using the CC2652R **AND** Z-Stack 3.0 (= Zigbee 3.0)

**Notes**
- This is experimental! (no guarantees)
- The device can be bought here: http://www.ti.com/tool/launchxl-cc26x2r1
- No additional flashing hardware is needed, it has an on-board programmer (no CC debugger needed).
- The board has +- the size of a Raspberry Pi
- Zigbee 3.0!
- Flash using [Flash Programmer v2](http://www.ti.com/tool/FLASH-PROGRAMMER)
- Firmware can be found here in the bin folder: https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/Z-Stack_3.x.0
- It's not possible to update your Zigbee 1.2 network (requires repairing)
- The network restore trick that we used in Zigbee 1.2 doesn't work anymore (due to better security policies). A backup of the CC2652R is taken at every shutdown of zigbee2mqtt, file can be found under `data/coordinator_backup.json`. This backup is automatically restored when starting with a new flashed CC2652R.
- Requires latest zigbee2mqtt dev branch.

JLFN, 2019-04-14:
@Koenkk can this board be bought from inside of Europe? I need to pay extra taxes if bought outside.

Koenkk, 2019-04-14:
@JLFN mine shipped from the Netherlands.

lolorc, 2019-04-14:
mine too
I was able to flash it with uniflash on linux.

JLFN, 2019-04-14:
> @JLFN mine shipped from the Netherlands.

Can you please give me link of the page where you bought it from that shipped from the NL.

Koenkk, 2019-04-14:
@JLFN just from TI.

Koenkk, 2019-04-14:
The Xiaomi end device undefined model ID issue had nothing to do with the CC2652R, it was actually a zigbee2mqtt bug, fixed in latest dev branch.

dh-harald, 2019-04-15:
I've got mine today morning (from NL as well). Today, there will be a big migration project :)

tunip, 2019-04-15:
Before I also "jump on the adventure train", I have 2 questions.

1. Range
Could someone rate the range of this device? Could we connect a antenna to increase the range?

2. Router
Is this device only for a coordinator role? Or would we see also in the future a router firmware?

Thx

salopette, 2019-04-15:
Hello, does that part have an antenna? What is the range of that? and pays off the switch from cc2531 to cc2652r ????

lolorc, 2019-04-15:
@salopette it has a pcb antenna and an unconnected connector, see https://e2e.ti.com/support/wireless-connectivity/bluetooth/f/538/t/723337

lolorc, 2019-04-15:
I only have osram plugs and aqara devices, aqara temp/humidity sensors are mostly ok to rebind, but it's not as straightforward for the other ones.

dh-harald, 2019-04-15:
I'm done with re-pairing... Looks good... Re-pairing done with QBKG11LM, QBKG04LM, RTCGQ11LM, GL-C-008. Link qualities are better than CC2531, but I still can't create a proper mesh network (based on network map). Routers are connected directly to the controller and that's all, that I can see...

lolorc, 2019-04-15:
i've been able to pair MCCGQ01LM, MFKZQ01LM and DJT11LM, but still there's something very wrong with aqara devices.
With the aqara gateway and another alternative gw (zigate), the pairing action is very straightforward, it only needs one action on the device, with those 3 models I had to do something like pressing the pairing button several times per second after the initial long press action, worst was DJT11LM

Koenkk, 2019-04-16:
@dh-harald do you see big improvements in link quality?

dh-harald, 2019-04-16:
@Koenkk with CC2531, I had 50-70 link quality, not I've 100-120.

JLFN, 2019-04-16:
@Koenkk  What is the maximum devices that can be added, it say that max memory is 
– 352KB of In-System Programmable Flash
– 256KB of ROM for Protocols and Firmware

lolorc, 2019-04-16:
This morning it looked like my cc2652 had crashed. z2m wasn't able to use its serial port anymore. Had to restart rpi. Most of the pairings have been lost.

Link quality looks also better than what I had with cheap cc2530+cc2591.

lolorc, 2019-04-16:
@tunip range looks better, antenna should be possible as there's a connector.
router should be doable, but currently the changes to build a router firmware are closed-source.

dh-harald, 2019-04-16:
> This morning it looked like my cc2652 had crashed. z2m wasn't able to use its serial port anymore. Had to restart rpi. Most of the pairings have been lost.
I've something similar ATM... All devices are offline, can't ping by the controller

lolorc, 2019-04-16:
@dh-harald if it was connected to a PC, I would try to use uniflash to reset the board.
Myself i need to find a way to program the board from my linux pc, I'll give a try to ser2net+socat.

dh-harald, 2019-04-16:
> @dh-harald if it was connected to a PC, I would try to use uniflash to reset the board.
> Myself i need to find a way to program the board from my linux pc, I'll give a try to ser2net+socat.

It's an arm64 box, so I'm afraid, I can't run uniflash on it :(

tunip, 2019-04-16:
@dh-harald 
Maybe this could help you:
https://askubuntu.com/questions/1036341/unplug-and-plug-in-again-a-usb-device-in-the-terminal
https://askubuntu.com/questions/645/how-do-you-reset-a-usb-device-from-the-command-line
https://unix.stackexchange.com/questions/165447/turning-off-power-to-usb-port-or-turn-off-power-to-entire-usb-subsystem

dh-harald, 2019-04-16:
@tunip I already tried them, but it didn't work.

Koenkk, 2019-04-16:
Mine also crashed after 2 days (Request timeout errors)

lolorc, 2019-04-16:
the mqtt message with empty action are not related to cc2652, are they ?
```
  serialport/stream binding.read finished +1ms
  cc-znp { sof: 254,
  cc-znp   len: 28,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'AF',
  cc-znp   cmd: 'incomingMsg',
  cc-znp   payload:
  cc-znp    { groupid: 0,
  cc-znp      clusterid: 18,
  cc-znp      srcaddr: 64732,
  cc-znp      srcendpoint: 2,
  cc-znp      dstendpoint: 1,
  cc-znp      wasbroadcast: 0,
  cc-znp      linkquality: 108,
  cc-znp      securityuse: 0,
  cc-znp      timestamp: 5007337,
  cc-znp      transseqnumber: 0,
  cc-znp      len: 8,
  cc-znp      data: <Buffer 18 38 0a 55 00 21 02 00> },
  cc-znp   fcs: 97,
  cc-znp   csum: 97 } +21ms
  serialport/stream _read reading +18ms
  serialport/binding-abstract read +19ms
  serialport/bindings/unixRead Starting read +18ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 18, srcaddr: 64732, srcendpoint: 2, dstendpoint: 1, wasbroadcast: 0, linkquality: 108, securityuse: 0, timestamp: 5007337, transseqnumber: 0, len: 8, data: <Buffer 18 38 0a 55 00 21 02 00> } +35ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":18,"srcaddr":64732,"srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"linkquality":108,"securityuse":0,"timestamp":5007337,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,56,10,85,0,33,2,0]}} +126ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +39ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +6ms
  serialport/bindings/poller Polling for "readable" +27ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":18,"srcaddr":64732,"srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"linkquality":108,"securityuse":0,"timestamp":5007337,"transseqnumber":0,"len":8,"data":{"0":24,"1":56,"2":10,"3":85,"4":0,"5":33,"6":2,"7":0},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":56,"cmdId":"report","payload":[{"attrId":85,"dataType":33,"attrData":2}]}} +5ms
  zigbee2mqtt:debug 4/16/2019, 8:13:05 PM Received zigbee message of type 'attReport' with data '{"cid":"genMultistateInput","data":{"presentValue":2}}' of device 'lumi.sensor_cube' (cube_02) of endpoint 2
  zigbee2mqtt:info 4/16/2019, 8:13:05 PM MQTT publish: topic 'zigbee2mqtt/cube_02', payload '{"action":"wakeup","linkquality":108,"last_seen":1555438385033,"elapsed":813581}'
  zigbee2mqtt:info 4/16/2019, 8:13:05 PM MQTT publish: topic 'zigbee2mqtt/cube_02', payload '{"action":""}'
```

lolorc, 2019-04-16:
another funny issue with a door window sensor, and I haven't touched the pair button, I was just playing with  a magnet. In the end the device is removed from the network

```
  serialport/stream binding.read finished +10ms                                                                                                                                                                    │
  cc-znp { sof: 254,                                                                                                                                                                                               │
  cc-znp   len: 27,                                                                                                                                                                                                │
  cc-znp   type: 'AREQ',                                                                                                                                                                                           │
  cc-znp   subsys: 'AF',                                                                                                                                                                                           │
  cc-znp   cmd: 'incomingMsg',                                                                                                                                                                                     │
  cc-znp   payload:                                                                                                                                                                                                │
  cc-znp    { groupid: 0,                                                                                                                                                                                          │
  cc-znp      clusterid: 6,                                                                                                                                                                                        │
  cc-znp      srcaddr: 33589,                                                                                                                                                                                      │
  cc-znp      srcendpoint: 1,                                                                                                                                                                                      │
  cc-znp      dstendpoint: 1,                                                                                                                                                                                      │
  cc-znp      wasbroadcast: 0,                                                                                                                                                                                     │
  cc-znp      linkquality: 135,                                                                                                                                                                                    │
  cc-znp      securityuse: 0,                                                                                                                                                                                      │
  cc-znp      timestamp: 7527883,                                                                                                                                                                                  ┤
  cc-znp      transseqnumber: 0,                                                                                                                                                                                   │
  cc-znp      len: 7,                                                                                                                                                                                              │
  cc-znp      data: <Buffer 18 12 0a 00 00 10 00> },                                                                                                                                                               │
  cc-znp   fcs: 49,                                                                                                                                                                                                │
  cc-znp   csum: 49 } +18ms                                                                                                                                                                                        │
  cc-znp { sof: 254,                                                                                                                                                                                               │
  cc-znp   len: 27,                                                                                                                                                                                                │
  cc-znp   type: 'AREQ',                                                                                                                                                                                           │
  cc-znp   subsys: 'AF',                                                                                                                                                                                           │ [... cut ...]
  zigbee2mqtt:warn 4/16/2019, 8:43:08 PM Message without device!                                                                                                                                                   │
  zigbee-shepherd:msgHdlr IND <-- ZDO:leaveInd +373ms                                                                                                                                                              │
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +4ms                                                                                                                                   │
```

lolorc, 2019-04-16:
looks like it happens very often :-(

```
  zigbee2mqtt:debug 4/16/2019, 8:22:47 PM Received zigbee message of type 'devLeaving' with data '"switch_square_01"' of endpoint undefined
  zigbee2mqtt:debug 4/16/2019, 8:42:06 PM Received zigbee message of type 'devLeaving' with data '"0xdoor_window_sensor_05"' of endpoint undefined
  zigbee2mqtt:debug 4/16/2019, 8:43:08 PM Received zigbee message of type 'devLeaving' with data '"0xdoor_window_sensor_04"' of endpoint undefined
  zigbee2mqtt:debug 4/16/2019, 8:50:59 PM Received zigbee message of type 'devLeaving' with data '"aqara_07"' of endpoint undefined
  zigbee2mqtt:debug 4/16/2019, 8:57:26 PM Received zigbee message of type 'devLeaving' with data '"aqara_05"' of endpoint undefined
  zigbee2mqtt:debug 4/16/2019, 9:32:16 PM Received zigbee message of type 'devLeaving' with data '"0xdoor_window_sensor_06"' of endpoint undefined
  zigbee2mqtt:debug 4/16/2019, 9:33:26 PM Received zigbee message of type 'devLeaving' with data '"0xdoor_window_sensor_07"' of endpoint undefined
  zigbee2mqtt:debug 4/16/2019, 9:50:50 PM Received zigbee message of type 'devLeaving' with data '"cube_01"' of endpoint undefined
  zigbee2mqtt:debug 4/16/2019, 9:56:36 PM Received zigbee message of type 'devLeaving' with data '"motion_lum_02"' of endpoint undefined
  zigbee2mqtt:debug 4/16/2019, 10:01:48 PM Received zigbee message of type 'devLeaving' with data '"switch_02"' of endpoint undefined
  zigbee2mqtt:debug 4/16/2019, 10:42:33 PM Received zigbee message of type 'devLeaving' with data '"door_window_sensor_03"' of endpoint undefined
  zigbee2mqtt:debug 4/16/2019, 10:57:39 PM Received zigbee message of type 'devLeaving' with data '"cube_03"' of endpoint undefined
  zigbee2mqtt:debug 4/16/2019, 11:02:43 PM Received zigbee message of type 'devLeaving' with data '"switch_01"' of endpoint undefined
```

Koenkk, 2019-04-17:
@lolorc the empty actions are probably some converter bug. I haven't seen the dev leaving yet (and I have many Xiaomi devices)

Mine crashed again this morning, but I was able to get it working again by only replugging the CC2652R (no restart of zigbee2mqtt). Trying to catch a `DEBUG=*` log now.

lolorc, 2019-04-17:
the leaveind issues might be related to https://github.com/fairecasoimeme/ZiGate/issues/38#issuecomment-397001660

it wasn't happening with cc2530+firmware from end of 2018.

Koenkk, 2019-04-17:
@lolorc good catch, can you check if this still happens with `20190417` firmware?

lolorc, 2019-04-17:
I'll try tomorrow, I've just seen your message and it's a bit too late :-)
in the mean time I've repaired everything but the routers to see if it could be related to the routers.

lolorc, 2019-04-18:
Without any router I haven't lost a device during the night. The 6 routers (5 osram plug + 1 cc2531) and some more important devices are paired to my old cc2530.
Will wait for my cc2652 to crash or for some devices to disappear before upgrading the firmware.

lolorc, 2019-04-18:
grmbl, still no devices gone, still no crash.
I will reintroduce routers one by one to reproduce, then I will try your latest firmware to confirm it's a good workaround, gonna take ages.. ;-)

Koenkk, 2019-04-18:
https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/p/794106/2937749#2937749 that probably answers our stability problems. Will publish a new firmware tonight.

Koenkk, 2019-04-18:
Status update:
- New firmware is available (https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/Z-Stack_3.x.0)
- I've currently migrated my whole production network to the CC2652R
- Except the crashes every day, I've experienced no issues (hopefully fixed with latest firmware)
- When updating the firmware, I took a different CC2652R with a freshly flashed firmware, my whole network was recovered.

lolorc, 2019-04-18:
ace !
other hints here : https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/p/759868/2808276#2808276
a bit sad the issue is still not resolved since nov 2018.

antst, 2019-04-19:
Did someone tried already to solder external antenna connector on those boards?

Koenkk, 2019-04-19:
Crashed again, reduced device tables further, new firmware is up: `20190419`.

lolorc, 2019-04-20:
mine without any router hasn't crashed yet, had to restart z2m to pair more devices.... (still running 20190415)
will add routers by the end of the week end.

lolorc, 2019-04-21:
@dh-harald I was able to flash my cc2652 connected on my raspberry. I used usbip on both my rpi and my linux laptop, and used uniflash to flash. The programming was slower (using wifi) but it worked.
Don't know If I used the correct options, but I had to unplug/plug the cc2652 to have its serial port working again.

haven't had time to add routers with old firmware and wait for devLeaving messages from devices. I upgraded to latest 20190419 firmware.

lolorc, 2019-04-21:
```
  serialport/stream binding.read finished +0ms            
  cc-znp { sof: 254,                          
  cc-znp   len: 13,                     
  cc-znp   type: 'AREQ',                          
  cc-znp   subsys: 'ZDO',                                                                                                                                                                                             cc-znp   cmd: 'leaveInd',                                                                                                                                                                                           cc-znp   payload:                                        
  cc-znp    { srcaddr: 62610,                                                                                                                                                             
  cc-znp      extaddr: '0xaqara_03',                                                                                                                                     
  cc-znp      request: 0,                                                                                                                                                                                             cc-znp      removechildren: 0,                                                                                                                                              
  cc-znp      rejoin: 1 },                                                                                                                                                                                            cc-znp   fcs: 166,                                                                                                                                                                                                  cc-znp   csum: 166 } +520ms 
  serialport/stream _read reading +6ms                            
  serialport/binding-abstract read +6ms                                          
  serialport/bindings/unixRead Starting read +6ms        
  cc-znp:AREQ <-- ZDO:leaveInd, { srcaddr: 62610, extaddr: '0xaqara_03', request: 0, removechildren: 0, rejoin: 1 } +518ms                                                                                            zigbee-shepherd Device: 0xaqara_03 leave the network. +35s                                                                                                                                                          zigbee2mqtt:debug 4/21/2019, 8:27:04 PM Received zigbee message of type 'devLeaving' with data '"0xaqara_03"' of endpoint undefined                                                                                 zigbee2mqtt:warn 4/21/2019, 8:27:04 PM Message without device!                                                                                                                                                      zigbee-shepherd:msgHdlr IND <-- ZDO:leaveInd +509ms                                                                                                                                                                 serialport/bindings/unixRead waiting for readable because of code: EAGAIN +5ms
```

it just happened, the only router connected is a cc2531 (cc2531_1.2.2a.44539_firmware)
in the mean time this device was working but shown as unconnected on the network map, as all my devices after the usbip tests and upgrade. I've removed the router, it might not be related to the router. will see quite quickly

lolorc, 2019-04-21:
just happened without a router, it's probably an issue after the upgrade. my devices probably want to rejoin the network after the usbip/upgrade break. (rejoin=1)
I guess I'll have to repair everything again, I'm a bit lost :)

Koenkk, 2019-04-22:
Did you set `permit_join: true` after flashing the device? (maybe this is needed for the first start?). `coordinator_backup.json` should contain a list of devices, but this is hidden in the `NIB`.

lolorc, 2019-04-22:
every time the cc2652 crashes or I restart it I loose the pairing with most of my devices.
I've just repaired 10ish devices and the coordinator crashed few minutes after. I had to unplugged it, z2m did not re-open the serial port. Once restarted, all devices are show as unconnected in the network map. In the next hours all my devices will send a ZDO:leaveInd with rejoin=1, rejoin=1 is not handled (https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/event_handlers.js#L341)

I thought z2m was supposed to maintain the paired device list in the cc2652.

Koenkk, 2019-04-23:
If I understand right, when you stop zigbee2mqtt, replug CC2652R, start zigbee2mqtt, you lose pairings?

lolorc, 2019-04-23:
yep.
I will try to reproduce again this evening with only one aqara device and a sniffer recording the traffic.

Koenkk, 2019-04-23:
New firmware is up, should improve stability and I've managed to completely disable the child ageing mechanism.

lolorc, 2019-04-23:
will try it right now, thanks !

lolorc, 2019-04-24:
I've paired 1 aqara device, stopped z2m, replug cc2562R, the aqara device is still there. it hasn't left the network after 12hours, but the network map shows it disconnected from the coord. So I guess the disabled child ageing is working but it's probably just a workaround.

Koenkk, 2019-04-24:
Is it an end device (battery powered)?

lolorc, 2019-04-24:
yes, a MCCGQ11LM

Koenkk, 2019-04-24:
Was it awake when you did the network scan? Otherwise I expect it to be disconnected.

lolorc, 2019-04-24:
My battery powered aqara devices have always been shown as connected whenever they were awake or not. The behaviour changes after replugging the coordinator.

antst, 2019-04-24:
Mine almost always disconnected.

I didn't got down to the logic, but it seems that after every replug my coordinator (2530) draws map in one of 2 distinctively different ways. And on one of them battery-powered switches are always disconnected, but fully functional.

Koenkk, 2019-04-24:
@lolorc based on my experience and @antst comment, I don't think this is an issue of the CC2652R, which means until now everything is working fine for both of us.

antst, 2019-04-24:
So, 2652 is stable now?

Koenkk, 2019-04-24:
@antst not yet, but will we know this in the coming days.

antst, 2019-04-24:
Tomorrow I will receive one :)

tunip, 2019-04-24:
Received the CC2652R today. Will report the next days how it runs with my spare devices.

brtwrs, 2019-04-24:
Nice work @Koenkk! Did you test the CC2652R with lange networks? (50+ devices)

[855 more replies]

closed 2020-12-11 by Koenkk