Koenkk/zigbee2mqtt #1430: Sniffer firmware for CC2531 flash error (Alternative flashing methods)

antti365, 2019-04-14:
Hi!
I am trying to flash the zigbee stick CC2531 with an alternative method of flashing through raspberry pi. 
[https://github.com/Koenkk/zigbee2mqtt.io/blob/master/information/alternative_flashing_methods.md](url)
Firmware `CC2531ZNP-Prod.hex` is flash without problems and errors, everything is fine. But the firmware for the sniffer with the firmware gives the following error:
`pi@rasp:/flash_cc2531 $ ./cc_erase
  ID = b524.
  erase result = 00a2.
pi@rasp:/flash_cc2531 $ ./cc_write zboss_sniffer.hex
  ID = b524.
  reading line 540.
  file loaded (544 lines read).
writing page   5/  5.
verifying page   5/  5.
error at 0x3, 0xff instead of 0x7f

 Errors found in 1 pages.`
How critical is this error and is it possible to flash without this error?

antti365, 2019-04-19:
@Koenkk How to contact the authors of the article above to solve my question?

Koenkk, 2019-04-19:
@jmichault can you help?

jmichault, 2019-04-24:
Hello, i fixed a bug in reading hex files.
This should work now.

caisyew, 2019-04-25:
Hello !
I have the same problem with PI 3+b

pi@hassbian:~/flash_cc2531 $ ./cc_write CC2531ZNP-Prod.hex 
  ID = b524.
  reading line 15490.
  file loaded (15497 lines read).
writing page 128/128.
verifying page   1/128.
error at 0x0, 0x0 instead of 0x2
verifying page 128/128.
 Errors found in 1 pages.
pi@hassbian:~/flash_cc2531 $ ./cc_erase
  ID = b524.
  erase result = 00a2.
pi@hassbian:~/flash_cc2531 $ ./cc_write CC2531ZNP-Prod.hex 
  ID = b524.
  reading line 15490.
  file loaded (15497 lines read).
writing page 128/128.
verifying page   1/128.
error at 0x0, 0x0 instead of 0x2
verifying page 128/128.
 Errors found in 1 pages.

jmichault, 2019-04-25:
Hello, i have made a workaround so that both tested files works.

caisyew, 2019-04-25:
Thanks!!! solved

closed 2019-04-26 by Koenkk