Koenkk/zigbee2mqtt #1431: Aqara devices lost connection to CC2530 coordinator

RIMperator, 2019-04-14:
I have approx. 3 Aqara devices (1 windows sensor, 1 motion sensor and 1 temperature sensor)  that have lost connection to my CC2530 plugged into a rpi 3B. I have quite a few window sensors right next to each other and all except for 1 are working fine. I tried short and long press (repairing 5s and reset >10s), removed and reinserted battery however nothing seems to get the devices back working. Any suggestions what I could do to get these devices back in working mode? Thanks!

zigbee2mqtt version: 0.2.0 (commit #b0d3c2f)

Koenkk, 2019-04-15:
First I would recommend updating to the latest zigbee2mqtt version and firmware.

RIMperator, 2019-04-15:
Ok, I thought that had already happened, but in this case I will retry.

RIMperator, 2019-04-15:
Looks like I had an error: "Please, commit your changes or stash them before you can merge." As I'm not experienced with Linux, but believe this is a common error I will search for a solution.

RIMperator, 2019-04-15:
OK, update of zigbee2mqtt and node.js succeeded. Now running zigbee2mqtt version 1.3.0 (commit #00ebd44) and node v10.15.3. However I see no difference regarding the two remaining devices (the temperature sensor came back online by pressing the button once already before the update). I've now removed the window sensor using mqtt publish and after a force remove it was confirmed that the device is removed. However repairing doesn't work and it appears that also no new devices can be added (currently 13 devices are joined according to log entry in the start sequence). Could this somehow be related to previously have already hit the limit of 15 devices? It appears that since then I can't add any devices anymore at all.

Koenkk, 2019-04-15:
@RIMperator no, you can re-flash without losing pairings.

RIMperator, 2019-04-15:
@Koenkk Thanks for your fast feedback! I will flash with CC2530_DEFAULT_20190223.zip and will post back if it makes any difference

RIMperator, 2019-04-15:
I've now flashed the CC2530 with CC2530_DEFAULT_20190223.zip, however when restarting zigbee2mqtt I get the following error: "Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)". Is there a procedure how I can verify if flashing was successful? The SmartRF Flash Programmer says "CC2531 - ID6121: Erase, program and verify OK".

RIMperator, 2019-04-15:
> I've now flashed the CC2530 with CC2530_DEFAULT_20190223.zip, however when restarting zigbee2mqtt I get the following error: "Error while starting zigbee-shepherd, attempting to fix... (takes 60 [...]

Reflashing with CC2531_DEFAULT_20190223.zip worked: "zigbee2mqtt:info 4/15/2019, 6:47:48 PM Coordinator firmware version: '20190223'"

Now I was able to pair a new Aqara window contact as well as rejoin the Aqara motion sensor that lost its connection. Tomorrow I will try to rejoin the Aqara window contact that lost its connection too as well as add a Hue Aura that I would like to use as a router (fingers crossed that it works, since it's not on the supported device list yet).

RIMperator, 2019-04-27:
Adding the LivingColors Aura worked and is now acting as a router allowing me to add additional devices. I‘m at 21 devices now. As the Aura was not in the supported list I had to add it. Since it‘s a first for me, can anyone help me how to make the pull request so it becomes available for everyone?

Koenkk, 2019-04-28:
@RIMperator you can either follow https://help.github.com/en/articles/creating-a-pull-request or post the section you added to `devices.js` here.

RIMperator, 2019-05-29:
@Koenkk Apologies for the late post, however here is the section I've added to devices.js:

    {
        zigbeeModel: ['LLC014'],
        model: '7099860PH',
        vendor: 'Philips',
        description: 'LivingColors Aura',
        extend: hue.light_onoff_brightness_colorxy,
    },

Koenkk, 2019-05-29:
@RIMperator device is supported in the dev branch know, can this be closed?

RIMperator, 2019-05-29:
Yes, many thanks!

closed 2019-05-29 by RIMperator