Koenkk/zigbee2mqtt #1434: Is rename feature broken?

EmmanuelLM, 2019-04-15:
Using version 1.3.0, when I try to rename an entity through mqtt with...

{"topic":"zigbee2mqtt/bridge/config/rename","payload":"{old: '0x00158d0002ede2ab', new: 'Utility light switch'}"}

...I get the following error:

  zigbee2mqtt:error 4/15/2019, 5:45:11 PM Invalid rename message format expected {old: 'friendly_name', new: 'new_name} got {old: '0x00158d0002ede2ab', new: 'Utility light switch'}

I'm probably missing something simple...

BudBundi, 2019-04-16:
Maybe  the spaces. What happen when you use underscores?

Koenkk, 2019-04-16:
Should be `{"old": '0x00158d0002ede2ab', "new": 'Utility light switch'}` (quotes are missing)

tb-killa, 2019-04-16:
@Koenkk we should integrate something like slugify convert as it would do in web development to prevent strange results.
I prever this as it build standard naming. 
What do you think?

Koenkk, 2019-04-17:
@tb-killa sounds good!

EmmanuelLM, 2019-04-17:
> Should be `{"old": '0x00158d0002ede2ab', "new": 'Utility light switch'}` (quotes are missing)

Sadly, this is invalid JSON format in the MQTT publish service I am using under home-assistant.... (i.e. it seems quotes are not within the payload).

tb-killa, 2019-04-18:
> @tb-killa sounds good!

@Koenkk maybe this would help ?
```
return string.toLowerCase()
  .replace(/[^\w\s-]/g, '') // remove non-word [a-z0-9_], non-whitespace, non-hyphen characters
  .replace(/[\s_-]+/g, '-') // swap any length of whitespace, underscore, hyphen characters with a single -
  .replace(/^-+|-+$/g, ''); // remove leading, trailing -
```

ShonkyCH, 2019-04-18:
> > Should be `{"old": '0x00158d0002ede2ab', "new": 'Utility light switch'}` (quotes are missing)
> 
> Sadly, this is invalid JSON format in the MQTT publish service I am using under home-assistant.... (i.e. it seems quotes are not allowed within the payload).

Escape them seems to work:
{
"topic":"zigbee2mqtt/bridge/config/rename",
"payload":"{\"old\": \"0x00158d0002ede2ab\", \"new\": \"Utility light switch\"}"
}

Koenkk, 2019-04-18:
@tb-killa yes, can you make a PR and introduce extra tests for this?

tb-killa, 2019-04-18:
@Koenkk yes

closed 2019-06-24 by stale[bot]

OmgImAlexis, 2020-09-19:
Any chance this could be reopened? I can't rename my devices as I just get an error and I'm guessing this is what's causing it as I'm using this with Home Assistant.

Koenkk, 2020-09-19:
@OmgImAlexis please provide the debug logging of this.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```