Koenkk/zigbee2mqtt #1435: delayed or wrong messages with Hue Motion sensor

vogler, 2019-04-16:
# Bug Report
## What happened
Just recently my Hue motion sensor started sending wrong messages with `occupancy:true` and messing up my light automation.
The first message below is correct, the last message is wrong (or delayed) since I wasn't in the bathroom anymore. I don't know if it issued these `genPowerCfg` warnings before, but they don't happen before every falsely sent message.

~~~
4/16/2019, 7:49:54 AM - info: MQTT publish: topic 'zigbee2mqtt/Hue_Motion_Bad', payload '{"battery":100,"linkquality":       26,"occupancy":true,"temperature":20.94,"illuminance":23530}'
...
4/16/2019, 7:54:53 AM - warn: No converter available for '9290012607' with cid 'genPowerCfg', type 'devChange' and data      '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":106}}'
4/16/2019, 7:54:53 AM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
4/16/2019, 7:56:39 AM - info: MQTT publish: topic 'zigbee2mqtt/Hue_Motion_Bad', payload '{"battery":100,"linkquality":       10,"occupancy":true,"temperature":20.94,"illuminance":0}'
~~~

Possible reasons are that I pulled the newest commits (before it was running 32db1fd07a46c26e820fb4d93038830641473e85), or that I added two new Aqara Door sensors. I now have 11 devices paired.
I assume my sensor didn't just start sending wrong information all of a sudden, but that the messages are delayed. Also because it doesn't happen randomly but only some time after a correctly send message.

## How to reproduce it (minimal and precise)
Trigger motion. It will still send a `occupancy: true` message a couple of minutes afterwards.

## Debug Info
zigbee2mqtt version: 1.3.0 dev branch (310f9892d6a8562bd648a4a554dc6d21e74f2003)
CC253X firmware version: not sure, flashed 19.12.2018

vogler, 2019-04-16:
Tried it again with a different sensor, after setting `{"occupancy_timeout": 11}`. Why is it sending so many messages? It wasn't like that before.
~~~
  zigbee2mqtt:info 4/16/2019, 6:44:43 PM MQTT publish: topic 'zigbee2mqtt/Hue_Motion_WZ', payload '{"occupancy":false,"linkquality":52,"illuminance":16738,"temperature":19.44,"battery":53}'
  zigbee2mqtt:info 4/16/2019, 6:44:45 PM MQTT publish: topic 'zigbee2mqtt/Hue_Motion_WZ', payload '{"occupancy":false,"linkquality":52,"illuminance":16738,"temperature":19.44,"battery":53}'
  zigbee2mqtt:info 4/16/2019, 6:45:09 PM MQTT publish: topic 'zigbee2mqtt/Hue_Motion_WZ', payload '{"occupancy":true,"linkquality":47,"illuminance":16656,"temperature":19.44,"battery":53}'
  zigbee2mqtt:info 4/16/2019, 6:45:13 PM MQTT publish: topic 'zigbee2mqtt/Hue_Motion_WZ', payload '{"occupancy":true,"linkquality":44,"illuminance":16645,"temperature":19.44,"battery":53}'
  zigbee2mqtt:info 4/16/2019, 6:45:18 PM MQTT publish: topic 'zigbee2mqtt/Hue_Motion_WZ', payload '{"occupancy":true,"linkquality":47,"illuminance":16666,"temperature":19.44,"battery":53}'
  zigbee2mqtt:info 4/16/2019, 6:45:23 PM MQTT publish: topic 'zigbee2mqtt/Hue_Motion_WZ', payload '{"occupancy":true,"linkquality":47,"illuminance":16645,"temperature":19.44,"battery":53}'
  zigbee2mqtt:info 4/16/2019, 6:45:28 PM MQTT publish: topic 'zigbee2mqtt/Hue_Motion_WZ', payload '{"occupancy":true,"linkquality":44,"illuminance":16656,"temperature":19.44,"battery":53}'
  zigbee2mqtt:info 4/16/2019, 6:45:33 PM MQTT publish: topic 'zigbee2mqtt/Hue_Motion_WZ', payload '{"occupancy":true,"linkquality":49,"illuminance":16625,"temperature":19.44,"battery":53}'
  zigbee2mqtt:info 4/16/2019, 6:45:38 PM MQTT publish: topic 'zigbee2mqtt/Hue_Motion_WZ', payload '{"occupancy":true,"linkquality":52,"illuminance":16614,"temperature":19.44,"battery":53}'
  zigbee2mqtt:info 4/16/2019, 6:45:44 PM MQTT publish: topic 'zigbee2mqtt/Hue_Motion_WZ', payload '{"occupancy":false,"linkquality":52,"illuminance":16614,"temperature":19.44,"battery":53}'
  zigbee2mqtt:info 4/16/2019, 6:45:46 PM MQTT publish: topic 'zigbee2mqtt/Hue_Motion_WZ', payload '{"occupancy":false,"linkquality":52,"illuminance":16614,"temperature":19.44,"battery":53}'
~~~

vogler, 2019-04-16:
Just to make sure I'm interpreting the docs correctly:
- `occupancy_timeout` in `configuration.yaml` is not supported (and therefore ignored) for Hue motion sensors? (It only mentions `RTCGQ01LM, RTCGQ11LM and AV2010/22`, whereas the Hue motion sensor is `SML001`.)
- If I want to change `occupancy_timeout`, I have to publish to `zigbee2mqtt/[DEVICE_ID]/set`? (Whereas for groups and binding `[friendly_name]` instead of `[DEVICE_ID]` is used.)
  - (There is no way to see that the change was successful or query the current state? (At least ` mosquitto_pub -t zigbee2mqtt/0x0017880103283554/get -m '{"occupancy_timeout": ""}'` doesn't do anything.))
  - Hue motion sensors can't be set up to not send a `"occupancy":false` message (`0 - 10: 10sec (min possible timeout)`), whereas the other motion sensors above can (`When set to 0 no occupancy: false is send.`).

Koenkk, 2019-04-18:
- occupancy_timeout cannot be used for SML001 (see point below)
- this is indeed true, as the SLM001 produces this message itself (whereas the Xiaomi sensors don't and we have to produce this synthetical message), to verify if it has been set correctly, just look at the logs, if it doesn't contain any error for that message send, it's OK.
- to further debug your issue, please enable debug logging (`log_level: debug`) (http://www.zigbee2mqtt.io/configuration/configuration.html)

vogler, 2019-04-18:
I pulled the newest commit, so far the problem hasn't appeared yet. Will upload a full log if it happens again.
However, I thought that some time ago the sensor didn't send that many messages but only one `occupancy:true` when movement was detected and you kept moving inside the `occupancy_timeout`.
Now it sends messages every 2-3s when moving (see below).
What does the red light on the sensor indicate? Does it only come on when motion was detected after the internal timeout?

~~~
❯ ag 'Motion_Bad.*occupancy":true' data/log/2019-04-18.19-41-54/log.txt
126:4/18/2019, 7:42:58 PM - info: MQTT publish: topic 'zigbee2mqtt/Hue_Motion_Bad', payload '{"battery":100,"linkquality":5,"occupancy":true,"temperature":21.51,"illuminance":0}'
129:4/18/2019, 7:42:58 PM - info: MQTT publish: topic 'zigbee2mqtt/Hue_Motion_Bad', payload '{"battery":100,"linkquality":2,"occupancy":true,"temperature":21.51,"illuminance":4573}'
138:4/18/2019, 7:43:00 PM - info: MQTT publish: topic 'zigbee2mqtt/Hue_Motion_Bad', payload '{"battery":100,"linkquality":2,"occupancy":true,"temperature":21.51,"illuminance":4573}'
154:4/18/2019, 7:43:03 PM - info: MQTT publish: topic 'zigbee2mqtt/Hue_Motion_Bad', payload '{"battery":100,"linkquality":7,"occupancy":true,"temperature":21.51,"illuminance":23283}'
181:4/18/2019, 7:43:23 PM - info: MQTT publish: topic 'zigbee2mqtt/Hue_Motion_Bad', payload '{"battery":100,"linkquality":10,"occupancy":true,"temperature":21.51,"illuminance":23306}'
184:4/18/2019, 7:43:28 PM - info: MQTT publish: topic 'zigbee2mqtt/Hue_Motion_Bad', payload '{"battery":100,"linkquality":15,"occupancy":true,"temperature":21.51,"illuminance":23306}'
186:4/18/2019, 7:43:28 PM - info: MQTT publish: topic 'zigbee2mqtt/Hue_Motion_Bad', payload '{"battery":100,"linkquality":15,"occupancy":true,"temperature":21.51,"illuminance":23227}'
191:4/18/2019, 7:43:29 PM - info: MQTT publish: topic 'zigbee2mqtt/Hue_Motion_Bad', payload '{"battery":100,"linkquality":18,"occupancy":true,"temperature":21.51,"illuminance":23227}'
194:4/18/2019, 7:43:32 PM - info: MQTT publish: topic 'zigbee2mqtt/Hue_Motion_Bad', payload '{"battery":100,"linkquality":15,"occupancy":true,"temperature":21.51,"illuminance":23227}'
196:4/18/2019, 7:43:33 PM - info: MQTT publish: topic 'zigbee2mqtt/Hue_Motion_Bad', payload '{"battery":100,"linkquality":15,"occupancy":true,"temperature":21.51,"illuminance":23240}'
206:4/18/2019, 7:43:38 PM - info: MQTT publish: topic 'zigbee2mqtt/Hue_Motion_Bad', payload '{"battery":100,"linkquality":18,"occupancy":true,"temperature":21.51,"illuminance":23270}'
218:4/18/2019, 7:43:42 PM - info: MQTT publish: topic 'zigbee2mqtt/Hue_Motion_Bad', payload '{"battery":100,"linkquality":15,"occupancy":true,"temperature":21.51,"illuminance":23274}'
229:4/18/2019, 7:43:44 PM - info: MQTT publish: topic 'zigbee2mqtt/Hue_Motion_Bad', payload '{"battery":100,"linkquality":10,"occupancy":true,"temperature":21.51,"illuminance":23234}'
232:4/18/2019, 7:43:44 PM - info: MQTT publish: topic 'zigbee2mqtt/Hue_Motion_Bad', payload '{"battery":100,"linkquality":7,"occupancy":true,"temperature":21.51,"illuminance":23238}'
237:4/18/2019, 7:43:45 PM - info: MQTT publish: topic 'zigbee2mqtt/Hue_Motion_Bad', payload '{"battery":100,"linkquality":15,"occupancy":true,"temperature":21.51,"illuminance":23238}'
239:4/18/2019, 7:43:49 PM - info: MQTT publish: topic 'zigbee2mqtt/Hue_Motion_Bad', payload '{"battery":100,"linkquality":13,"occupancy":true,"temperature":21.51,"illuminance":23265}'
269:4/18/2019, 7:44:32 PM - info: MQTT publish: topic 'zigbee2mqtt/Hue_Motion_Bad', payload '{"battery":100,"linkquality":5,"occupancy":true,"temperature":21.51,"illuminance":23005}'
273:4/18/2019, 7:44:33 PM - info: MQTT publish: topic 'zigbee2mqtt/Hue_Motion_Bad', payload '{"battery":100,"linkquality":15,"occupancy":true,"temperature":21.51,"illuminance":23005}'
276:4/18/2019, 7:44:34 PM - info: MQTT publish: topic 'zigbee2mqtt/Hue_Motion_Bad', payload '{"battery":100,"linkquality":13,"occupancy":true,"temperature":21.51,"illuminance":23005}'
~~~

vogler, 2019-04-18:
Mh, it's still happening...
If you search here for `10:06`: https://termbin.com/gbc1
That message was falsely sent ~5min after the last correct `occupancy:true` message.
It always seems to be a little more than 5 minutes.

Koenkk, 2019-04-19:
On `10:06:27` the sensor reports it's temperature state, this causes all properties of the device to be re-send (so also the `occupancy: true` that was retrieved from a zigbee message on `9:53:29`). 

On `10:06:40` a occupancy false message is received from the, all succeeding message contain `occupancy: false` (until true is reported again).

I don't see any unexpected behavior yet.

vogler, 2019-04-19:
Ok, so then the problem is that the `occupancy: 0` zigbee message is no longer sent (or received) correctly.
If I search for `occupancy":(0|1).*8d8bc`:
~~~
 9:52:07 ... "occupancy":1 ...
 9:52:17 ... "occupancy":0 ... // as it should be (10s later)
 9:53:29 ... "occupancy":1 ...
10:06:40 ... "occupancy":0 ... // 13min too late
~~~

Koenkk, 2019-04-19:
To verify that, let's check the sniffing (http://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

vogler, 2019-04-19:
I have the CC2531 stick connected to my rpi as the coordinator and a currently unused CC2530.
How should I do it? Can I use the CC2530 via USB (CP2102) on a MacBook or on Windows for sniffing?

vogler, 2019-04-19:
CC2530 doesn't seem to work 😞 
I tried with the `ZBOSS` firmwares `SmartRf05EB` and `CC2531` on Windows and with `whsniff` using `sniffer_fw_cc2530.hex` on macOS.

Koenkk, 2019-04-19:
Could you buy another CC2531?

vogler, 2019-04-19:
I don't really want to if I don't need it for anything else. Also, shipping would take weeks.
I could flash the CC2530 as a coordinator and sniff with the CC2531 if the problem persists, right? However I don't like the idea of having to repair everything. Can I copy the paired devices between the two?
Can we rule out some other things?
- It reports `"battery":100` - I guess changing the battery wouldn't help, unless this value is not reliable.
- `linkquality` ranges from `0` to `42` - I will try to pair another (newer) Tradfri lamp tomorrow and put it inbetween. I already have one inbetween but it sometimes loses connection. I also tried with this lamp disconnected, didn't have any influence on the problem.

Koenkk, 2019-04-19:
You don't need to re-pair device when switching from coordinator (http://www.zigbee2mqtt.io/information/what_does_and_doesnt_require_repairing.html)

vogler, 2019-05-15:
Aqara Door sensors still worked after the change to CC2530, the rest didn't; also the networkmap showed no connections anymore. Repaired everything.
Now using `CC2530_20190425` and stopped using homeassisant. Everything seems to be fine again - no more spammed and delayed messages from the Hue motion sensors.

closed 2019-05-15 by vogler