Koenkk/zigbee2mqtt #1437: CC2530 + CC2592 = Very long range solution

kirovilya, 2019-04-16:
[image]

The guys from our community tried to use TI cc2530 modules from EBYTE with a new cc2592 amplifier with a power of 100mW [E18-MS1PA1-PCB](https://www.aliexpress.com/item/CDEBYTE-2Pcs-Lot-Smart-Home-Automation-Module-E18-MS1PA1-PCB-2-4GHz-CC2530-Wireless-Zigbee-Module/32803068018.html). For this module, we had to build [a special Z-Stack ZNP firmware](https://github.com/kirovilya/files/blob/master/CC2530_CC2592_ZNP.hex), since there was no firmware for such an amplifier. And the results exceeded expectations!

The tests with the “round button” from Xiaomi showed that the coordinator get the signals even at a distance of 400m of direct visibility. In the house, the button works through 2 reinforced concrete walls. Those the amplifier greatly increases the range and sensitivity.

Another indicative criterion is the parameters of the link_quality signals from devices (they can be seen, for example, on the network map in iobroker.zigbee or zigbee2mqtt). For this module, link_quality shows 36–40 and good work, where the coordinator cc2531 showed 2–5 and there were interruptions in communication (3 foam-block walls, one thin wooden wall).

There is a version of this module with an external antenna connection. And there is also a 500mW [E18-2G4Z27SI](https://www.aliexpress.com/item/E18-2G4Z27SI-Zigbee-CC2530-PA-CC2592-2-4Ghz-500mW-IoT-uhf-Wireless-Transceiver-27dBm-2-4/32965861242.html) module, but we didn’t feel any significant difference in comparison with the 100mW module (or we couldn’t correct start it in the firmware yet), the current consumption of the 500mW module coincided with the 100mW module.

On the basis of these modules, the craftsmen made themselves coordinators and routers to cover with the network not only a residential building, but also the area around it. 

[image]
[image]

The user Jager share [his stick circuit](https://www.dropbox.com/s/vqok0rphdwvwbf3/USB_dongle_zigbee.rar?dl=0) in [his blog](http://modkam.ru/?p=1017).

[image]

[image]

ps. sorry for my English :)

lolorc, 2019-04-16:
sounds nice !
have you tried the ipx version E18-MS1PA1-IPX ?

feel free to submit a PR for cc2530+cc2592 support :-)

Jager-f, 2019-04-16:
> have you tried the ipx version E18-MS1PA1-IPX ?
This module [works](http://modkam.ru/?p=1009)

lolorc, 2019-04-16:
i guess it works, just wondering if link quality comparaisons have been done with this module as well.

kirovilya, 2019-04-16:
@lolorc 

> feel free to submit a PR for cc2530+cc2592 support :-)

yes. late

Jager-f, 2019-04-16:
> i guess it works, just wondering if link quality comparaisons have been done with this module as well.
Real tests have not yet done. The manufacturer claims this, but the data are clearly overestimated.
[image]

lolorc, 2019-04-16:
> @lolorc
> 
> > feel free to submit a PR for cc2530+cc2592 support :-)
> 
> yes. later

I hope so, because I just ordered some ;-)

lolorc, 2019-04-17:
sure, we've been asking about info about the router code so we can build it ourselves, but the question has been left unanswered...
https://github.com/Koenkk/Z-Stack-firmware/issues/50
https://github.com/Koenkk/Z-Stack-firmware/issues/21
https://github.com/Koenkk/zigbee2mqtt/issues/165#issuecomment-401628955
https://github.com/Koenkk/zigbee2mqtt/issues/494
...
;-)

kirovilya, 2019-04-17:
@wunderbaum I do not have the source code for the router. Therefore, we checked on the module with 500mW - firmware of router with cc2591 came up there

lolorc, 2019-04-17:
the firmwares for cc2591 should probably work on cc2592, they probably lack changes to increase the power output.

kirovilya, 2019-04-17:
sorry...

@lolorc did not work for cc2592. I had to do a bit in the source code

wunderbaum, 2019-04-17:
Maybe someone can contact ptvoinfo for providing a router firmware for this special device.

MiloshCZ, 2019-04-17:
@kirovilya Nice work! Have you tried it with IKEA bulbs? IKEA bulbs are, ee, hmm, "quite strange" :-D

antst, 2019-04-19:
What is the maximum number of devices with this firmware?

kirovilya, 2019-04-19:
@antst I assembled the firmware based on the ZNP - there are 15 direct connect devices by default. a little later I make the PR for firmware in the Z-StacK-firmware.

Are you russian? find me in telegram :)

kirovilya, 2019-04-19:
@MiloshCZ not test yet

battuashwik, 2019-04-19:
@kirovilya , Hi, I have the exact same hardware, I am having issues with range which i suspect is to do with my amplifier CC2592 settings in Zstack 3.0 ZNP. I have tested with this https://github.com/kirovilya/files/blob/master/CC2530_CC2592_ZNP.hex and having no issues at all with the range but i have some custom settings enabled in ZNP which i need for my use case.

Also my code is working perfectly fine for me on CC2530 + RFX2401c (HAL_PA_LNA setting for amplifier) .Can you please help with this. Its very urgent for me. It will be great if you could share the Zstack code patch that you have applied for ZNP for getting cc2592 amplifier working.

kirovilya, 2019-04-20:
@battuashwik  Very simple: you must specify directive 
`HAL_PA_LNA_CC2592`
instead of directive 
`HAL_PA_LNA`
Try it. Perhaps both should be used. I'm still testing, too, maybe something else needs to be changed.

battuashwik, 2019-04-20:
Hey, tried that already, not working. I tried applying patches mentioned in https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/p/758643/2802562#2802562 , still no luck, was wondering if you have done something else to get it working.

kirovilya, 2019-04-20:
@battuashwik look at the proposed patch https://github.com/Koenkk/Z-Stack-firmware/issues/71, I think the main changes for you will be in the hal_board_cfg.h file

battuashwik, 2019-04-21:
> @battuashwik look at the proposed patch [Koenkk/Z-Stack-firmware#71](https://github.com/Koenkk/Z-Stack-firmware/issues/71), I think the main changes for you will be in the hal_board_cfg.h file

Hey thanks, got it working and works much better than my rfx2401c amplifier.

kirovilya, 2019-04-24:
Hey @ptvoinfo ! Can you build "router-firmware" with cc2530+cc2592 ?

wunderbaum, 2019-04-29:
Whow, my order came quick! I already have the small item.
Hopefully @ptvoinfo makes us a routing firmware.

I keep my fingers crossed and owe you a beer ;)

ptvoinfo, 2019-04-29:
The firmware is here:
https://ptvo.info/wp-content/uploads/2018/09/cc2530_cc2591_1.2.2a.44539_firmware.zip
https://ptvo.info/wp-content/uploads/2018/09/cc2530_cc2592_1.2.2a.44539_firmware.zip

Ormerod2, 2019-04-29:
I am using this Chip now for quite a while but I made a UART Version out of it since I don’t line USB dongles. 
https://drive.google.com/file/d/1YfYTegq1n7-A1CzsvAAA9TlK87XtGEHA/view?usp=drivesdk

Currently I am running the KronKk Firmware but I will try kirovilya‘s tonight.

drbios, 2019-05-09:
Hello guys ...2 question ...
To program this little friend ... it's the same pin connection  that  is used for cc2530? 

Where is the appropriate coordinator and router for this lovely units?

Best regards from Chile 👌

wunderbaum, 2019-05-10:
The firmware is just 2 steps above:
https://github.com/Koenkk/zigbee2mqtt/issues/1437#issuecomment-487658726

For programming use the pinout from the flasher here:
Chapter 2, cc debugger pinout and connect the P2.1, P2.2 etc to the according ones on the CC2592-module as described here:
https://www.aliexpress.com/item/CDEBYTE-2Pcs-Lot-Smart-Home-Automation-Module-E18-MS1PA1-PCB-2-4GHz-CC2530-Wireless-Zigbee-Module/32803068018.html

drbios, 2019-05-10:
Thanks for your help my friend ...today I will start my experimentation 👌👌

drbios, 2019-05-13:
Hello again ....
Any one knows some kind of breakout board for this units?

drbios, 2019-05-13:
I have doubts about the connections...the cc2530 flashing procedure indicates use 2 vcc connections  between the module and the cc debuger...but according to the manual of these modules only have 1 vcc pin (number 2) ...where is the second vcc pin?

kirovilya, 2019-05-13:
@drbios you need to connect the 2nd and 9th contact between each other, then the ccdebugger will work fine
[image]

drbios, 2019-05-13:
@kirovilya doing that means provide 3.3v + 3.3v = 6.6v to the pin2 of this modules?

kirovilya, 2019-05-13:
@drbios why 6.6v? just connect the pins (blue line)
[image]

drbios, 2019-05-13:
Really sorry about 6.6v comment...I though that pin2 and pin9 of the debugger were 3.3 volt 🤦‍♂️
Now I will try again with this conection configuration.

 Thanks Thanks thanks 👌👌👌

drbios, 2019-05-13:
now it works !!
this is my "programer interfaces" XD

[image]

thanks

linusali, 2019-05-17:
Thanks a lot @kirovilya and @ptvoinfo for the solution. And thanks to Jager for the PCB and the parts list. It really works. I have a coordinator and a router for the time being and have connected close to 24 devices. All works well.  I have used the coordinator firmware from https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/bin/CC2530_CC2592_20190515.zip and router firmware from http://ptvo.info/wp-content/uploads/2018/09/cc2530_cc2592_1.2.2a.44539_firmware.zip 

I have a small issue though , with the current board and connections, I could not tell if the board is connected to a network or not. So is there a way I can add a LED to indicate this ?

Thanks in Advance. And once again great work.

antst, 2019-05-21:
I still have issue with stability of firmware.
“Naked” cc2530 works like charm in test setup with 20 devices.
With this fronted coordinator hangs after sometime, like it used to be with older 2530.
I assume that adding fronted somehow reduces available memory.
Btw, maxheap is different for naked 2530 and with frontend.

Is it possible to make firmware for 12 devices max or so? Even 10.
But not 5, as in max_stability

XRyu, 2019-06-23:
> I am using this Chip now for quite a while but I made a UART Version out of it since I don’t line USB dongles.
> https://drive.google.com/file/d/1YfYTegq1n7-A1CzsvAAA9TlK87XtGEHA/view?usp=drivesdk
> 
> Currently I am running the KronKk Firmware but I will try kirovilya‘s tonight.

@Ormerod2 could you provide the files for the UART version? i like this :)

drbios, 2019-06-24:
for me it was impossible to build... my soldering skills ain't good enough

halfcherry, 2019-08-11:
Will this work?
https://item.taobao.com/item.htm?spm=a1z10.5-c-s.w4002-16785486380.12.c60918aaYpf6dL&id=583299208437

httpedo13, 2020-05-01:
For this usb card with E18-2G4Z27SI which firmware is compatible? Is an antenna also needed?

https://it.aliexpress.com/item/4000834424643.html?spm=a2g0o.productlist.0.0.7fc37f39juEj2Y&algo_pvid=83a57da3-8ed6-423f-b8ce-422c3d4a8488&algo_expid=83a57da3-8ed6-423f-b8ce-422c3d4a8488-2&btsid=0ab6f83115883377035427337e52aa&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

lehungmt94, 2020-05-25:
Hello kirovilya,
You can tutorial me how to build ZNP firmware
This is my preprocessing

```
BDB_FINDING_BINDING_CAPABILITY_ENABLED=0
TC_LINKKEY_JOIN
HAL_UART_DMA_RX_MAX=128
MAXMEMHEAP=2765
NWK_MAX_DEVICE_LIST=10
ZDSECMGR_TC_DEVICE_MAX=30
DISABLE_GREENPOWER_BASIC_PROXY
POWER_SAVING
FEATURE_SYSTEM_STATS
MT_GP_CB_FUNC
ASSERT_RESET
MAKE_CRC_SHDW

```

it's not working

kirovilya, 2020-05-25:
@leanderseidlitz Use ready-made firmware from here https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin including for cc2592.
And here is the build instruction https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/COMPILE.md

lehungmt94, 2020-05-25:
> @leanderseidlitz Use ready-made firmware from here https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin including for cc2592.
> And here is the build instruction https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/COMPILE.md

Nice instruction, thanks @leanderseidlitz

codmpm, 2020-07-10:
Hey guys. Just wanted to let you know, that I've built a module based on the E18-MS1PA1-IPX for the GPIO header of the Pi: https://allgeek.de/2020/06/22/zigbee-modul-fuer-den-raspberrypi-cc2530-mit-homegear-oder-zigbee2mqtt/ (german)

It's open source but we also sell these. So hit me up, if you need anything.

linusali, 2020-07-10:
> Hey guys. Just wanted to let you know, that I've built a module based on the E18-MS1PA1-IPX for the GPIO header of the Pi: https://allgeek.de/2020/06/22/zigbee-modul-fuer-den-raspberrypi-cc2530-mit-homegear-oder-zigbee2mqtt/ (german)
> 
> [image]
> 
> It's open source but we also sell ready made modules. So hit me up, if you need anything.

Wow that's nice. How can I get one or two long range one? Does it come with the same firmware ?

codmpm, 2020-07-10:
They are sold in our online Shop, the link is in the blog article. We preflash the modules with Z-Stack Coordinator Firmware 1.2 oder 3.0.x.
Don't want to hijack the issue for marketing, so hit me up with an email at blog@allgeek.de.

XRyu, 2020-07-11:
@codmpm i sadly can't find the files that are open source :) I have some of the SMD-modules here, as i wanted to start something myself but never did :) is there a chance to get the platine and casedesign for the Long Range Router somewhere? :) love the sleek look!

codmpm, 2020-07-12:
Hey @XRyu, the case is here: https://www.thingiverse.com/thing:4192704
Did not manage to put the pcb files on github, yet.

Get in touch with me. Maybe I can get you some prototype PCB's where the LED of the router was not wired correctly. My email is in the comment before yours.

lily148, 2020-09-01:
Sorry, I am very new to this. Can I direct solder the E18-MS1PA1-IPX to Raspberry Pi uart without any PCB board?

codmpm, 2020-09-02:
@lily148 get in touch with me via blog@allgeek.de... I'll send you one pcb for the E18-MS1PA1-IPX.

fahimehorvatinia, 2020-09-06:
Hey, 
I was working on e18-ms1pa1-ipx and I want to know is  CC2530ZNP-Pro-Secure_LinkKeyJoin.hex for coordinator and router-cc2530-cc2591-diag.hex for router works fine together?

And I want to know that is it possible to connect this module directly to arduino (tx and rx) as coordinator or router?

lily148, 2020-09-11:
@fahimehorvatinia , I tested direct solder the pin to my Pi tx and rx, it works. No extra PCB is needed but you need to have better soldering skill. Otherwise, @codmpm 's PCB module might be better choice.

fahimehorvatinia, 2020-09-11:
@lily148 , 
Hi, 
I connected this module directly to Computer via (usb to ttl) module, but i couldn't set up communication . 
how did you connect it directly?
this is my pin connection. 
VCC->3.3
GND->GND
P0.3 ->TX 
P0.2 -> Rx
and this is the error that I got. 
[image]

lily148, 2020-09-11:
@fahimehorvatinia , what is did was direct connect the board to the Pi uart port. That's it.
By the way, had you already flash the firmware or download your program into the board via CC debugger?

fahimehorvatinia, 2020-09-11:
@lily148 I've download cc2530-cc2592.hex file to the module using cc debugger. 
[image]
And after that I connected e18-ms1pa1-ipx module to computer(using usb to ttl converter). 
Is there any software that I make sure connection is ok? 
I just used ebyte software configuration to get parameters and it failed.

lily148, 2020-09-12:
@fahimehorvatinia , I am not sure... because I am not using the ebyte software nor using any USB to ttl. I just direct connect to the uart pin on the Pi.

benjaminbear, 2021-01-11:
@fahimehorvatinia also tried to connect via ttl over usb with no success. Have you found a solution?

fahimehorvatinia, 2021-01-13:
> @fahimehorvatinia also tried to connect via ttl over usb with no success. Have you found a solution?

After many attempts, I finally succeeded. I will share useful content in this field with you.
[https://github.com/epcbtech/ArduinoZigBee](url)
[https://epcb.vn/products/zigbee-shield](url)