Koenkk/zigbee2mqtt #1440: friendly_names in state topics instead of DEVICE_ID

nibi79, 2019-04-17:
hi,

I've updated my installation to version 1.3.0. I don't know if this behavior was in 1.2.0 too, because I don't use friendly_names before.

When I configure friendly_names zigbee2mqtt publishes all messages to:
`zigbee2mqtt/[friendly_name]`
instead of:
`zigbee2mqtt/[DEVICE_ID]`
 
Is that correct ?

Let me give you an example with my Osram plug.
`conficuration.yaml`

> '0x7cb03eaa00b1d78a':
>     friendly_name: 'myplug'
>     retain: false

Topic: `homeassistant/sensor/0x7cb03eaa00b1d78a/linkquality/config`

> {
>   "unit_of_measurement": "-",
>   "value_template": "{{ value_json.linkquality }}",
>   **"state_topic": "zigbee2mqtt/0x7cb03eaa00b1d78a",**
>   "json_attributes_topic": "zigbee2mqtt/0x7cb03eaa00b1d78a",
>   "name": "0x7cb03eaa00b1d78a_linkquality",
>   "unique_id": "0x7cb03eaa00b1d78a_linkquality_zigbee2mqtt",
>   "device": {
>     "identifiers": "zigbee2mqtt_0x7cb03eaa00b1d78a",
>     "name": "0x7cb03eaa00b1d78a",
>     "sw_version": "Zigbee2mqtt 1.3.0",
>     "model": "Smart+ plug (AB3257001NJ)",
>     "manufacturer": "OSRAM"
>   },
>   "availability_topic": "zigbee2mqtt/bridge/state"
> }

Topic: `homeassistant/switch/0x7cb03eaa00b1d78a/switch/config`
> {
>   "payload_off": "OFF",
>   "payload_on": "ON",
>   "value_template": "{{ value_json.state }}",
>   **"command_topic": "zigbee2mqtt/0x7cb03eaa00b1d78a/set",**
>   **"state_topic": "zigbee2mqtt/0x7cb03eaa00b1d78a",**
>   "name": "0x7cb03eaa00b1d78a_switch",
>   "unique_id": "0x7cb03eaa00b1d78a_switch_zigbee2mqtt",
>   "device": {
>     "identifiers": "zigbee2mqtt_0x7cb03eaa00b1d78a",
>     "name": "0x7cb03eaa00b1d78a",
>     "sw_version": "Zigbee2mqtt 1.3.0",
>     "model": "Smart+ plug (AB3257001NJ)",
>     "manufacturer": "OSRAM"
>   },
>   "availability_topic": "zigbee2mqtt/bridge/state"
> }

But when I now publish a message to `zigbee2mqtt/0x7cb03eaa00b1d78a/get`  or `zigbee2mqtt/myplug/get` zigbee2mqtt only and always replies to `zigbee2mqtt/myplug`:

> {
>   "state": "ON",
>   "linkquality": 0,
>   "device": {
>     "ieeeAddr": "0x7cb03eaa00b1d78a",
>     "friendlyName": "myplug",
>     "type": "Router",
>     "nwkAddr": 34286,
>     "manufId": 48042,
>     "manufName": "OSRAM",
>     "powerSource": "Mains (single phase)",
>     "modelId": "Plug 01",
>     "hwVersion": "unknown",
>     "swBuildId": "unknown",
>     "dateCode": "unknown",
>     "status": "offline"
>   }
> }

Koenkk, 2019-04-17:
Can you provide your zigbee2mqtt startup log?

nibi79, 2019-04-17:
sorry their were some retained messages which results in some strange behaviour. 
I edited my post a second ago.

Koenkk, 2019-04-17:
Yes, device id is the `friendly_name` from `configuration.yaml`.

nibi79, 2019-04-17:
Thank you for the clarification!

closed 2019-04-17 by nibi79