Koenkk/zigbee2mqtt #1444: Not able to decode the writeRsp message

he902, 2019-04-18:
The Wireshark showed that **writeRsp** message received from my Yunmi Lock device:
```
ZigBee Cluster Library Frame, Command: Write Attributes Response, Seq: 12
    Frame Control Field: Profile-wide (0x18)
    Sequence Number: 12
    Command: Write Attributes Response (0x04)
    Status Record
        Status: Unsupported Attribute (0x86)
        Attribute: Unknown (0x410d)
```

But zigbee2mqtt was not able to decode it:
```
  serialport/bindings/unixRead Starting read +8ms
  serialport/bindings/unixRead Finished read 31 bytes +0ms
  serialport/stream binding.read finished +12ms
  cc-znp { sof: 254,
  cc-znp   len: 26,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'AF',
  cc-znp   cmd: 'incomingMsg',
  cc-znp   payload:
  cc-znp    { groupid: 0,
  cc-znp      clusterid: 257,
  cc-znp      srcaddr: 38955,
  cc-znp      srcendpoint: 1,
  cc-znp      dstendpoint: 1,
  cc-znp      wasbroadcast: 0,
  cc-znp      linkquality: 78,
  cc-znp      securityuse: 0,
  cc-znp      timestamp: 7888958,
  cc-znp      transseqnumber: 0,
  cc-znp      len: 6,
  cc-znp      data: <Buffer 18 02 04 86 0d 41> },
  cc-znp   fcs: 120,
  cc-znp   csum: 120 } +17ms
  serialport/stream _read reading +6ms
  serialport/binding-abstract read +8ms
  serialport/bindings/unixRead Starting read +14ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 257, srcaddr: 38955, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 78, securityuse: 0, timestamp: 7888958, transseqnumber: 0, len: 6, data: <Buffer 18 02 04 86 0d 41> } +25ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +26ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +34ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +12ms
```

I have tried to add 'writeRsp' into `/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/af.js`, but no luck:
```
            const cmdIDs = ['on', 'offWithEffect', 'step', 'stop', 'hueNotification', //Philips-RWL020-RWL021
                            'off', 'stepColorTemp', 'moveWithOnOff', 'move', 'moveHue', 'moveToSaturation', //OSRAM-Switch 4x EU-LIGHTIFY
                            'stopWithOnOff', 'moveToLevelWithOnOff', 'toggle', 'tradfriArrowSingle', 'tradfriArrowHold', 'tradfriArrowRelease',
                            'stepWithOnOff', 'moveToColorTemp', 'moveToColor', 'onWithTimedOff', 'recall',
                            'operationEventNotification', 'unlockDoorRsp', 'getRfidCodeRsp', 'getPinCodeRsp', 'getLogRecordRsp',
                            'getUserTypeRsp', 'getUserStatusRsp', 'writeRsp'];
```

he902, 2019-04-18:
There is:
```
        "writeRsp": {
            "params":[
                {"extra":"writeRsp"}
            ],
            "knownBufLen": 0
        },
```
defined in `/opt/zigbee2mqtt//node_modules/zcl-packet/lib/defs/zcl_meta.json`.
But I don't understand how `extra` works.

he902, 2019-04-19:
Upgraded to the the latest 1.3.1, now it can decode the `writeRsp`:
```
  serialport/bindings/unixRead Starting read +0ms
  serialport/bindings/unixRead Finished read 31 bytes +1ms
  serialport/stream binding.read finished +3ms
  cc-znp { sof: 254,
  cc-znp   len: 26,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'AF',
  cc-znp   cmd: 'incomingMsg',
  cc-znp   payload:
  cc-znp    { groupid: 0,
  cc-znp      clusterid: 257,
  cc-znp      srcaddr: 38955,
  cc-znp      srcendpoint: 1,
  cc-znp      dstendpoint: 1,
  cc-znp      wasbroadcast: 0,
  cc-znp      linkquality: 78,
  cc-znp      securityuse: 0,
  cc-znp      timestamp: 1231647,
  cc-znp      transseqnumber: 0,
  cc-znp      len: 6,
  cc-znp      data: <Buffer 18 02 04 86 0d 41> },
  cc-znp   fcs: 152,
  cc-znp   csum: 152 } +8ms
  serialport/stream _read reading +5ms
  serialport/binding-abstract read +7ms
  serialport/bindings/unixRead Starting read +6ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 257, srcaddr: 38955, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 78, securityuse: 0, timestamp: 1231647, transseqnumber: 0, len: 6, data: <Buffer 18 02 04 86 0d 41> } +9ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":257,"srcaddr":38955,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":78,"securityuse":0,"timestamp":1231647,"transseqnumber":0,"len":6,"data":{"type":"Buffer","data":[24,2,4,134,13,65]}} +9ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +15ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +7ms
  serialport/bindings/poller Polling for "readable" +15ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":257,"srcaddr":38955,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":78,"securityuse":0,"timestamp":1231647,"transseqnumber":0,"len":6,"data":{"0":24,"1":2,"2":4,"3":134,"4":13,"5":65},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":2,"cmdId":"writeRsp","payload":[{"status":134,"attrId":16653}]}} +7ms
```

he902, 2019-04-19:
But not able to see below format of DEBUG message:
```
Received zigbee message of type 'xxxx' with data '{"cid":"yyyy","data":{...}}' of device 'FNB56-DOR23FB1.3' (0x00158d0002acba74) of endpoint 1
```
Without:
**type 'xxxx' 
"cid":"yyyy"**
I'm not able to add support of this message into toZigbee.js.

Koenkk, 2019-04-19:
I don't understand what do you see of `Received zigbee message... ` of this message?

he902, 2019-04-20:
I mean the DEBUG log output at:
[zigbee2mqtt/lib/controller.js](https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/controller.js#L135)
```
        logger.debug(
            `Received zigbee message of type '${message.type}' with data '${JSON.stringify(message.data)}'` +
            (device ? ` of device '${device.modelId}' (${device.ieeeAddr})` : '') +
            (message.groupid ? ` with groupID ${message.groupID}` : '') +
            (message.endpoints && message.endpoints.length ? ` of endpoint ${message.endpoints[0].epId}` : '')
        );
```
Such code piece has not been executed for `writeRsp`.
Or say the `onZigbeeMessage()` has not been triggered.

he902, 2019-04-20:
In `/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/af.js`, there is piece of code:
```
            const cmdIDs = [
                'on', 'offWithEffect', 'step', 'stop', 'hueNotification',
                'off', 'stepColorTemp', 'moveWithOnOff', 'move', 'moveHue', 'moveToSaturation',
                'stopWithOnOff', 'moveToLevelWithOnOff', 'toggle', 'tradfriArrowSingle', 'tradfriArrowHold', 'tradfriArrowRelease',
                'stepWithOnOff', 'moveToColorTemp', 'moveToColor', 'onWithTimedOff', 'recall', 'arm', 'panic', 'emergency',
                'operationEventNotification', 'unlockDoorRsp', 'getRfidCodeRsp', 'getPinCodeRsp', 'getLogRecordRsp',
                'getUserTypeRsp', 'getUserStatusRsp', 'writeRsp'
            ];

            if (frameType === 1 && cmdIDs.includes(msg.zclMsg.cmdId) && msg.zclMsg.payload) {
                af.controller._shepherd.emit('ind:cmd', targetEp, msg.clusterid, msg.zclMsg.payload, msg.zclMsg.cmdId, msg);
            }
```
My `writeRsp` comes with `frameType = 0`, so not triggered the `onZigbeeMessage()` in zigbee2mqtt's `controller.js`.

he902, 2019-04-20:
I modified `/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/af.js` as:
```
            if ((frameType === 0 || frameType === 1) && cmdIDs.includes(msg.zclMsg.cmdId) && msg.zclMsg.payload) {
                af.controller._shepherd.emit('ind:cmd', targetEp, msg.clusterid, msg.zclMsg.payload, msg.zclMsg.cmdId, msg);
            }
```

Now, I can see the familiar debug log, and be able to add new converter to support it:
```
4/20/2019, 11:27:31 AM - info: Zigbee publish to device '0x00158d0002acba74', closuresDoorLock - write - [{"attrId":16653,"dataType":32,"attrData":0}] - {"manufSpec":0,"disDefaultRsp":1} - null
4/20/2019, 11:27:31 AM - debug: Received zigbee message of type 'cmdWriteRsp' with data '{"cid":"closuresDoorLock","data":[{"status":134,"attrId":16653}]}' of device 'FNB56-DOR23FB1.3' (0x00158d0002acba74) of endpoint 1
4/20/2019, 11:27:31 AM - warn: No converter available for 'FNB56-DOR23FB1.3' with cid 'closuresDoorLock', type 'cmdWriteRsp' and data '{"cid":"closuresDoorLock","data":[{"status":134,"attrId":16653}]}'
4/20/2019, 11:27:31 AM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```

closed 2019-06-26 by stale[bot]