Koenkk/zigbee2mqtt #1445: Z-Stack 3 on CC2531 adventures

Koenkk, 2019-04-18:
Texas Instruments has recently released a new high performance Zigbee chip: the CC2652R. This chip is much faster and contains much more memory than the CC2530/CC2531. 

This issue is created in order to gather feedback of users using the CC2652R **AND** Z-Stack 3.0 (= Zigbee 3.0)

**You are about to enter a highly experimental area (no guarantees)**

**Notes**
- Zigbee 3.0!
- Firmware can be found here in the bin folder: https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/Z-Stack_3.0.x
- It's not possible to migrate your Zigbee 1.2 network (requires repairing)
- The network restore trick that we used in Zigbee 1.2 doesn't work anymore (due to better security policies). A backup of the CC2531 is taken at every shutdown of zigbee2mqtt, file can be found under `data/coordinator_backup.json`. This backup is automatically restored when starting with a new flashed CC2531.
- Requires latest zigbee2mqtt dev branch.

**Known issues**
- Pan ID and extended pan ID can't be set (probably not something to worry about)
- Pairing only seems possible during the first 180 seconds (even with `permit_join: true`, a workaround is to restart zigbee2mqtt.

jarrah31, 2019-04-23:
What are the benefits of Zigbee 3.0 for zigbee2mqtt? Will it allow more than the default 15 devices on a CC2531 coordinator (and if so how many)? Do we need new firmware on the CC2531 routers as well?

btw, thanks @koenkk for a fantastic product and all the effort you put into this project, it is very much appreciated!

Koenkk, 2019-04-23:
@jarrah31 it also has a direct children limit of 15. No new router firmware is needed (zigbee 1.2 devices can join a zigbee 3.0 network).

I've just released a new firmware `20190423` (https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/Z-Stack_3.0.x/bin).)
- Increased stability
- Try to fix issue that kicked of Xiaomi devices from the network
- Increased number of direct children from 10 to 15

Z0l, 2019-04-27:
Hey, I'm new to this whole Zigbee thing but I got this one up and running with the future intent of integrating with Openhab to finally make my Hue Tap do things Philips never dreamed of :) Is there documentation available how GPP devices can be paired? Perhaps something specific for Hue Tap? I'm running 20190423 with the latest dev Docker container.

Koenkk, 2019-04-28:
@Z0l you can try to pair it, setup procedure seems to be here: https://huetips.com/tutorials/how-to-add-a-hue-tap/  (10 seconds press a button).

runningman84, 2019-04-28:
I have flashed my cc2531 with CC2531ZNP-with-SBL.hex:
```
sudo ./cc-tool -e -w CC2531ZNP-with-SBL.hex                                                                                                                                            ±[●][master]
[sudo] Passwort für phil: 
  Programmer: CC Debugger
  Target: CC2531
  Erasing flash...
  Completed       
  Writing flash (241 KB)...
  Completed (17.97 s.)
```

but the stick does not seem to work with zigbee2mqtt anymore:
```
> zigbee2mqtt@1.3.1 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 4/28/2019, 9:59:45 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-04-28.21-59-45'
  zigbee2mqtt:info 4/28/2019, 9:59:46 PM Starting zigbee2mqtt version 1.3.1 (commit #39d6a8e)
  zigbee2mqtt:info 4/28/2019, 9:59:46 PM Starting zigbee-shepherd
  zigbee2mqtt:info 4/28/2019, 9:59:51 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
  zigbee2mqtt:info 4/28/2019, 10:00:51 PM Starting zigbee-shepherd
  zigbee2mqtt:error 4/28/2019, 10:00:56 PM Error while starting zigbee-shepherd!
  zigbee2mqtt:error 4/28/2019, 10:00:56 PM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 4/28/2019, 10:00:56 PM Failed to start
        {"message":"rsp error: 2","stack":"Error: rsp error: 2\n    at rspHdlr (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/controller.js:309:29)\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:264:13)\n    at Object.onceWrapper (events.js:277:13)\n    at CcZnp.emit (events.js:189:13)\n    at CcZnp._mtIncomingDataHdlr (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:366:14)\n    at Immediate._onImmediate (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:342:22)\n    at runCallback (timers.js:705:18)\n    at tryOnImmediate (timers.js:676:5)\n    at processImmediate (timers.js:658:5)"}
  zigbee2mqtt:error 4/28/2019, 10:00:56 PM Exiting...
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.3.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.3.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-04-28T21_00_56_252Z-debug.log
```

after pressing the reset button on the stick I get this error:
```
pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.3.1 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 4/28/2019, 10:03:41 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-04-28.22-03-41'
  zigbee2mqtt:info 4/28/2019, 10:03:42 PM Starting zigbee2mqtt version 1.3.1 (commit #39d6a8e)
  zigbee2mqtt:info 4/28/2019, 10:03:42 PM Starting zigbee-shepherd
  zigbee2mqtt:info 4/28/2019, 10:03:44 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
   zigbee2mqtt:info 4/28/2019, 10:04:44 PM Starting zigbee-shepherd
  zigbee2mqtt:error 4/28/2019, 10:04:45 PM Error while starting zigbee-shepherd!
  zigbee2mqtt:error 4/28/2019, 10:04:45 PM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 4/28/2019, 10:04:45 PM Failed to start
        {"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:261:22)\n    at Object.onceWrapper (events.js:277:13)\n    at CcZnp.emit (events.js:189:13)\n    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:240:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
  zigbee2mqtt:error 4/28/2019, 10:04:45 PM Exiting...
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.3.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.3.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-04-28T21_04_45_888Z-debug.log
```

the same stick worked fine with the old version:
```
[... 1 lines cut ...]

> zigbee2mqtt@1.3.1 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 4/28/2019, 8:29:24 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-04-28.20-29-24'
  zigbee2mqtt:info 4/28/2019, 8:29:25 PM Starting zigbee2mqtt version 1.3.1 (commit #39d6a8e)
  zigbee2mqtt:info 4/28/2019, 8:29:25 PM Starting zigbee-shepherd
  zigbee2mqtt:info 4/28/2019, 8:29:25 PM zigbee-shepherd started
  zigbee2mqtt:info 4/28/2019, 8:29:25 PM Coordinator firmware version: '20190223'
  zigbee2mqtt:info 4/28/2019, 8:29:25 PM Currently 1 devices are joined:
  zigbee2mqtt:info 4/28/2019, 8:29:25 PM 0x7cb03eaa00a6f7a3 (0x7cb03eaa00a6f7a3): AB3257001NJ - OSRAM Smart+ plug (Router)
  zigbee2mqtt:warn 4/28/2019, 8:29:25 PM `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 4/28/2019, 8:29:25 PM Allowing new devices to join.
  zigbee2mqtt:warn 4/28/2019, 8:29:25 PM Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 4/28/2019, 8:29:25 PM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 4/28/2019, 8:29:25 PM Connecting to MQTT server at mqtt://192.168.178.71
  zigbee2mqtt:info 4/28/2019, 8:29:26 PM zigbee-shepherd ready
[... cut ...] aa00a6f7a3","name":"0x7cb03eaa00a6f7a3","sw_version":"Zigbee2mqtt 1.3.1","model":"Smart+ plug (AB3257001NJ)","manufacturer":"OSRAM"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 4/28/2019, 8:29:26 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.3.1","commit":"39d6a8e","coordinator_firmware":20190223,"log_level":"info","permit_join":true}'
  zigbee2mqtt:info 4/28/2019, 8:29:26 PM Successfully configured 0x7cb03eaa00a6f7a3 (0x7cb03eaa00a6f7a3)
```

runningman84, 2019-04-28:
Btw. the same error happens in the dev branch (commit #4472c01)

Koenkk, 2019-04-29:
Can you provide the output of `DEBUG=* npm start` (on https://pastebin.com/)

Z0l, 2019-04-29:
For me it works with the dev version:

`4/28/2019, 8:58:36 PM - info: Logging to directory: '/app/data/log/2019-04-28.20-58-36'
4/28/2019, 8:58:36 PM - info: Starting zigbee2mqtt version 1.3.1 (commit #unknown)
4/28/2019, 8:58:36 PM - info: Starting zigbee-shepherd
4/28/2019, 8:58:38 PM - info: zigbee-shepherd started
4/28/2019, 8:58:38 PM - info: Coordinator firmware version: '20190425'
4/28/2019, 8:58:38 PM - info: Currently 0 devices are joined:
4/28/2019, 8:58:38 PM - warn: `permit_join` set to  `true` in configuration.yaml.
4/28/2019, 8:58:38 PM - warn: Allowing new devices to join.
4/28/2019, 8:58:38 PM - warn: Set `permit_join` to `false` once you joined all devices.
4/28/2019, 8:58:38 PM - info: Zigbee: allowing new devices to join.
4/28/2019, 8:58:38 PM - info: Connecting to MQTT server at mqtt://localhost
4/28/2019, 8:58:38 PM - info: zigbee-shepherd ready
4/28/2019, 8:58:38 PM - info: Connected to MQTT server
4/28/2019, 8:58:38 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
4/28/2019, 8:58:38 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.3.1","commit":"unknown","coordinator":20190425,"log_level":"info","permit_join":true}'`

However nothing happens when I keep pushing the button for 10 sec (I tried pushing all 4). Is the limitation on pairing has to happen in the first 180 seconds still in place? Also, is that normal that the led turns off after some time (didn't measure how long, but when I plug the USB in and start the docker container, the LED turns off).

lolorc, 2019-04-29:
@Z0l the pairing limitation is actually a zigbee 3.0 feature :-)
https://www.ti.com/lit/an/swra615/swra615.pdf "Permit join can no longer be enabled forever as networks will automatically close joining after a maximum of 254 seconds."
just resend the mqtt cmd permit_join true when the 254s have passed.

runningman84, 2019-04-29:
@lolorc it would be cool if zigbee2mqtt would expose a switch like "permit_join" to homeassistant.

lolorc, 2019-04-29:
@runningman84 a hack is probably already possible. using zigbee2mqtt/bridge/config/permit_join as a mqtt switch with retain true. I guess it would also be possible to tell home assistant to set it to false 254s after it has been set to true.
but I agree it would better to be able to read from zigbee2mqtt/bridge/config/permit_join and get the updated status, and also having z2m setting it to false after 254s

Koenkk, 2019-04-29:
@runningman84 @lolorc don't spend time on this. Soon I will implement something that zigbee2mqtt does this automatically.

runningman84, 2019-04-29:
@Koenkk
> Can you provide the output of `DEBUG=* npm start` (on https://pastebin.com/)

https://pastebin.com/h0hFLEuA

Koenkk, 2019-04-29:
@runningman84 after updating, did you do a `rm -rf node_modules && npm install`?

runningman84, 2019-04-29:
I just run rm -rf node_modules && npm install and flashed the firmware once again. Now it seems to start:
```
pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.3.1 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 4/29/2019, 8:49:59 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-04-29.20-49-59'
  zigbee2mqtt:info 4/29/2019, 8:50:00 PM Starting zigbee2mqtt version 1.3.1 (commit #a54e256)
  zigbee2mqtt:info 4/29/2019, 8:50:00 PM Starting zigbee-shepherd
  zigbee2mqtt:info 4/29/2019, 8:50:02 PM zigbee-shepherd started
  zigbee2mqtt:info 4/29/2019, 8:50:02 PM Coordinator firmware version: '20190425'
  zigbee2mqtt:info 4/29/2019, 8:50:02 PM Currently 1 devices are joined:
  zigbee2mqtt:info 4/29/2019, 8:50:02 PM 0x7cb03eaa00a6f7a3 (0x7cb03eaa00a6f7a3): AB3257001NJ - OSRAM Smart+ plug (Router)
  zigbee2mqtt:warn 4/29/2019, 8:50:02 PM `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 4/29/2019, 8:50:02 PM Allowing new devices to join.
  zigbee2mqtt:warn 4/29/2019, 8:50:02 PM Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 4/29/2019, 8:50:02 PM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 4/29/2019, 8:50:02 PM Connecting to MQTT server at mqtt://192.168.178.71
  zigbee2mqtt:info 4/29/2019, 8:50:02 PM zigbee-shepherd ready
  zigbee2mqtt:info 4/29/2019, 8:50:02 PM Connected to MQTT server
  zigbee2mqtt:info 4/29/2019, 8:50:02 PM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 4/29/2019, 8:50:02 PM MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa00a6f7a3', payload '{"state":"ON","linkquality":0}'
  zigbee2mqtt:info 4/29/2019, 8:50:02 PM MQTT publish: topic 'homeassistant/switch/0x7cb03eaa00a6f7a3/switch/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state }}","command_topic":"zigbee2mqtt/0x7cb03eaa00a6f7a3/set","state_topic":"zigbee2mqtt/0x7cb03eaa00a6f7a3","name":"0x7cb03eaa00a6f7a3_switch","unique_id":"0x7cb03eaa00a6f7a3_switch_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x7cb03eaa00a6f7a3","name":"0x7cb03eaa00a6f7a3","sw_version":"Zigbee2mqtt 1.3.1","model":"Smart+ plug (AB3257001NJ)","manufacturer":"OSRAM"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 4/29/2019, 8:50:02 PM MQTT publish: topic 'homeassistant/sensor/0x7cb03eaa00a6f7a3/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x7cb03eaa00a6f7a3","json_attributes_topic":"zigbee2mqtt/0x7cb03eaa00a6f7a3","name":"0x7cb03eaa00a6f7a3_linkquality","unique_id":"0x7cb03eaa00a6f7a3_linkquality_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x7cb03eaa00a6f7a3","name":"0x7cb03eaa00a6f7a3","sw_version":"Zigbee2mqtt 1.3.1","model":"Smart+ plug (AB3257001NJ)","manufacturer":"OSRAM"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 4/29/2019, 8:50:02 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.3.1","commit":"a54e256","coordinator":20190425,"log_level":"info","permit_join":true}'
  zigbee2mqtt:warn 4/29/2019, 8:50:14 PM No converter available for 'AB3257001NJ' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"manufacturerName":"OSRAM","modelId":"Plug 01","powerSource":1}}'
  zigbee2mqtt:warn 4/29/2019, 8:50:14 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 4/29/2019, 8:50:14 PM No converter available for 'AB3257001NJ' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":12,"stackVersion":2,"hwVersion":1,"dateCode":"20140331DEOS****","swBuildId":"V1.04.12"}}'
  zigbee2mqtt:warn 4/29/2019, 8:50:14 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 4/29/2019, 8:50:15 PM No converter available for 'AB3257001NJ' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"deviceEnabled":1}}'
  zigbee2mqtt:warn 4/29/2019, 8:50:15 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 4/29/2019, 8:50:15 PM No converter available for 'AB3257001NJ' with cid 'genIdentify', type 'devChange' and data '{"cid":"genIdentify","data":{"identifyTime":0}}'
  zigbee2mqtt:warn 4/29/2019, 8:50:15 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 4/29/2019, 8:50:15 PM No converter available for 'AB3257001NJ' with cid 'genGroups', type 'devChange' and data '{"cid":"genGroups","data":{"nameSupport":0}}'
  zigbee2mqtt:warn 4/29/2019, 8:50:15 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 4/29/2019, 8:50:15 PM No converter available for 'AB3257001NJ' with cid 'genScenes', type 'devChange' and data '{"cid":"genScenes","data":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}}'
  zigbee2mqtt:warn 4/29/2019, 8:50:15 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:info 4/29/2019, 8:50:16 PM MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa00a6f7a3', payload '{"state":"ON","linkquality":84}'
  zigbee2mqtt:warn 4/29/2019, 8:50:16 PM No converter available for 'AB3257001NJ' with cid 'haElectricalMeasurement', type 'devChange' and data '{"cid":"haElectricalMeasurement","data":{"measurementType":8,"activePower":400}}'
[... cut ...]
  zigbee2mqtt:info 4/29/2019, 8:50:16 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:warn 4/29/2019, 8:50:53 PM Failed to configure 0x7cb03eaa00a6f7a3 (0x7cb03eaa00a6f7a3) ('Error: Timed out after 10000 ms') (attempt #1)
  zigbee2mqtt:warn 4/29/2019, 8:50:53 PM This can be ignored if the device is working properly
```

But the OSRAM plug does not seem to work well with this firmware or software. State changes are not reported and there are these timed out error messages.

runningman84, 2019-04-29:
I just removed the plug from the outlet and put it back in... now the state change is recognized:
```
  zigbee2mqtt:warn 4/29/2019, 8:50:53 PM Failed to configure 0x7cb03eaa00a6f7a3 (0x7cb03eaa00a6f7a3) ('Error: Timed out after 10000 ms') (attempt #1)
  zigbee2mqtt:warn 4/29/2019, 8:50:53 PM This can be ignored if the device is working properly
  zigbee2mqtt:info 4/29/2019, 8:53:20 PM Successfully configured 0x7cb03eaa00a6f7a3 (0x7cb03eaa00a6f7a3)
  zigbee2mqtt:info 4/29/2019, 8:53:22 PM MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa00a6f7a3', payload '{"state":"OFF","linkquality":86}'
  zigbee2mqtt:info 4/29/2019, 8:53:24 PM MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa00a6f7a3', payload '{"state":"ON","linkquality":84}'
  zigbee2mqtt:info 4/29/2019, 8:53:25 PM MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa00a6f7a3', payload '{"state":"OFF","linkquality":86}'
  zigbee2mqtt:info 4/29/2019, 8:53:26 PM MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa00a6f7a3', payload '{"state":"ON","linkquality":81}'
  zigbee2mqtt:info 4/29/2019, 8:53:27 PM MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa00a6f7a3', payload '{"state":"OFF","linkquality":86}'
  zigbee2mqtt:info 4/29/2019, 8:53:28 PM MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa00a6f7a3', payload '{"state":"ON","linkquality":84}'
  zigbee2mqtt:info 4/29/2019, 8:53:28 PM MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa00a6f7a3', payload '{"state":"OFF","linkquality":89}'
```

GlitchHound, 2019-05-07:
I've got this running, I took the plunge straight into Z-Stack 3 since I don't have any of my network setup. No bugs to report yet, but are there any instructions on how to use touchlink in this? Or is the whole point that we now _could_ make touchlink work, rather than that it already _is working_?

My hue bulbs don't even like resetting with the API touchlink trick on the original bridge, but using my Conbee (which I'm hoping to replace for the more open zigbee2mqtt) I was able to reset them and then get them onto zigbee2mqtt, so I'd love to see touchlink without any other device.

Only a couple of devices so far, and I'm not doing anything sophisticated, but it is working with what I've paired up (a GE switch and a hue bulb)

Koenkk, 2019-05-08:
@GlitchHound good idea! I will look into this, please keep an eye on https://github.com/Koenkk/zigbee2mqtt/issues/1509

sehraf, 2019-05-19:
Do i understand this correctly:
> The network restore trick that we used in Zigbee 1.2 doesn't work anymore (due to better security policies). A backup of the CC2531 is taken at every shutdown of zigbee2mqtt, file can be found under [...]

Does this imply that i can even change the hardware without the need to repair all devices? Or does this "only" cover firmware changes?

Koenkk, 2019-05-19:
@sehraf yes, hardware can be changed without repairing (was also possible with 1.2). Note that when going from zigbee 1.2 to 3.0 you have to re-pair.

drbios, 2019-05-20:
<script src="https://gist.github.com/drbios/8a9894ea5131d97a99b0d7f2bbc291b5.js"></script>

drbios, 2019-05-20:
everything Working fine but when i stop the zigbee2mqtt service with availability_timeout: option in the configuration.yaml this is the log:

May 20 11:50:54 hassbian systemd[1]: Stopping zigbee2mqtt...
May 20 11:50:54 hassbian npm[11064]: #33[32m zigbee2mqtt:info#033[39m 5/20/2019, 11:50:54 AM MQTT publish: topic 'zigbee2mqtt/0x00158d0001c9affc/availability', payload 'offline'
May 20 11:50:54 hassbian npm[11064]: #33[32m zigbee2mqtt:info#033[39m 5/20/2019, 11:50:54 AM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
May 20 11:50:54 hassbian npm[11064]: #33[32m zigbee2mqtt:info#033[39m 5/20/2019, 11:50:54 AM Disconnecting from MQTT server
May 20 11:50:54 hassbian mosquitto[3833]: 1558367454: Client mqttjs_0943bd74 disconnected.
May 20 11:50:54 hassbian systemd[1]: zigbee2mqtt.service: Main process exited, code=killed, status=15/TERM
May 20 11:50:54 hassbian npm[11064]: #33[32m zigbee2mqtt:info#033[39m 5/20/2019, 11:50:54 AM MQTT publish: topic 'zigbee2mqtt/0x00158d0001c9affc/availability', payload 'offline'
May 20 11:50:54 hassbian npm[11064]: events.js:174
May 20 11:50:54 hassbian npm[11064]: throw er; // Unhandled 'error' event
May 20 11:50:54 hassbian npm[11064]: ^
May 20 11:50:54 hassbian npm[11064]: Error: client disconnecting
May 20 11:50:54 hassbian npm[11064]: at MqttClient._checkDisconnecting (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:370:26)
May 20 11:50:54 hassbian npm[11064]: at MqttClient.publish (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:408:12)
May 20 11:50:54 hassbian npm[11064]: at MQTT.publish (/opt/zigbee2mqtt/lib/mqtt.js:95:21)
May 20 11:50:54 hassbian npm[11064]: at DeviceAvailability.publishAvailability (/opt/zigbee2mqtt/lib/extension/deviceAvailability.js:146:19)
May 20 11:50:54 hassbian npm[11064]: at zigbee.getDevices.filter.forEach (/opt/zigbee2mqtt/lib/extension/deviceAvailability.js:106:39)
May 20 11:50:54 hassbian npm[11064]: at Array.forEach ()
May 20 11:50:54 hassbian npm[11064]: at DeviceAvailability.stop (/opt/zigbee2mqtt/lib/extension/deviceAvailability.js:106:14)
May 20 11:50:54 hassbian npm[11064]: at extensions.filter.forEach (/opt/zigbee2mqtt/lib/controller.js:177:64)
May 20 11:50:54 hassbian npm[11064]: at Array.forEach ()
May 20 11:50:54 hassbian npm[11064]: at Controller.stop (/opt/zigbee2mqtt/lib/controller.js:177:47)
May 20 11:50:54 hassbian npm[11064]: Emitted 'error' event at:
May 20 11:50:54 hassbian npm[11064]: at MqttClient._checkDisconnecting (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:370:12)
May 20 11:50:54 hassbian npm[11064]: at MqttClient.publish (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:408:12)
May 20 11:50:54 hassbian npm[11064]: [... lines matching original stack trace ...]
May 20 11:50:54 hassbian npm[11064]: at Controller.stop (/opt/zigbee2mqtt/lib/controller.js:177:47)
May 20 11:50:54 hassbian systemd[1]: zigbee2mqtt.service: Succeeded.
May 20 11:50:54 hassbian systemd[1]: Stopped zigbee2mqtt.

Koenkk, 2019-05-20:
@drbios fixed in latest dev.

drbios, 2019-05-20:
@Koenkk excellent!! i will update 
excellent thank you very much ... amazing work

Koenkk, 2019-05-24:
CC2530 and  CC2530_CC2591 firmwares are also available: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.0.x/bin

ekaats, 2019-05-25:
I started using the Z-Stack 3 a couple of days ago. So far I haven't had any stability issues as I had with the older stack. However, somehow I cannot get my two IKEA Floalt to connect. Other IKEA bulbs do not pose a problem, and even battery powered Xiaomi connect with little to no problem. Anyone else had problems with these lamps?

hanzoh, 2019-05-28:
I have updated my CC2530_CC2591 to the Z-Stack 3 firmware and have repaired all my devices.
After a few minutes though, all devices are going offline.
Here is my config and I have attached a log with debug enabled.
I'm running the latest Docker dev image.

´´´yaml
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://192.168.1.10'
serial:
  port: /dev/zigbee
advanced:
  pan_id: 0x1a62
  rtscts: false
  log_level: debug
  availability_timeout: 60
´´´

[zigbee.log](https://github.com/Koenkk/zigbee2mqtt/files/3228686/zigbee.log)

hanzoh, 2019-05-28:
When I disable the availability_timeout, it last a bit longer, but then crashes after the second or so network update for refreshing the network map in Home Assistant. I have now disabled that automation as well and will monitor how long it lasts. This heavy polling lets the CC2530 crash somehow.

Koenkk, 2019-05-28:
Can you post the `log_level: debug` when this happens (http://www.zigbee2mqtt.io/configuration/configuration.html)

milakov, 2019-05-29:
Works fine for me on cc2531

drbios, 2019-05-29:
Hello ... happens the same to me using zs3 and a cc2530+2591,suddenly everything goes offline and I had to reinstall everything  again ...no possibilitie of repairing.
Now I am using   zs1.2.2 and not using availability_timeout it's much more stable.

hanzoh, 2019-05-29:
I have reenabled the /10m network scan for the map in home assistant, and so far it has been stable. So it really must be the availability check that crashes the network. Had to unplug and replug the USB to serial converter to get it started again.

hanzoh, 2019-05-29:
Okay, 1,5 hours later, and it crashed again.
@Koenkk Please find attached the debug log since z2m started this morning at 2:00AM (automatic update to latest-dev Docker image).
It fails at 5/29/2019, 5:50:00 PM, I reenabled the network map refresh (every 10 minutes) at 5/29/2019, 3:52:41 PM.

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3233793/log.txt)

Koenkk, 2019-05-29:
@hanzoh probably happens because of https://github.com/Koenkk/zigbee2mqtt/issues/652#issuecomment-496842986. Without the networkmap everything should be working fine.

ekaats, 2019-06-02:
Today I found out that the Z-Stack 3 firmware might have some issues with an explicitly set network key. After reading about the (possible?) security settings when not using a network key, I decided to add one and change my network to another channel as well. (re)pairing went smooth but after awhile I noticed instability in the network, particularly from Ikea devices. Furthermore, upon reset they seemed to be unpaired or otherwise unreachable. Other devices also showed faults but these or for the most part sensors that just didn't register anything, so I guess they were unpaired/unreachable as well.

Removing the custom channel did not help. Instead it made repairing impossible somehow. Changing the network key did help, or so it seemed. I was able to restart zigbee2mqtt and reconnect with my devices. To be sure I reflashed the old Z-Stack and sure enough, using the same settings both custom channel and network key worked.

So, tl;dr: Setting network key  (and possibly channel) seems broken on Z-Stack CC2531.

milakov, 2019-06-02:
Non-default channel works fine for me with different xiaomi devices in the network with cc2531 flashed with z-stack 3

CypherMK, 2019-06-03:
Will there ever be an option to migrate all my configured devices from 1.x to 3.x? 
I have 2 hue bulbs which I can't reset, cause I don't have the remote.

Koenkk, 2019-06-03:
@CypherMK probably not

faronov, 2019-06-04:
can you build this firmware for CC2530+CC2592 ?

matlab22, 2019-06-05:
Where does the device limit of 40devices comes from? I already have about 50 devices in my network. Does this mean I have to split up the network if I ever upgrade to 3.0?

Koenkk, 2019-06-05:
@faronov does the Z-Stack 1.2 CC2592 work stable now?

@matlab22 this is due to a new security requirement in Zigbee 3. Note that this only applies to Zigbee 3 devices, meaning that you can have max 40 Zigbee 3 devices in your network AND an unlimited amount of Zigbee 1.2 devices.

matlab22, 2019-06-05:
> 
> @matlab22 this is due to a new security requirement in Zigbee 3. Note that this only applies to Zigbee 3 devices, meaning that you can have max 40 Zigbee 3 devices in your network AND an unlimited amount of Zigbee 1.2 devices.

How do I know which is connected by 1.2 and which by 3.0? I have devices that claim to be 3.0 connected to your v1.3 firmware. So I assume they are backwards compatible. I have a lot of Xiaomi stuff, do you know which version they are using?

faronov, 2019-06-05:
@Koenkk I only today finishing assemble and tried right now to flash 1.2, but I have an idea to use some security zones and I see that you get some progress about stack 3
for my devices stack 3 more stable than 1.2

Koenkk, 2019-06-06:
@matlab22 it may be listed in the device specification, but currently there is no way of easily knowing this.

kiwinol, 2019-06-06:
I still haven't made the switch to the Zigbee 3 firmware. I am wondering how it compares stability wise on the CC2531 and when you would start recommending that people use this by default over the Zigbee 1.2 firmware :)

santirguez, 2019-06-06:
I migrated my network to Z-Stack 3 mainly because on Z-Stack 1.2 most of my Xiaomi sensors were removed every 1 or 2 days making me to repair them continuously.
In Z-Stack 3 is not happening, or at least that often (I had to repair two sensors after a couple of weeks of use) and it is quite stable.
There are sometimes that the sensors do not respond immediately and they start responding after a couple of minutes, but I still don't know why this is happening as I don't see anything in logs,

kiwinol, 2019-06-06:
Thanks for sharing your experience. I am finding my setup is pretty stable at the moment, except the the latest firmware, I had to roll back to 20190223 firmware.

marc-gist, 2019-06-14:
> I migrated my network to Z-Stack 3 mainly because on Z-Stack 1.2 most of my Xiaomi sensors were removed every 1 or 2 days making me to repair them continuously.
> In Z-Stack 3 is not happening, or at [...]

Have you separated out your WiFi channel (on all APs) from the Zigbee one?  I have over 20 Xiaomi sensors, never drop now that I separated out my 2.4Ghz channels to ensure no interference with Zigbee radio.  I'm on firmware v1.2 20190223

I have some samsung and other Zigbee devices as well on the network. all working great!  I have two routers, CC2531 and a CC2530 external antenna. and the Samsung plug and one Light bulb show as routers as well.

So i'm afraid to change ... i'll setup a new dev environment perhaps and move a few sensors to it.

valexi7, 2019-07-06:
Everytime I need to restart zigbee2mqtt service my network is messed up and I need to reset bulbs to get them working again. It is annoying... CC2531 with Z-Stack 3.0 firmware and latest zigbee2mqtt dev git channel...

dzungpv, 2019-07-24:
I have this error with CC2530+cc2591 firmware  CC2530_CC2591_20190523.zip:

```
	{"message":"Cannot read property 'zclClustersReq' of null","stack":"TypeError: Cannot read property 'zclClustersReq' of null\n    at /app/node_modules/zigbee-herdsman/dist/lib/components/event_handlers.js:243:32\n    at _fulfilled (/app/node_modules/zigbee-herdsman/node_modules/q/q.js:854:54)\n    at /app/node_modules/zigbee-herdsman/node_modules/q/q.js:883:30\n    at Promise.promise.promiseDispatch (/app/node_modules/zigbee-herdsman/node_modules/q/q.js:816:13)\n    at /app/node_modules/zigbee-herdsman/node_modules/q/q.js:877:14\n    at runSingle (/app/node_modules/zigbee-herdsman/node_modules/q/q.js:137:13)\n    at flush (/app/node_modules/zigbee-herdsman/node_modules/q/q.js:125:13)\n    at process._tickCallback (internal/process/next_tick.js:61:11)"}
  zigbee2mqtt:error 7/24/2019, 6:09:45 PM 
	{"message":"Cannot read property 'zclClustersReq' of null","stack":"TypeError: Cannot read property 'zclClustersReq' of null\n    at /app/node_modules/zigbee-herdsman/dist/lib/components/event_handlers.js:243:32\n    at _fulfilled (/app/node_modules/zigbee-herdsman/node_modules/q/q.js:854:54)\n    at /app/node_modules/zigbee-herdsman/node_modules/q/q.js:883:30\n    at Promise.promise.promiseDispatch (/app/node_modules/zigbee-herdsman/node_modules/q/q.js:816:13)\n    at /app/node_modules/zigbee-herdsman/node_modules/q/q.js:877:14\n    at runSingle (/app/node_modules/zigbee-herdsman/node_modules/q/q.js:137:13)\n    at flush (/app/node_modules/zigbee-herdsman/node_modules/q/q.js:125:13)\n    at process._tickCallback (internal/process/next_tick.js:61:11)"}
```
The reason i change to zigbee 3.0 because some not solved bug with CC2630/cc2650 end devices (in side TI stack) with router

dzungpv, 2019-08-02:
@Koenkk How to make Zigbee 3.0 more stable? CC2652R1 have some hardware bug but CC2530/2531 good enough,  I will invest time to it if you need help

Koenkk, 2019-08-02:
@dzungpv I'm currently still investigating the CC2652R issues. CC2538 might also be an option #1568

dzungpv, 2019-08-02:
@Koenkk I only try with CC2530/2531, but i have some problem with z2mqtt, every time restart z2mqtt it will start pairing from the start, so it is about z2mqtt? I read some one use Ztool solution to start the Z 3.0 network and it work stable.

Koenkk, 2019-08-02:
@dzungpv you mean that all your pairing are lost?

dzungpv, 2019-08-02:
@Koenkk Yes, and it start re-pairing all the devices, I setup test network with only 1 router and two end devices

Koenkk, 2019-08-03:
@dzungpv strange, I currently use the CC2531 + zigbee 3 in production and it's working reliable for 2 months now. 

Probably it zigbee-herdsman detect that the configuration is not up-to-date and therefore reinitialises the device, can you provide the log when running with `DEBUG=* npm start`?

dzungpv, 2019-08-09:
@Koenkk I connect 22 device in the network, 2 router and it start error when reboot/start:
```
  zigbee2mqtt:error 8/9/2019, 8:08:25 PM 
	{"message":"Cannot read property 'zclClustersReq' of null","stack":"TypeError: Cannot read property 'zclClustersReq' of null\n    at /zigbee2mqtt-1.5.1/node_modules/zigbee-herdsman/dist/lib/components/event_handlers.js:243:32\n    at _fulfilled (/zigbee2mqtt-1.5.1/node_modules/zigbee-herdsman/node_modules/q/q.js:854:54)\n    at /zigbee2mqtt-1.5.1/node_modules/zigbee-herdsman/node_modules/q/q.js:883:30\n    at Promise.promise.promiseDispatch (/zigbee2mqtt-1.5.1/node_modules/zigbee-herdsman/node_modules/q/q.js:816:13)\n    at /zigbee2mqtt-1.5.1/node_modules/zigbee-herdsman/node_modules/q/q.js:877:14\n    at runSingle (/zigbee2mqtt-1.5.1/node_modules/zigbee-herdsman/node_modules/q/q.js:137:13)\n    at flush (/zigbee2mqtt-1.5.1/node_modules/zigbee-herdsman/node_modules/q/q.js:125:13)\n    at process._tickCallback (internal/process/next_tick.js:61:11)"}
```

dzungpv, 2019-08-10:
@Koenkk never mind, i delete all and start over, now it work fine for 48 hours, 2 CC2530 Z-Stack 3 router with 24 end devices on CC2530+CC2591, include those Xiaomi sensors: 2 motion, 3 emperature, 1 door.  I will report back it not stable.

choeflake, 2019-08-28:
@Koenkk I could not find any note about a 40 device limit in Zigbee 3. Is there a device limit in combination with the CC2530/31 and Zigbee 3 or general for Zigbee 3?
It's hard for me to believe that is is a general device limit in Zigbee 3.

Koenkk, 2019-08-29:
Only in combination with the cc2531/cc2530.

[72 more replies]

closed 2020-11-10 by github-actions[bot]