Koenkk/zigbee2mqtt #1446: Sunricher DIM Lighting

rotda, 2019-04-18:
Hi ,

I have an Sunricher DIM Lighting device .
Bought in Germany :
https://www.iluminize.com/de/shop/led-steuerung/led-controller/product/520-511-012-zigbee-controller-8a.html

I have added the device in devices.js
```

{
    zigbeeModel: ['DIM Lighting'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '511.10', // Vendor model number, look on the device for a model number
    vendor: 'iluminize', // Vendor of the device (only used for documentation and startup logging)
    description: 'LED ', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    supports: 'on/off', // Actions this device supports (only used for documentation)
    fromZigbee: [fz.light_brightness, fz.light_state],
    toZigbee: [tz.on_off, tz.light_brightness, tz.ignore_transition, tz.light_alert],

}

```

But it's send no informations to zigbee2mqtt :-(

Database.json

```
{
	"id": 3,
	"type": "Router",
	"ieeeAddr": "0x000d6f001374850c",
	"nwkAddr": 49999,
	"manufId": 4107,
	"manufName": "Sunricher",
	"powerSource": "Mains (single phase)",
	"modelId": "DIM Lighting",
	"epList": [1, 242],
	"status": "offline",
	"joinTime": null,
	"endpoints": {
		"1": {
			"profId": 260,
			"epId": 1,
			"devId": 257,
			"inClusterList": [0, 3, 4, 5, 6, 8, 2821, 4096],
			"outClusterList": [25, 4096],
			"clusters": {
				"genBasic": {
					"dir": {
						"value": 1
					},
					"attrs": {
						"9": 255,
						"10": {},
						"11": "NULL",
						"30721": 255,
						"30722": 255,
						"30723": 255,
						"30724": 255,
						"36864": 2,
						"65533": 1,
						"zclVersion": 3,
						"appVersion": 0,
						"stackVersion": 0,
						"hwVersion": 0,
						"manufacturerName": "Sunricher",
						"modelId": "DIM Lighting",
						"dateCode": "NULL",
						"powerSource": 1,
						"appProfileVersion": 255,
						"swBuildId": "2.4.1_r30"
					}
				},
				"genIdentify": {
					"dir": {
						"value": 1
					},
					"attrs": {
						"65533": 1,
						"identifyTime": 0
					}
				},
				"genGroups": {
					"dir": {
						"value": 1
					},
					"attrs": {
						"65533": 1,
						"nameSupport": 0
					}
				},
				"genScenes": {
					"dir": {
						"value": 1
					},
					"attrs": {
						"65533": 1,
						"count": 0,
						"currentScene": 0,
						"currentGroup": 0,
						"sceneValid": 0,
						"nameSupport": 0
					}
				},
				"genOnOff": {
					"dir": {
						"value": 1
					},
					"attrs": {
						"65533": 1,
						"onOff": 1,
						"globalSceneCtrl": 1,
						"onTime": 0,
						"offWaitTime": 0
					}
				},
				"genLevelCtrl": {
					"dir": {
						"value": 1
					},
					"attrs": {
						"15": 1,
						"16384": 255,
						"65533": 1,
						"currentLevel": 254,
						"remainingTime": 0,
						"onOffTransitionTime": 1,
						"onLevel": 255
					}
				},
				"genOta": {
					"dir": {
						"value": 2
					},
					"attrs": {}
				},
				"haDiagnostic": {
					"dir": {
						"value": 1
					},
					"attrs": {
						"65533": 1,
						"numberOfResets": 0,
						"averageMacRetryPerApsMessageSent": 0,
						"lastMessageLqi": 0,
						"lastMessageRssi": 0
					}
				},
				"lightLink": {
					"dir": {
						"value": 3
					},
					"attrs": {
						"65533": 1
					}
				}
			}
		},
		"242": {
			"profId": 41440,
			"epId": 242,
			"devId": 102,
			"inClusterList": [33],
			"outClusterList": [33],
			"clusters": {
				"genGreenPowerProxy": {
					"dir": {
						"value": 3
					},
					"attrs": {}
				}
			}
		}
	},
	"_id": "nMrodiX4LQXCMcBC"
}

```

Koenkk, 2019-04-19:
Are you able to control the device?

rotda, 2019-04-19:
No Sorry :
  zigbee2mqtt:error 4/19/2019, 8:47:21 AM Zigbee publish to device '0x000d6f001374850c', genOnOff - {state: on} - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: Unrecognized command

Command:

mosquitto_pub  -h 127.0.0.1 -m "{"state": "ON"}" -t zigbee2mqtt/0x000d6f001374850c/set

Koenkk, 2019-04-19:
Try with `mosquitto_pub -h 127.0.0.1 -m "ON" -t zigbee2mqtt/0x000d6f001374850c/set`

rotda, 2019-04-19:
Thx , but now

```
 zigbee2mqtt:info 4/19/2019, 8:48:42 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: Unrecognized command","meta":{"entity":{"ID":"0x000d6f001374850c","type":"device","friendlyName":"0x000d6f001374850c"},"message":"{state: ON}"}}'
  zigbee2mqtt:info 4/19/2019, 9:04:20 AM Zigbee publish to device '0x000d6f001374850c', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 4/19/2019, 9:04:20 AM MQTT publish: topic 'zigbee2mqtt/0x000d6f001374850c', payload '{"state":"ON"}'
/opt/zigbee2mqtt/lib/extension/deviceReceive.js:109
                if (c.cid === cid) {
                      ^

TypeError: Cannot read property 'cid' of undefined
    at mappedDevice.fromZigbee.filter (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:109:23)
    at Array.filter (<anonymous>)
    at DeviceReceive.onZigbeeMessage (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:105:52)
    at extensions.filter.forEach (/opt/zigbee2mqtt/lib/controller.js:145:31)
    at Array.forEach (<anonymous>)
    at Controller.onZigbeeMessage (/opt/zigbee2mqtt/lib/controller.js:145:14)
    at Zigbee.onMessage (/opt/zigbee2mqtt/lib/zigbee.js:224:18)
    at ZShepherd.emit (events.js:189:13)
    at ZShepherd.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/shepherd.js:192:14)
    at ZShepherd.emit (events.js:189:13)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.3.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.3.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-04-19T07_04_21_158Z-debug.log

```

Koenkk, 2019-04-19:
can you post your `devices.js` file?

rotda, 2019-04-19:
[devices.txt](https://github.com/Koenkk/zigbee2mqtt/files/3097318/devices.txt)

rotda, 2019-04-19:
So now a little bit further to an solution , i can now switch on/off but still some errors

```
  zigbee2mqtt:info 4/19/2019, 6:14:42 PM MQTT publish: topic 'zigbee2mqtt/0x000d6f001374850c', payload '{"state":"ON","linkquality":0}'
  zigbee2mqtt:warn 4/19/2019, 6:14:42 PM No converter available for '511.10' with cid 'genOnOff', type 'devChange' and data '{"cid":"genOnOff","data":{"onOff":1}}'
  zigbee2mqtt:warn 4/19/2019, 6:14:42 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 4/19/2019, 6:14:42 PM No converter available for '511.10' with cid 'genOnOff', type 'devChange' and data '{"cid":"genOnOff","data":{"onOff":1}}'
```

Koenkk, 2019-04-19:
Please try with:

```js
{
    zigbeeModel: ['DIM Lighting'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '511.10', // Vendor model number, look on the device for a model number
    vendor: 'iluminize', // Vendor of the device (only used for documentation and startup logging)
    description: 'LED ', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    extend: generic.light_onoff_brightness,

},
```

rotda, 2019-04-25:
Hi thx this work. Can add this also to the device.js for  the future ?

Koenkk, 2019-04-26:
Yes, it will be supported in the dev branch in a few hours.

closed 2019-04-26 by Koenkk

johnnychicago, 2019-12-09:
Hello --
I have a 511.012, which seems closely related. I added following to devices.js:

`    {                
        zigbeeModel: ['511.012'],
        model: '511.012',
        vendor: 'Iluminize',
        description: 'Zigbee LED-Controller ',
        extend: generic.light_onoff_brightness,
    },`
and I can switch the dimmer by publishing to the respective MQTT topic as mentioned earlier in the thread. However the device does not seem to find its way into my home assistant instance.
Can I ask you to add the 511.012 to devices.js, and would you by chance know where I should look to understand why it's not coming up in home assistant?

Koenkk, 2019-12-11:
@johnnychicago the reason that it's not showing is that it is not defined in `homeassistant.js`. Anyway I've just added the device and it will be supported in a few hours in the dev branch (together with home assistant discovery).

johnnychicago, 2019-12-12:
Thanks a million. I've just pulled the latest-dev docker, and it seems not yet in. Will try again these days.

Koenkk, 2019-12-12:
Working on it right now, should work within 2 hours.

johnnychicago, 2019-12-13:
Works like a charm. Thanks a lot!