Koenkk/zigbee2mqtt #1447: Raw mode for Ikea ICTC-G-1

milakov, 2019-04-18:
Hi,
Support for Ikea ICTC-G-1 hides all raw packets from the device, the device basically looks like the value from 0 to 255. While this is very covenient in some cases, it is limiting in others. In particular, if the dimmer is used to adjust the brightness of the lamp, but dimmer is not the only way to adjust that brightness, then the value of the dimmer can easily get out of sync with the brightness of the lamp, this value cannot be used any longer.

The solution would be to allow (optionally) switch support for Ikea ICTC-G-1 to the "raw" mode, where all the packets from the device are reported as is to mqtt. No "brightness" in this mode. Thus it would be easy to have an automation triggered by each such message and adjusting the brightness of the lamp accordingly.

Koenkk, 2019-04-19:
This could be accomplished relatively easy, just set the `payload.action` depending on the code before (https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L123).

I think this can always be send, so no extra option is needed.

milakov, 2019-04-19:
payload.action is probably fine already.

The messages are now accumulated - debounce(250) at https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L89, so if I rotate the dimmer and stop doing it within 1/4 of a second then all I get is rotate_stop, no info on the direction the rotation happened.
And frequency 1/4 doesn't allow responsive automations.

Would it be possible to remove this debounce altogether or make it configurable for this device the same way as for temperature/humidity xiaomi sensors?

Koenkk, 2019-04-19:
To skip the debounce, refactor e.g. https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L1614   to 

```js
cmd_move: {
        cid: 'genLevelCtrl',
        type: 'cmdMove',
        convert: (model, msg, publish, options) => {
                 ictcg1(model, msg, publish, options, 'move');
                 const direction = msg.data.data.movemode === 1 ? 'left' : 'right';
                 return {action: `rotate_${direction}`};
         },
    },
```

Something similar needs to be done for all converters till https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L1634

milakov, 2019-04-19:
I can try that. Do you want me to create a PR afterwards? Some information is still lost this way, for example the speed of rotation or whether it was a quick rotation.

Koenkk, 2019-04-19:
Yes, please make a pr

Koenkk, 2019-04-22:
merged, thanks!

closed 2019-04-22 by Koenkk