Koenkk/zigbee2mqtt #1449: "unknown" device in log without corresponding entry in configuration.yaml

aktentasche, 2019-04-19:
# Bug Report
## What happened
My debug log shows "unknown" devices, more specifically Hue RGB lamps. I own a total of four of them, those are listed in my configuration.yaml

The "unknown" devices have ieeeAdr that are different from my known ones. I have no clue where they come from, it could be a neighbours lamp.

## What did you expect to happen
If a device is not listed in the configuration.yaml it should not appear in any part of the software (even if it just the log) after a restart of zigbee2mqtt when permit_join is set to false.

## How to reproduce it (minimal and precise)
Unknown

## Debug Info
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3097931/log.txt)

In state.json the "unknown" devices do not appear.

zigbee2mqtt version: 1.3.0 (commit #00ebd44)
CC253X firmware version: CC2531ZNP-Pro-Secure_LinkKeyJoin_mod.hex from So 04 Nov 2018 00∶01∶56 CET (cannot remember the source)

`4/19/2019, 12:37:04 PM - info: hue_rgb_lamp_1 (0x0017880104e3babd): 9290012573A - Philips Hue white and color ambiance E26/E27/E14 (Router)
4/19/2019, 12:37:04 PM - info: hue_rgb_lamp_2 (0x0017880104562b52): 9290012573A - Philips Hue white and color ambiance E26/E27/E14 (Router)
4/19/2019, 12:37:04 PM - info: hue_rgb_lamp_3 (0x00178801039e7d85): 9290012573A - Philips Hue white and color ambiance E26/E27/E14 (Router)
4/19/2019, 12:37:04 PM - info: hue_rgb_lamp_4 (0x0017880104d32b21): 9290012573A - Philips Hue white and color ambiance E26/E27/E14 (Router)
**4/19/2019, 12:37:04 PM - info: unknown (0x001788010284138b): 9290012573A - Philips Hue white and color ambiance E26/E27/E14 (Router)
4/19/2019, 12:37:04 PM - info: unknown (0x0017880102841370): 9290012573A - Philips Hue white and color ambiance E26/E27/E14 (Router)
4/19/2019, 12:37:04 PM - info: unknown (0x001788010284137d): 9290012573A - Philips Hue white and color ambiance E26/E27/E14 (Router**)`

Koenkk, 2019-04-19:
The actual list is taken from `database.db`, can you post this file?

aktentasche, 2019-04-21:
Ah, I did not know. I see there are way too many entries, probably because I quite often did a repairing. Would it be advisable to maybe empty database.db, backup configuration.yaml and start from the scratch (i.e. repair all devices again)?

[database.txt](https://github.com/Koenkk/zigbee2mqtt/files/3101098/database.txt)

Koenkk, 2019-04-22:
You could either re-pair everything, or just delete the 3 unknown entries from `datbase.db`.

aktentasche, 2019-06-21:
I deleted database.db and started from scratch, closing.

closed 2019-06-21 by aktentasche