Koenkk/zigbee2mqtt #1450: No more Battery info on Xiaomi devices?

Bastiencc, 2019-04-19:
Hi

I remove/re-pair some Xiaomi/aqara devices today, and it seems there is no more battery info in 1.3.1 (it was ok before removing) :

MFKZQ01LM - Xiaomi Mi/Aqara smart home cube (EndDevice) : 
```
Apr 19 16:47:00 raspberrypi npm[2258]:   zigbee2mqtt:info 4/19/2019, 4:47:00 PM MQTT publish: topic 'zigbee2mqtt/aqara_cube', payload '{"side":0,"linkquality":31,"action":"slide"}'
```

WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice) : 
```
Apr 19 16:43:50 raspberrypi npm[2258]:   zigbee2mqtt:info 4/19/2019, 4:43:50 PM MQTT publish: topic 'zigbee2mqtt/cigare_capteur', payload '{"temperature":23.39,"linkquality":92,"humidity":55.39,"pressure":1002}'
```

Bastiencc, 2019-04-20:
ok so, after severals hours, battery info came back on payload...
```
Apr 20 08:23:15 raspberrypi npm[2258]:   zigbee2mqtt:info 4/20/2019, 8:23:15 AM MQTT publish: topic 'zigbee2mqtt/aqara_cube', payload '{"side":0,"linkquality":23,"battery":91,"voltage":2985,"action":"slide"}'
```
Strange , but it's ok now ! Maybe a bad pairing process? 

I close this issue

closed 2019-04-20 by Bastiencc

philipparndt, 2019-04-20:
AFAIK the battery info will only be send from time to time to save battery.
I can see the same behavior on my Xiaomi sensors.