Koenkk/zigbee2mqtt #1452: Can not pair aqara window sensor

Ulli2k, 2019-04-20:
I tried to pair the aqara window contact sensor several times and do not get it paired.
I get the following log entries
`zigbee2mqtt:info 4/20/2019, 8:26:13 AM New device 'undefined' with address 0x00158d0002ecf09c connected!

  zigbee2mqtt:info 4/20/2019, 8:26:13 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00158d0002ecf09c","meta":{}}'

  zigbee2mqtt:warn 4/20/2019, 8:26:13 AM Device with modelID 'undefined' is not supported.

  zigbee2mqtt:warn 4/20/2019, 8:26:13 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

  zigbee2mqtt:warn 4/20/2019, 8:26:13 AM Device with modelID 'undefined' is not supported.

  zigbee2mqtt:warn 4/20/2019, 8:26:13 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

  zigbee2mqtt:warn 4/20/2019, 8:26:20 AM Device with modelID 'undefined' is not supported.

  zigbee2mqtt:warn 4/20/2019, 8:26:20 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

  zigbee2mqtt:warn 4/20/2019, 8:26:20 AM Device with modelID 'undefined' is not supported.

  zigbee2mqtt:warn 4/20/2019, 8:26:20 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

  zigbee2mqtt:warn 4/20/2019, 8:26:34 AM Message without device!

  zigbee2mqtt:warn 4/20/2019, 8:26:51 AM Device with modelID 'undefined' is not supported.

  zigbee2mqtt:warn 4/20/2019, 8:26:51 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

  zigbee2mqtt:warn 4/20/2019, 8:26:59 AM Device with modelID 'undefined' is not supported.

  zigbee2mqtt:warn 4/20/2019, 8:26:59 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html`

I use the following firmware
zigbee2mqtt:info 4/20/2019, 8:33:50 AM Starting zigbee2mqtt version 1.3.0 (commit #unknown)

  zigbee2mqtt:info 4/20/2019, 8:33:50 AM Starting zigbee-shepherd

  zigbee2mqtt:info 4/20/2019, 8:33:52 AM zigbee-shepherd started

  zigbee2mqtt:info 4/20/2019, 8:33:52 AM Coordinator firmware version: '20181024'

Ulli2k, 2019-04-20:
If I press the pairing button just one time I get
`  zigbee2mqtt:info 4/20/2019, 8:36:26 AM MQTT publish: topic 'zigbee2mqtt/0x00158d000312220c', payload '{"contact":false,"linkquality":120}',
  zigbee2mqtt:info 4/20/2019, 8:36:26 AM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.3.0","commit":"unknown","coordinator":20181024,"log_level":"info","permit_join":false}',
  zigbee2mqtt:info 4/20/2019, 8:36:32 AM Zigbee: disabling joining new devices.,
  zigbee2mqtt:info 4/20/2019, 8:36:32 AM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.3.0","commit":"unknown","coordinator":20181024,"log_level":"info","permit_join":false}',
  zigbee2mqtt:info 4/20/2019, 8:36:38 AM Zigbee: allowing new devices to join.,
  zigbee2mqtt:info 4/20/2019, 8:36:38 AM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.3.0","commit":"unknown","coordinator":20181024,"log_level":"info","permit_join":true}',
  zigbee2mqtt:info 4/20/2019, 8:37:37 AM Connecting with device...,
  zigbee2mqtt:info 4/20/2019, 8:37:37 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}',
  zigbee2mqtt:info 4/20/2019, 8:37:41 AM Connecting with device...,
  zigbee2mqtt:info 4/20/2019, 8:37:41 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}',
  zigbee2mqtt:info 4/20/2019, 8:37:44 AM Connecting with device...,
  zigbee2mqtt:info 4/20/2019, 8:37:44 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}',
  zigbee2mqtt:info 4/20/2019, 8:37:47 AM Connecting with device...,
  zigbee2mqtt:info 4/20/2019, 8:37:47 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}',
  zigbee2mqtt:info 4/20/2019, 8:37:50 AM Connecting with device...,
  zigbee2mqtt:info 4/20/2019, 8:37:50 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}',
  zigbee2mqtt:info 4/20/2019, 8:37:50 AM Device incoming...,
  zigbee2mqtt:info 4/20/2019, 8:37:50 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}',
  zigbee2mqtt:info 4/20/2019, 8:37:50 AM New device 'undefined' with address 0x00158d0002ecf09c connected!,
  zigbee2mqtt:info 4/20/2019, 8:37:50 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00158d0002ecf09c","meta":{}}',
  zigbee2mqtt:warn 4/20/2019, 8:37:50 AM Device with modelID 'undefined' is not supported.,
  zigbee2mqtt:warn 4/20/2019, 8:37:50 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html,
  zigbee2mqtt:warn 4/20/2019, 8:37:50 AM Device with modelID 'undefined' is not supported.,
  zigbee2mqtt:warn 4/20/2019, 8:37:50 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html,
  zigbee2mqtt:warn 4/20/2019, 8:38:28 AM Device with modelID 'undefined' is not supported.,
  zigbee2mqtt:warn 4/20/2019, 8:38:28 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html,
  zigbee2mqtt:warn 4/20/2019, 8:38:28 AM Device with modelID 'undefined' is not supported.,
  zigbee2mqtt:warn 4/20/2019, 8:38:28 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html,
  zigbee2mqtt:warn 4/20/2019, 8:39:12 AM Message without device!,
  zigbee2mqtt:info 4/20/2019, 8:40:08 AM Device incoming...,
  zigbee2mqtt:info 4/20/2019, 8:40:08 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}',
  zigbee2mqtt:warn 4/20/2019, 8:40:08 AM Device with modelID 'undefined' is not supported.,
  zigbee2mqtt:warn 4/20/2019, 8:40:08 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html,
  zigbee2mqtt:warn 4/20/2019, 8:40:08 AM Device with modelID 'undefined' is not supported.,
  zigbee2mqtt:warn 4/20/2019, 8:40:08 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html,
  zigbee2mqtt:info 4/20/2019, 8:41:06 AM MQTT publish: topic 'zigbee2mqtt/0x00158d0002e93c1e', payload '{"temperature":25.07,"linkquality":39,"humidity":31.25,"pressure":954,"battery":80,"voltage":2965}',
  zigbee2mqtt:info 4/20/2019, 8:41:06 AM MQTT publish: topic 'zigbee2mqtt/0x00158d0002e93c1e', payload '{"temperature":25.07,"linkquality":39,"humidity":28.86,"pressure":954,"battery":80,"voltage":2965}',
  zigbee2mqtt:info 4/20/2019, 8:41:06 AM MQTT publish: topic 'zigbee2mqtt/0x00158d0002e93c1e', payload '{"temperature":25.07,"linkquality":39,"humidity":28.86,"pressure":954,"battery":80,"voltage":2965}',
Using '/app/data' as data directory,
,
> zigbee2mqtt@1.3.0 start /app,
> node index.js,
,
  zigbee2mqtt:info 4/20/2019, 8:42:41 AM Logging to directory: '/app/data/log/2019-04-20.08-42-41',
  zigbee2mqtt:info 4/20/2019, 8:42:41 AM Starting zigbee2mqtt version 1.3.0 (commit #unknown),
  zigbee2mqtt:info 4/20/2019, 8:42:41 AM Starting zigbee-shepherd,
  zigbee2mqtt:info 4/20/2019, 8:42:43 AM zigbee-shepherd started,
  zigbee2mqtt:info 4/20/2019, 8:42:43 AM Coordinator firmware version: '20181024',
  zigbee2mqtt:info 4/20/2019, 8:42:43 AM Currently 13 devices are joined:,
  zigbee2mqtt:info 4/20/2019, 8:42:43 AM 0x00158d0002e98f0b (0x00158d0002e98f0b): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice),
[... cut ...]
  zigbee2mqtt:info 4/20/2019, 8:44:29 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00158d0002ecf09c","meta":{}}',
  zigbee2mqtt:warn 4/20/2019, 8:44:29 AM Device with modelID 'undefined' is not supported.,
  zigbee2mqtt:warn 4/20/2019, 8:44:29 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html,
  zigbee2mqtt:warn 4/20/2019, 8:44:29 AM Device with modelID 'undefined' is not supported.,
  zigbee2mqtt:warn 4/20/2019, 8:44:29 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html,`

Fabiancrg, 2019-04-20:
I just tried to pair the same device and it worked (latest DEV branch update and FW 20190223).
But after that I tried to pair an Aqara motion sensor and had exactly the same issue.
What I had to do is remove the device, restart zigbee2mqtt and retried, had to do it two or three times to finally had the sensor working correctly.

Here is the log after the first pairing attempt:
````
Apr 20 22:01:56 NUC npm[1158]:   zigbee2mqtt:info 4/20/2019, 10:01:56 PM MQTT publish: topic 'zigbee2mqtt/Kitchen_Smoke', payload '{"battery":100,"voltage":3175,"linkquality":39}'
Apr 20 22:01:56 NUC npm[1158]:   zigbee2mqtt:info 4/20/2019, 10:01:56 PM MQTT publish: topic 'zigbee2mqtt/Stairwell_Light', payload '{"state":"OFF","brightness":200,"color_temp":155,"linkquality":39}'
Apr 20 22:02:22 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:02:22 PM No converter available for 'ZNLDP12LM' with cid 'msOccupancySensing', type 'devChange' and data '{"cid":"msOccupancySensing","data":{"occupancy":1}}'
Apr 20 22:02:22 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:02:22 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Apr 20 22:02:35 NUC npm[1158]:   zigbee2mqtt:info 4/20/2019, 10:02:35 PM New device 'undefined' with address 0x00158d00030aa20d connected!
Apr 20 22:02:35 NUC npm[1158]:   zigbee2mqtt:info 4/20/2019, 10:02:35 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00158d00030aa20d","meta":{}}'
Apr 20 22:02:35 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:02:35 PM Device with modelID 'undefined' is not supported.
Apr 20 22:02:35 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:02:35 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Apr 20 22:02:35 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:02:35 PM Device with modelID 'undefined' is not supported.
Apr 20 22:02:35 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:02:35 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Apr 20 22:02:41 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:02:41 PM Device with modelID 'undefined' is not supported.
Apr 20 22:02:41 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:02:41 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Apr 20 22:02:42 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:02:42 PM Device with modelID 'undefined' is not supported.
Apr 20 22:02:42 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:02:42 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Apr 20 22:02:47 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:02:47 PM Device with modelID 'undefined' is not supported.
Apr 20 22:02:47 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:02:47 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Apr 20 22:02:47 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:02:47 PM Device with modelID 'undefined' is not supported.
Apr 20 22:02:47 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:02:47 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Apr 20 22:02:53 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:02:53 PM Device with modelID 'undefined' is not supported.
Apr 20 22:02:53 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:02:53 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Apr 20 22:02:53 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:02:53 PM Device with modelID 'undefined' is not supported.
Apr 20 22:02:53 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:02:53 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Apr 20 22:02:59 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:02:59 PM Device with modelID 'undefined' is not supported.
Apr 20 22:02:59 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:02:59 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Apr 20 22:03:01 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:03:01 PM Device with modelID 'undefined' is not supported.
Apr 20 22:03:01 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:03:01 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Apr 20 22:03:01 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:03:01 PM Device with modelID 'undefined' is not supported.
Apr 20 22:03:01 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:03:01 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Apr 20 22:03:08 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:03:08 PM Device with modelID 'undefined' is not supported.
Apr 20 22:03:08 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:03:08 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Apr 20 22:03:10 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:03:10 PM Device with modelID 'undefined' is not supported.
Apr 20 22:03:10 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:03:10 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Apr 20 22:03:12 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:03:12 PM Device with modelID 'undefined' is not supported.
Apr 20 22:03:12 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:03:12 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Apr 20 22:03:14 NUC npm[1158]:   zigbee2mqtt:warn 4/20/2019, 10:03:14 PM Device with modelID 'undefined' is not supported.
[... cut ...]
So I deleted it again, retried, same thing, finally I restarted again z2m and it's currently working normally... all this is very unusual but it seems I am having issues with paring since the upgrade to 1.3.x.
I had that with a Bosch motion sensor, impossible to pair, the next day when I retried it worked.

I have also a Heiman Smart remote controller, used to worked very well and suddenly stopped working (probably when the upgrade was done but cannot confirm), so I retried the pairing, it's seen as unknown sometimes, sometimes it's recofnized but I cannot get any status when I press the button. Will open another issue for this one.

Koenkk, 2019-04-22:
The pairing of Xiaomi sensor will be greatly improved in the next release (1.4) (see #1401). 

This has already been implemented in the dev branch (http://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html). Once a Xiaomi sensor is not properly recognized (`undefined `modelid` blablabla) it can easily be fixed by pressing the reset button once.

closed 2019-04-22 by Koenkk