Koenkk/zigbee2mqtt #1453: Delay on Aqara motion sensor

pimseb, 2019-04-20:
I've been using the official Xiaomi gateway until today. I'm now testing zigbee2mqtt.
I have 3 Aqara motion sensor that will be linked later to an alarm. My zigbee2mqtt is linked to domoticz.
When I walk in front of a motion sensor, it does detect a motion and I see in the log that a motion has been detected  (in domoticz aswell). The problem is that it only detects the motion once. If I come again in front of the sensor, zigbee2mqtt won't send an "on" signal again. It seems to be blocked during 1:30 min
This behaviour doesn't come from the Aqara motion sensor unit itself because I don't have this problem with the official Xiaomi getaway. If I pass 3 times in 30 seconds in front of the motion sensor, the xiaomi getaway will send 3 "on" to domoticz.

Sorry for bad english :)
Thanks 

## Debug Info
zigbee2mqtt version: zigbee2mqtt@1.3.1 start /opt/zigbee2mqtt
CC253X firmware version: CC2531_DEFAULT_20190223

timants, 2019-04-20:
i always had this problem with my aqara motion sensors , seems like it goes in sleep after it detects motion to save battery, (i even had this with the original xiaomi hub) , i hope someone has a solution to this , becouse they are useless with this behavoir .  i sub to this topic to follow it .

aktentasche, 2019-04-21:
Maybe OT, but I think technology-wise PIR sensors can only detect a movement every 15s or so because they need to settle again. 90 seconds is too long though that is clear.

md-reddevil, 2019-04-22:
Same issue here.

FixxCZ, 2019-04-22:
Same issue, Even when an device specific config occupancy_timeout is set for few seconds it still have that 90 seconds cool down. Very annoying.

andreasbrett, 2019-04-22:
Unfortunately this is normal with battery-driven PIR sensors. Consider yourself walking around in a room with a PIR sensor. It would be very power consuming for the device to send each and every move of yours via zigbee radio to the network. The cooldown is the only way to have this sensor's battery last some time.

It's a typical trade off. There's power consumption and precision. Can't have low power consumption AND precision. It's either or. There are zigbee PIR sensors that are wall-plugged out there which would give you the precision you're all asking for.

pajikos, 2019-04-29:
Has anyone tried to set occupancy_timeout? RTCGQ01LM, RTCGQ11LM seems to be aqara devices, so if the documentation is correct, you should be able to this timeout individually: https://www.zigbee2mqtt.io/configuration/device_specific_configuration.html

aXionDevelopment, 2019-04-29:
> 
> 
> Has anyone tried to set occupancy_timeout? 

Just tried it on a RTCGQ11LM with a timeout of 20 seconds. Seems to work fine!

Koenkk, 2019-04-29:
@aXionDevelopment a timeout of 20 seconds makes no sense, as the device only reports the occupancy every 60 seconds. Meaning that for 40 seconds you won't get occupancy.

aXionDevelopment, 2019-04-29:
@Koenkk , yep, you're completely right. I was playing with the device and just noticed that.

aXionDevelopment, 2019-04-29:
I guess that's why the default timeout value (false message) is set to 90 seconds, so the sensor is able to re-trigger (after 60 sec) if the room is still occupied, and if it detects nothing for 30 more seconds, an "occupancy: false" message will be sent. I believe that's a good enough trade-off.

pimseb, 2019-04-29:
When I use the official Xiaomi getaway I have this kind of log :
[https://i.ibb.co/gmdV1pg/Screenshot-20190429-203616-01.jpg](url)
As you see the getaway receives the motion information many times in one minute.
In domoticz I can setup an auto off 5 seconds delay so at 20:36 I see 5 ON and 5 0FF in domoticz logs.

When I use zigbee2mqtt my logs look like this :

Koenkk, 2019-04-29:
@pimseb this is only right after pairing, see https://community.smartthings.com/t/making-xiaomi-motion-sensor-a-super-motion-sensor/139806

CypherMK, 2019-05-02:
Wanted to ask the same question as OP. But came to the same conclusions:

- After initial motion detection, you have to wait at least 60 seconds for the next motion detection.
- During the first 60 seonds, no motion will be detected, even with "no_occupancy_since" option.
- You can use the "occupancy_timeout" to increase the default 90 seconds window, before the 
   "occupancy: false" message is given. Gives you a longer period after the initial 60 seconds.

So there is no other option, to change this 60 second limitation, except hardware modding?

md-reddevil, 2019-05-02:
As @pimseb mentioned, if the Motion Sensor is connected to XIAOMI Gateway motion information is updated multiple times per minute. If this is the case then the problem is not in the hardware but is in the integration with zigbee2mqtt.

Koenkk, 2019-05-02:
@md-reddevil this is also the case with zigbee2mqtt, but only right after pairing (same for the xiaomi gateway)https://github.com/Koenkk/zigbee2mqtt/issues/1453#issuecomment-487705315

closed 2019-07-08 by stale[bot]

RestOp, 2019-10-01:
> @pimseb this is only right after pairing, see https://community.smartthings.com/t/making-xiaomi-motion-sensor-a-super-motion-sensor/139806

Hello,

first of all, thanks a lot for zigbee2mqtt))) Really Great project!
I would like to ask what have г meant with this post? That г need to modify it to get 5 sec delay?