Koenkk/zigbee2mqtt #1454: Xiaomi Switch QBKG03LM - Failed to ping

VivantSenior, 2019-04-21:
# Bug Report
## What happened
Hello,

I'm unable to use Xiaomi Wired Wall Switch (version without neutral line) - QBKG03LM. After successful pairing I can see in the logs:
> Ping 0x00158...(basic)
> Failed to ping 0x00158...
Detailed logs: https://pastebin.com/raw/HDnBUssU.

@Koenkk created similar issue there: #943 but not the same.

Thank you for any advice,
Vivant

## What did you expect to happen
Xiaomi Switch should show its current state in Home Assistant and respond on requested state change.

## How to reproduce it (minimal and precise)
Pair Xiaomi Wired Wall Switch (QBKG03LM) device.

## Debug Info
zigbee2mqtt version: 1.3.1
CC2531 firmware version: 20190223 (default)
Hass.io version: 0.91.4
Hass.io supervisor: 160

Koenkk, 2019-04-22:
Are you also unable to turn it on/off? What error is logged in that case?

VivantSenior, 2019-04-22:
@Koenkk, yes, I'm unable to turn it on or off. See the logs:

> error: Zigbee publish to device '0x001...', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - 2 failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
> info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.","meta":{"entity":{"ID":"0x001...","type":"device","friendlyName":"Bedroom_wall_switch"},"message":"OFF"}}'

I've just removed this wall switch with the idea of repairing it with zigbee2mqtt but I'm having trouble with full removal of the device from Home Assistant (hass.io).

Koenkk, 2019-04-22:
It doesn't have to be removed from home assisant, can you try re-pairing it?

VivantSenior, 2019-04-22:
After re-pairing (at least trying to) I have an error:
> Cannot get the Node Descriptor of the Device: 0x001... (Error: Timed out after 10000 ms)

Koenkk, 2019-04-23:
Can you restart zigbee2mqtt and retry pairing?

EmmanuelLM, 2019-04-24:
Hello, I am having a similar "failed to ping" error after (say) 24-48h of no use.
To re-pair it, this is what I do:
- allow joining in zigbee2mqtt
- remove the device via the mqtt command
- after that the device will automatically re-pair

VivantSenior, 2019-04-24:
> Can you restart zigbee2mqtt and retry pairing?

As far as I remember, I have restarted zigbee2mqtt and whole hass.io instance before trying to re-pair. During the weekend will try to do that again.

JSCarter1, 2019-04-25:
I also experienced this issue with the same switch.
I assumed it was something to do with the switch sleeping (presumably to reduce power consumption?).

I added a home assistant automation that calls zigbee2mqtt/<device_id>/get every 15 minutes - since then I haven't had any issues. Other things were changed at around the same time, so this may not have been the thing that fixed it.

Koenkk, 2019-04-26:
For those still having this issue, can you check if this has been fixed with the latest dev firmware? https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/Z-Stack_Home_1.2/bin

Koenkk, 2019-04-26:
Please test with the latest dev branch, things should have been improved (https://github.com/Koenkk/zigbee2mqtt/issues/943#issuecomment-487172996). If you have any issues, please report there.

closed 2019-04-26 by Koenkk

VivantSenior, 2019-04-26:
> For those still having this issue, can you check if this has been fixed with the latest dev firmware? https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/Z-Stack_Home_1.2/bin

Unfortunately still receiving an error:
> --REPAIRING--
> **debug: PM Ping 0x001... (basic)**
> **error: Failed to ping 0x001...**
> --REMOVING WALL SWITCH--
> debug: Received MQTT message on 'zigbee2mqtt/bridge/config/remove' with data '0x001...'
> warn: Failed to remove '0x001...', trying force remove...
> info: Force removed 0x001...
> info: Successfully removed 0x001...
> info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"0x001..."}'
> --REPAIRING--
> **error: Cannot get the Node Descriptor of the Device: 0x001... (Error: Timed out after 10000 ms)**

Koenkk, 2019-04-26:
After the removal, can you restart zigbee2mqtt and make sure to pair the switch close to the coordinator?

VivantSenior, 2019-04-27:
I'm still receiving the "Failed to ping" error: https://pastebin.com/raw/cs3rztEM.
I think there are two issues:
- unable to properly remove old instance of the wall switch (error "Cannot get the Node Descriptor of the Device" after trying to pair the device again after the removal by using mqtt message: [zigbee2mqtt/bridge/config/remove](https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html)
- "Failed to ping" error.

Today I'll mount two new Xiaomi Wall Switches (QBKG03LM) next to the coordinator and see if any issue will appear.

VivantSenior, 2019-04-28:
I successfully paired different Xiaomi Wall Switch (QBKG03LM): https://pastebin.com/raw/qMmdEANn but was still receiving "Failed to ping" on old switch.

--- (later that day) ---

I have connected old Xiaomi wall switch closer to the coordinator, removed it from zigbee2mqtt (using mqtt message), paired it again, disconnected it physically from the wall, moved to the old place and now it's working...

Will close the issue.

Thanks,
Viv

EmmanuelLM, 2019-04-28:
Not so fast, sadly mine is still showing ping errors... And it is
permanently set up about 2m from the coordinator...

VivantSenior, 2019-04-29:
Weird, over the night I had few "Failed to ping" on all Xiaomi Wall Switches (near and far from the coordinator). When I turn on debug log level, shepard debug set to true and restart zigbee2mqtt I can't see any "Failed to ping", just "Successfully pinged". I will monitor the logs...

VivantSenior, 2019-04-30:
No "Failed to ping" over the night. But I have found weird bug. When I connect the load only on one of the button (second button had connected cables to it but on the other end of the cable there was nothing) the light connected to first button started to flickers for a split of the second even when I didn't turn it on or off. It repeated every minute or two.

EmmanuelLM, 2019-04-30:
That is interesting because in both of the switches, I only have one line
connected (L2 I think but would have to check) - could it be that the
switch needs L1 connected to draw power (or even both)?

What's your setup (if I recall, you have several)?

VivantSenior, 2019-04-30:
Yeah, I have three of them. Two of them are in the same room. I have all lines connected but on line didn't had the load on it. After attaching simple bulb flickering stopped.

My idea was that maybe response on a ping from HA caused the flickering (drawing power). I don't know if this is more of a firmware issue than zigbee2mqtt...

Koenkk, 2019-04-30:
@VivantSenior it seems to be a hardware design issue: https://github.com/Koenkk/zigbee2mqtt/issues/943#issuecomment-485510942 https://github.com/Koenkk/zigbee2mqtt/issues/943#issuecomment-486362505

VivantSenior, 2019-05-13:
[image]

@Koenkk, https://pastebin.com/raw/Ji72Rkfp. Do you need debug info?

Koenkk, 2019-05-14:
Could you sniff the zigbee traffic when failed to ping happens? (http://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

VivantSenior, 2019-05-30:
```
zigbee2mqtt:debug 5/30/2019, 12:45:52 PM Ping 0x00...1 (basic)
zigbee2mqtt:debug 5/30/2019, 12:45:52 PM Ping 0x00...2 (basic)
zigbee2mqtt:debug 5/30/2019, 12:45:52 PM Ping 0x00...3 (basic)
```

I'm a bit confused right now... I have restarted Zigbee2Mqtt HA addon several times (to allow join few new battery devices). I will monitor the logs for the "Failed to ping" error.

Weird part is that despite the error I was able turn on/off switches remotely.