Koenkk/zigbee2mqtt #1455: How to disable the extra APS: Ack

he902, 2019-04-21:
With zigbee sniffing, I noticed an extra APS: Ack responded from my Yunmi Lock device for each Request I sent to it.

How to disable such extra APS: Ack via setting:
`Acknowledgement Request: False` in the `Frame Control Field` in the APS layer?

[image]

he902, 2019-04-21:
With the Yunmi vendor's coordinator (gateway), there is no such extra APS: Ack responded from my Yunmi Lock device:

You can see, there is:
`Acknowledgement Request: False` in the `Frame Control Field` in the APS layer:

[image]

Koenkk, 2019-04-22:
I've found some documentation here: http://weber.itn.liu.se/~qinye29/tne090/Z-Stack%20Developer's%20Guide.pdf (8. End-to-end acknowledgements)

Currently zigbee-shepherd doesn't allow to pass this option externally. However you should be able to achieve this by changing https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/af.js#L787 to `    var afOptions = ZSC.AF.options.DISCV_ROUTE;    // ACK_REQUEST (0x10), DISCV_ROUTE (0x20)`

Why do you require this?

he902, 2019-04-22:
Thanks @Koenkk, it works!

The reason I require this is that I'm trying to fully simulate the zigbee conversation between Yunmi vendor's coordinator and the Lock device.

The root cause is under investigation for why the PIN code sent to the Lock device is failed.
Another action I'm taking is to clone the MAC address, see discussion [here](https://github.com/Koenkk/Z-Stack-firmware/issues/72) (need your help too).

closed 2019-04-29 by he902