Koenkk/zigbee2mqtt #1456: Updated to latest Dev version and unable to start.

GForce2010, 2019-04-21:
I've just updated to the latest version like I've done several times but now I am unable to start Zigbee2MQTT

I just get the following error.

```
pi@raspberrypi1:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.3.1 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 4/21/2019, 5:29:21 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-04-21.17-29-21'
internal/modules/cjs/loader.js:584
    throw err;
    ^

Error: Cannot find module 'unpi'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:582:15)
    at Function.Module._load (internal/modules/cjs/loader.js:508:25)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:22:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:6:12)
    at Module._compile (internal/modules/cjs/loader.js:701:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:712:10)
    at Module.load (internal/modules/cjs/loader.js:600:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:539:12)
    at Function.Module._load (internal/modules/cjs/loader.js:531:3)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.3.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.3.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-04-21T16_29_22_171Z-debug.log
```

aktentasche, 2019-04-21:
Try running

`npm install -g unpi`

in any location

GForce2010, 2019-04-22:
```
pi@raspberrypi1:~ $ npm install -g unpi
npm WARN checkPermissions Missing write access to /usr/lib/node_modules
npm WARN bl@0.7.0 requires a peer of stream-browserify@* but none is installed. You must install peer dependencies yourself.

npm ERR! path /usr/lib/node_modules
npm ERR! code EACCES
npm ERR! errno -13
npm ERR! syscall access
npm ERR! Error: EACCES: permission denied, access '/usr/lib/node_modules'
npm ERR!  { [Error: EACCES: permission denied, access '/usr/lib/node_modules']
npm ERR!   stack:
npm ERR!    'Error: EACCES: permission denied, access \'/usr/lib/node_modules\'',
npm ERR!   errno: -13,
npm ERR!   code: 'EACCES',
npm ERR!   syscall: 'access',
npm ERR!   path: '/usr/lib/node_modules' }
npm ERR!
npm ERR! The operation was rejected by your operating system.
npm ERR! It is likely you do not have the permissions to access this file as the current user
npm ERR!
npm ERR! If you believe this might be a permissions issue, please double-check the
npm ERR! permissions of the file and its containing directories, or try running
npm ERR! the command again as root/Administrator (though this is not recommended).

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-04-22T11_18_58_701Z-debug.log

```

Koenkk, 2019-04-22:
In the zigbee2mqtt folder, do `rm -rf node_modules && npm install`.

GForce2010, 2019-04-22:
still getting the same error. If I switch to the master branch it runs fine again

Koenkk, 2019-04-23:
What is the output of `ls node_modules` ?

closed 2019-04-29 by GForce2010

httpedo13, 2019-05-11:
I have the same error, but I use Hass.io and zigbee2mqtt add on

GForce2010, 2019-05-11:
In the end i removed the dev version and installed the latest master release as that had all the features i wanted.

httpedo13, 2019-05-11:
i have the stable version and I had this error:
[image]

remixyz, 2019-05-11:
Same Error here! Just reinstalled my Pi (switch to SSD), and now i get this error. Yesterday with the old release, everything worked fine.

remixyz, 2019-05-13:
"Fixed" this error by dowgrading to 1.2.1...