Koenkk/zigbee2mqtt #1457: Empty click message for Xiaomi WXKG02LM wall switch

aXionDevelopment, 2019-04-21:
I was having issues with re-pairing my two WXKG02LM switches but after upgrading to the latest zigbee2MQTT dev version, at least one of them can be paired again without issues. Only it's sending two MQTT messages after a click action, where the last message shows an empty click parameter. 

log details: 

```
> info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.3.1","commit":"69e9dbc","coordinator":20190223,"log_level":"debug","permit_join":true}'
> 
>  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":6,"srcaddr":53636,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":92,"securityuse":0,"timestamp":9717277,"transseqnumber":0,"len":11,"data":{"type":"Buffer","data":[24,119,10,0,0,16,1,0,0,16,0]}} +0ms
>   zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +10s
>   zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":6,"srcaddr":53636,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":92,"securityuse":0,"timestamp":9717277,"transseqnumber":0,"len":11,"data":{"0":24,"1":119,"2":10,"3":0,"4":0,"5":16,"6":1,"7":0,"8":0,"9":16,"10":0},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":119,"cmdId":"report","payload":[{"attrId":0,"dataType":16,"attrData":1},{"attrId":0,"dataType":16,"attrData":0}]}} +33ms
>   zigbee2mqtt:debug 4/21/2019, 6:49:38 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'lumi.sensor_86sw2Un' (0x00158d000254ce3a) of endpoint 1
>   zigbee2mqtt:info 4/21/2019, 6:49:38 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000254ce3a', payload '{"linkquality":92,"battery":100,"voltage":3035,"click":"left"}'
>   zigbee2mqtt:info 4/21/2019, 6:49:38 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000254ce3a', payload '{"linkquality":92,"battery":100,"voltage":3035,"click":""}'
>   zigbee-shepherd:request REQ --> ZDO:bindReq +10s
>   zigbee2mqtt:debug 4/21/2019, 6:49:39 PM Received MQTT message on 'zigbee2mqtt/bridge/config/devices' with data '1555865379506'
```

Tried to re-pair but that didn't solve the issue.

 Any help is appreciated

neilvangeffen, 2019-04-22:
Thats normal right? It just changes the click to left and then clears it again right there after...

`  zigbee2mqtt:info 4/22/2019, 5:51:44 PM MQTT publish: topic 'zigbee2mqtt/Hallway Switch', payload '{"linkquality":115,"state_left":"ON","state_right":"OFF"}'
  zigbee2mqtt:info 4/22/2019, 5:51:45 PM MQTT publish: topic 'zigbee2mqtt/Hallway Switch', payload '{"linkquality":115,"state_left":"ON","state_right":"OFF","click":"left"}'
  zigbee2mqtt:info 4/22/2019, 5:51:45 PM MQTT publish: topic 'zigbee2mqtt/Hallway Switch', payload '{"linkquality":115,"state_left":"ON","state_right":"OFF","click":""}'
`

aXionDevelopment, 2019-04-22:
Thinking about it, you are probably right. It makes it possible to detect a state change when you press the button repeatedly (I don't mean double click). It used to be one message, but i recall having problems detecting state changes and therefore used the mqtt message as trigger input in the past.

GForce2010, 2019-04-22:
I had the same issue with my Xioami wireless button. 

This is new issue as it didn't happen a few versions back and is not happening in the 1.3 master branch.

This is causing my automation issues as it's receiving the single click to start running my rule but the empty click is being sent so quickly afterwards that my rules don't get a chance to check the state of the button.

aXionDevelopment, 2019-04-22:
> This is causing my automation issues as it's receiving the single click to start running my rule but the empty click is being sent so quickly afterwards that my rules don't get a chance to check the state of the button.

I haven't updated my automation rules yet, but it seems logical to me that the event triggers are queued. You could just ignore the second trigger (put in an extra condition). Or isn't your trigger fired by a state change of the click action?

GForce2010, 2019-04-22:
I could, and most likely will, rewrite my rules to be able to deal with this but my point was I believe this is a bug and shouldn't be happening as it's only appeared in the last couple of commits on the dev branch.

Koenkk, 2019-04-22:
This is expected behaviour which allows you to write automation in the following form: http://www.zigbee2mqtt.io/integration/home_assistant.html#via-home-assistant-entity

GForce2010, 2019-04-22:
I don't use home assistant, i use OpenHAB hence my problem.

If that's how it's going to work from now on I will just have to rewrite my rules to cope.

Koenkk, 2019-04-22:
@GForce2010 this message should only be published when `homeassistant: true` is set in `configuration.yaml` (https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/deviceReceive.js#L187).

GForce2010, 2019-04-22:
That makes sense now, I'd forgotten I turned that on. I was helping to test a Zigbee2MQTT binding and was asked to set that to true and forgot to set to back to false.

Apologies for being an idiot.

aXionDevelopment, 2019-04-23:
Thanks for de replies. the reason for the second mqqt message has been made clear, and I guess this issue can be closed now.

closed 2019-04-23 by aXionDevelopment