Koenkk/zigbee2mqtt #1458: Z-Stack 3.0 device limit? have 22 devices (multiple routers) and cant connect more

neilvangeffen, 2019-04-22:
Hey all. Im new to using GitHub and all of these issue things etc, so please forgive me if im in the wrong place, but i found lots of help in previous issues, so going with it....

I've been having issues with getting new sensors to join with V1.2 and i figured it was because i had too many devices connected to my coordinator. So i had another CC2531 so put router FW on it, but that wouldnt connect (probably because coordinator had too many connections already?)
So tried to find a way to remove end devices so i could add more routers, but that didnt work. Is this possible?

Then i saw the dev had Z-Stack 3.0 going, so figured "fuck it" and flashed the latest Z 3.0 to CC2531 and got the dev version of zigbee2mqtt running on hassio and started re-pairing all my devices again.

I started with my routers (a CC2531 with router FW, then a wall plug (ZNCZ02LM) and finally 2  wall switches (QBKG03LM) which i then unplugged after pairing as i need to hard wire them into the house one day.
That all worked, so added 12 temp sensors, smoke detector, 5 door/window sensors for a total of 22 devices (but 2 routers disconnected)
As a quick note, i tried pairing multiple temp sensors by walking around the house after restarting zigbee2mqtt and then holding all my temp sensor buttons down for a few seconds to get them to pair, but it would only pair with one. So from then on, i would only attempt to pair one thing at a time and restarted zigbee2mqtt before each pairing.

Then it was time to add some motion sensors but they kept timing out. "Cannot get the Node Descriptor of the Device"
so tried a different one. Same thing.
Tried a push button. Same thing
Tried another wall switch (battery powered one) and same thing
These things all worked perfectly before but now just wont pair and time out.

I figured i was hitting some sort of limit on devices on my coordinator? So tried pairing the switch in the garage near the CC2531 router on the other end of the house. Still no dice.

Plugged my bench test wall switch/router back in and tried to re-paid a few devices. Still nothing. Any idea how to get some more devices going? Want a few more switches around the house....

One curious thing though.... the router sends some debug info etc and it actually has a regular entry for the single button switch but it's not showing up in the database, and pressing buttons sends to MQTT messages etc through... Not sure if thats from when i attempted to pair it last or not, but thought i'd mention it!

`topic 'zigbee2mqtt/CC2531 Router', payload '{"state":true,"linkquality":39,"description":"4604/0x00158D0001B18732","type":"CHILD_RFD","rssi":170}'`
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3102443/log.txt) 

## Debug Info
zigbee2mqtt version: 1.3.1
CC253X firmware version: 20190418

Koenkk, 2019-04-22:
When you see `"Cannot get the Node Descriptor of the Device"` it might help to restart zigbee2mqtt and then try to join a different device. (when you see this message, the devices get's stuck in the joining loop and zigbee-shepherd won't allow other devices to join).

neilvangeffen, 2019-04-22:
I tried 4 different devices multiple times with reboots each time.
Im happy to try it a few more times if you think it will help, but I have tried that a fair few times already

scuppasteve, 2019-04-23:
I am having this same issue. When i enable pairing, i start getting timeout errors of devices i tried to pair but did not succeed. I have five devices that nonstop throw errors. I can't find their id's in any file to delete them so i can pair again from the beginning.

Koenkk, 2019-04-23:
@neilvangeffen did you follow the pairing procedure from: http://www.zigbee2mqtt.io/getting_started/pairing_devices.html#xiaomi

@scuppasteve also with z-stack 3 firmware?

Koenkk, 2019-04-23:
@neilvangeffen a new CC2531 is available which allows 5 more devices to be connected and fixes some issues. (https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/Z-Stack_3.0.x/bin).

If you have any more issues (with Z-Stack 3 firmware), please continue in #1445

closed 2019-04-23 by Koenkk