Koenkk/zigbee2mqtt #1459: [Feature Request] Non-Json GET

andreasbrett, 2019-04-22:
With `zigbee2mqtt/[DEVICE_ID]/set/brightness` one can already **set** the brightness of a specific device without using JSON formatted values.

How about making this also available for **get**. Thus we'd be able to also retrieve values by subscribing to that particular MQTT topic `zigbee2mqtt/[DEVICE_ID]/get/brightness`.

I know there is already a way to *ask* for specific values (https://github.com/Koenkk/zigbee2mqtt.io/blob/dev/information/mqtt_topics_and_message_structure.md#zigbee2mqttdevice_idget) but I think it's better to subscribe to a MQTT topic and actively wait for changes rather than polling.

GForce2010, 2019-04-22:
It is already possible for Zigbee2MQTT to send and receive values in non JSON format see this issue for how to enable it in your configuration #493

andreasbrett, 2019-04-22:
> It is already possible for Zigbee2MQTT to send and receive values in non JSON format see this issue for how to enable it in your configuration #493

That's cool. Didn't know it got implemented already (I commented in that post some months ago...). Will try it out, thanks!

closed 2019-04-22 by andreasbrett