Koenkk/zigbee2mqtt #1460: Xiaomi WSDCGQ01LM device send data, but not visible on HA devices / entities

thrasheri, 2019-04-22:
I set up Zigbee2Mqtt and home assistant internal mqtt server to connect all my zigbee devices to HA. The setup went fine and log state that Zigbee2Mqtt is connected to the mqtt server. I paired one Xiaomi WSDCGQ01LM and Zigbee2Mqtt is receiving the values from the sensor. Unfortunately the is no sign of the sensor in HA entities / devices. Did I miss something? 

**Home assistant configuration.yaml :**

```
mqtt:
  password: mqttpassword
  discovery: true
```

**zigbee2mqtt configuration.yaml:**

```
devices:
  '0x00158d000255927f':
    friendly_name: lampo_elmerinhuone
    homeassistant:
      device:
        identifiers: '0x00158d000255927f'
        name: elmerinhuone
      temperature:
        name: elmerinhuone
        unique_id: lampo_elmerinhuone'
    retain: false
homeassistant: true
mqtt:
  base_topic: zigbee2mqtt
  password: mypassword
  server: mqtt://homeassistant
  user: homeassistant
permit_join: true
serial:
  port: /dev/ttyACM0
```

**Zigbee2Mqtt Logs:** 

` zigbee2mqtt:info 4/22/2019, 11:25:00 AM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 4/22/2019, 11:25:00 AM Connecting to MQTT server at mqtt://homeassistant
  zigbee2mqtt:info 4/22/2019, 11:25:00 AM zigbee-shepherd ready
  zigbee2mqtt:info 4/22/2019, 11:25:02 AM Connected to MQTT server
  zigbee2mqtt:info 4/22/2019, 11:25:02 AM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 4/22/2019, 11:25:02 AM MQTT publish: topic 'zigbee2mqtt/lampo_elmerinhuone', payload '{"temperature":22.42,"linkquality":52,"humidity":29.93,"battery":91,"voltage":2985}'
  zigbee2mqtt:info 4/22/2019, 11:25:02 AM MQTT publish: topic 'homeassistant/sensor/0x00158d000255927f/temperature/config', payload '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","state_topic":"zigbee2mqtt/lampo_elmerinhuone","json_attributes_topic":"zigbee2mqtt/lampo_elmerinhuone","name":"elmerinhuone","unique_id":"lampo_elmerinhuone'","device":{"identifiers":"0x00158d000255927f","name":"elmerinhuone","sw_version":"Zigbee2mqtt 1.3.1","model":"MiJia temperature & humidity sensor  (WSDCGQ01LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'`

Bastiencc, 2019-04-23:
Hi
did you setup/enable  discovery in HA config menu?

thrasheri, 2019-04-23:
I have discovery : true under mqtt on home assistant. See config on the first post. It should be enough right?

Bastiencc, 2019-04-23:
What do you have there : https://ibb.co/1KwMTX7 
(sorry my HA is in French)

Ukkopaha, 2019-04-23:
I have the same issue. I have tried Xiaomi WSDCGQ11LM temperature sensor, Xiaomi WXKG11LM wireless switch, Ikea Trådfri LED driver and some Airam Zigbee smart lights. Every device connects and data can be seen in logs. The HA Integrations pages shows empty in MQTT panel.

I have tried re-installing the zigbee2mqtt and configuring it again. I also re-flashed the CC2531-stick and re-installed the whole Home Assistant, but still no luck.

When I installed zigbee2mqtt -plugin for the first time few months ago, the devices appeared automatically to the devices and entities. They worked a few weeks but then they disappeared. I don't have any memories that was it after some configuration or updating or not.

Koenkk, 2019-04-24:
Can you provide the startup logs?

Ukkopaha, 2019-04-26:
Here is home-assistant.log. I think that these doesn't affiliate to this issue.

`2019-04-26 19:32:53 WARNING (MainThread) [homeassistant.components.http] Configuring api_password via the http component has been deprecated. Use the legacy api password auth provider instead. For instructions, see https://www.home-assistant.io/docs/authentication/providers/#legacy-api-password
2019-04-26 19:32:53 WARNING (MainThread) [homeassistant.components.http.auth] legacy_api_password support has been enabled.
`

thrasheri, 2019-04-28:
I checked the same config and this was there: 

2019-04-28 15:54:02 ERROR (MainThread) [homeassistant.components.device_tracker] Unable to load /config/known_devices.yaml: Config file not found: /config/known_devices.yaml

thrasheri, 2019-05-01:
I got rid of that error by creating the file, but it didn't help with the main issue.

emiguerol, 2019-05-26:
I have the same problem. No appears device on integrations.

maorbenamer, 2019-07-14:
Hello,
Did any of you solve the problem?

closed 2019-09-19 by stale[bot]

DerpyDoom, 2019-10-30:
Just got my sensor today. The exact same thing happens to me too.

Log shows that the sensor is sending data, but not visible on HA.

daveLeiser, 2020-04-12:
Got exactly the same problem. Zigbee2mqtt log saying its connected to the mosquitto broker and seeing states reported from paired devices but they don't show up on the mqtt integration page. Was anyone able to solve the problem?
What I don't get: In the zigbee2mqtt config I need to set a base topic. But in the mosquitto broker config i do not need a topic on which he should listen right?