Koenkk/zigbee2mqtt #1463: Support Reenforce LED Panel with Docker

danst0, 2019-04-22:
Hi,

I am running zigbee2mqtt from Docker. I would like to add my new Reenforce LED Panel (https://www.conrad.de/de/p/led-panel-eek-led-a-e-40-w-warm-weiss-neutral-weiss-renkforce-1408617-1408617.html).

The error message I receive is:
`
  zigbee2mqtt:warn 4/22/2019, 2:22:09 PM No converter available for 'LED1545G12' with cid 'haDiagnostic', type 'devChange' and data '{"cid":"haDiagnostic","data":{"lastMessageLqi":116,"lastMessageRssi":-71}}',
  zigbee2mqtt:warn 4/22/2019, 2:22:09 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
`

I do not have access to the folders mentioned in the "how to".

How should I proceed?

Daniel

tb-killa, 2019-04-22:
We need the line with ``WARN Device with modelID 'XXXXX' is not supported.`` from your log to build the needing converter part.

danst0, 2019-04-22:
I do not have any other warnings in my log. Are there other logs?

tb-killa, 2019-04-22:
Show the content of you ``database.db`` file.
Do you use portainer for docker management?

danst0, 2019-04-22:
`
{"id":3,"type":"EndDevice","ieeeAddr":"0x00158d00025d8edc","nwkAddr":34528,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.weather","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,1026,1027,1029,65535],"outClusterList":[0,4,65535],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"msTemperatureMeasurement":{"dir":{"value":1},"attrs":{}},"msPressureMeasurement":{"dir":{"value":1},"attrs":{}},"msRelativeHumidity":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":3},"attrs":{}}}}},"_id":"43Yy6lWKEPspM1eG"}
{"id":4,"type":"EndDevice","ieeeAddr":"0x00158d0002e23a66","nwkAddr":5541,"manufId":4151,"epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,1,3,1024,1030,1280,65535],"outClusterList":[0,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"msIlluminanceMeasurement":{"dir":{"value":1},"attrs":{}},"msOccupancySensing":{"dir":{"value":1},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"7v9Zk9ZuOfoymDe4"}
{"id":2,"type":"Router","ieeeAddr":"0x000b57fffe48fe2f","nwkAddr":30788,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E27 WS opal 980lm","epList":[1],"status":"online","joinTime":1555871344,"endpoints":{"1":{"profId":49246,"epId":1,"devId":544,"inClusterList":[0,3,4,5,6,8,768,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"10":{},"65533":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"manufacturerName":"IKEA of Sweden","modelId":"TRADFRI bulb E27 WS opal 980lm","dateCode":"20170331","powerSource":1}},"genIdentify":{"dir":{"value":1},"attrs":{"65533":1,"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"65533":1,"nameSupport":0}},"genScenes":{"dir":{"value":3},"attrs":{"65533":1,"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attrs":{"16387":1,"65533":1,"onOff":1,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"dir":{"value":1},"attrs":{"15":0,"16384":254,"65533":1,"currentLevel":254,"remainingTime":0,"onOffTransitionTime":5,"onLevel":255}},"genOta":{"dir":{"value":2},"attrs":{}},"genPollCtrl":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{"15":0,"16397":0,"16398":370,"65533":1,"remainingTime":0,"currentX":30138,"currentY":26909,"colorTemperature":370,"colorMode":2,"numPrimaries":0,"enhancedColorMode":2,"colorCapabilities":16,"colorTempPhysicalMin":250,"colorTempPhysicalMax":454}},"haDiagnostic":{"dir":{"value":1},"attrs":{"65533":1,"numberOfResets":0,"macRxBcast":0,"macTxBcast":0,"aPSRxBcast":0,"aPSTxBcast":0,"nwkFcFailure":0,"packetBufferAllocateFailures":0,"averageMacRetryPerApsMessageSent":0,"lastMessageLqi":124,"lastMessageRssi":-69}},"lightLink":{"dir":{"value":3},"attrs":{"65533":1}}}}},"_id":"RrXikdffqLO45KRq"}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0014b93022","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,11],"status":"online","joinTime":1555854733,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[0,1281,3],"outClusterList":[1280,1282,3],"clusters":{"genBasic":{"dir":{"value":1},"acls":{"zclVersion":"R","hwVersion":"R","manufacturerName":"R","modelId":"R","dateCode":"R","powerSource":"R","locationDesc":"RW","physicalEnv":"RW","deviceEnabled":"RW"},"attrs":{"zclVersion":1,"hwVersion":1,"manufacturerName":"sivann inc.","modelId":"hiver0001","dateCode":"20170407","powerSource":1,"locationDesc":"    ","physicalEnv":0,"deviceEnabled":1}},"ssIasAce":{"dir":{"value":1}},"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}},"ssIasWd":{"dir":{"value":2}},"genIdentify":{"dir":{"value":3},"acls":{"identifyTime":"RW"},"attrs":{"identifyTime":0}}}}},"_id":"n0P9dp9WB5rRzMNi"}
{"$$indexCreated":{"fieldName":"id","unique":true,"sparse":false}}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0014b93022","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1555917536,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265, [... cut ...] olorTemperature":370,"colorMode":2,"numPrimaries":0,"enhancedColorMode":2,"colorCapabilities":16,"colorTempPhysicalMin":250,"colorTempPhysicalMax":454}},"haDiagnostic":{"dir":{"value":1},"attrs":{"65533":1,"numberOfResets":0,"macRxBcast":0,"macTxBcast":0,"aPSRxBcast":0,"aPSTxBcast":0,"nwkFcFailure":0,"packetBufferAllocateFailures":0,"averageMacRetryPerApsMessageSent":0,"lastMessageLqi":116,"lastMessageRssi":-71}},"lightLink":{"dir":{"value":3},"attrs":{"65533":1}}}}},"_id":"RrXikdffqLO45KRq"}

`

danst0, 2019-04-22:
> Do you use portainer for docker management?

Yes, I use portainer.

danst0, 2019-04-29:
Any way I can help?

Koenkk, 2019-04-30:
`LED1545G12` is a tradfri bulb, looking at your `database.db` I don't  see the device, did you pair it?

danst0, 2019-05-01:
Ok, I see, the panel has not even been recongnized yet. In the manual it says that pairing mode is active during the first 5 seconds after power on. However even after a few times of power-cycling, I do not see anything in the logs. This might be too short for zigbee2mqtt?

Koenkk, 2019-05-02:
Is this a zigbee device at all? I don't see it mentioned on https://www.conrad.de/de/p/renkforce-1408617-led-panel-eek-led-a-e-40-w-warm-weiss-neutral-weiss-1408617.html also https://produktinfo.conrad.com/datenblaetter/1400000-1499999/001408617-an-01-ml-LED_PANEL_40_W_KW_WW__DIMMBA_de_en_fr_nl.pdf doesn't mention it.

danst0, 2019-05-05:
My mistake: The panel is not Zigbee enabled, however the required LED driver is: https://www.conrad.de/de/p/renkforce-zigzee-modul-1408619-led-treiber-dimmbar-1408619.html

The manual did not give me any hints on the pairing process;-(

tb-killa, 2019-05-05:
@danst0 why there are so many qr code buttons mentioned by the manual? Normally the device should be in pairing mode if you power them on.
Doesn't the device support any button or other interaction?

nvoelzow, 2019-07-09:
The LED drivers don't have any pairing button, the whole system including their wifi-gateway is using QR codes to bind the LED drivers to the gateway inside the android/ios app - I've also had no luck here getting anything displayed whatsoever from zigbee2mqtt with debug enabled when power-cycling those drivers...

tb-killa, 2019-07-09:
If you scan the qr code with any reader app what does this return?

nvoelzow, 2019-07-09:
It contains just a single number like 4157758 - which is also printed underneath the code

tb-killa, 2019-07-09:
Hmm strange.
Please Describe in detail how the original commissioning is done with the gateway.
Do you have another CC2531 stick to use a sniffer ?

tb-killa, 2019-07-09:
I found in one pdf document (Gateway-Manual) one Info about adding Device with "Manual Mode":
``How the binding mode is activated depends on
the lamp. For the LED driver, it is active for 5 seconds once the driver is supplied with voltage, for
example``

nvoelzow, 2019-07-09:
I just checked with my CC2531 stick with sniffer FW to see if it announces anything, and it sends a packet every 15s after being powered up:

[image]
[LED-Driver.psd.zip](https://github.com/Koenkk/zigbee2mqtt/files/3374705/LED-Driver.psd.zip)

I'll see if I can get the gateway to play nicely and record the packets of binding a newly added driver (unfortunately the app for the gateway and/or the gateway are terribly unreliable to use here for some reason)

nvoelzow, 2019-07-09:
by the way, the "manual" for the driver unfortunately doesn't say anything about the Zigbee setup part at all:
http://downloads.cdn.re-in.de/1400000-1499999/001408619-an-01-ml-LED_TREIBER_INKL__ZIGBEE_MOD_de_en_fr_nl.pdf

tb-killa, 2019-07-09:
Maybe the Device need special Channel to get correctly working ?

nvoelzow, 2019-07-09:
hmm tried it with channels 11, 15, 20 and 25 - nothing shows up in zigbee2mqtt...

Just to make sure, I should be able to just pull the source on an RPi3, run npm install, start it and get at least some meaningful messages when a new device is powered up and in pairing mode, right?
Is there any way to make sure the dongle is doing what it's supposed to? Flashed it with the CC2531ZNP-Prod.hex according to the docs..

/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.4.0 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 7/9/2019, 10:54:21 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-07-09.22-54-21'
  zigbee2mqtt:debug 7/9/2019, 10:54:21 PM Removing old log directory '/opt/zigbee2mqtt/data/log/2019-07-09.22-14-48'
  zigbee2mqtt:debug 7/9/2019, 10:54:22 PM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/opt/zigbee2mqtt/data/database.db","coordBackupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 7/9/2019, 10:54:22 PM Loaded state from file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:debug 7/9/2019, 10:54:22 PM Saving state to file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:info 7/9/2019, 10:54:22 PM Starting zigbee2mqtt version 1.4.0 (commit #927c4db)
  zigbee2mqtt:info 7/9/2019, 10:54:22 PM Starting zigbee-shepherd
  zigbee2mqtt:info 7/9/2019, 10:54:24 PM zigbee-shepherd started
  zigbee2mqtt:info 7/9/2019, 10:54:24 PM Coordinator firmware version: '20190608'
  zigbee2mqtt:debug 7/9/2019, 10:54:24 PM zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":"11","panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"ieeeAddr":"0x00124b001938f081","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20190608},"startTime":1562709264,"joinTimeLeft":0}
  zigbee2mqtt:info 7/9/2019, 10:54:24 PM Currently 0 devices are joined:
  zigbee2mqtt:warn 7/9/2019, 10:54:24 PM `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 7/9/2019, 10:54:24 PM Allowing new devices to join.
  zigbee2mqtt:warn 7/9/2019, 10:54:24 PM Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 7/9/2019, 10:54:24 PM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 7/9/2019, 10:54:24 PM Connecting to MQTT server at mqtt://localhost
  zigbee2mqtt:info 7/9/2019, 10:54:24 PM zigbee-shepherd ready
  zigbee2mqtt:info 7/9/2019, 10:54:24 PM Connected to MQTT server
  zigbee2mqtt:info 7/9/2019, 10:54:24 PM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 7/9/2019, 10:54:24 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.4.0","commit":"927c4db","coordinator":20190608,"log_level":"debug","permit_join":true}'
  zigbee2mqtt:debug 7/9/2019, 10:54:24 PM Mounted the cieApp (epId 11)

nvoelzow, 2019-07-13:
ok, just received an osram smart+ light and everything there is working beautifully straight out of the box as expected, so the stick and zigbee2mqtt seem to be fine..

qosch, 2019-08-03:
Any news on this? I bought the driver only for a project and thought it wouldn't be too hard to add support for it but as it doesn't even show up in zigbee2mqtt I wouldn't know where to start. I had the device open and found the original manufacturer (it's only rebranded by Renkforce) but forgot to take pictures/note the original manufacturers name. I did take a picture of the Zigbee module though:
[image]

nvoelzow, 2019-08-04:
I found the website of the company who makes the modules, but couldn't find any useful docs about it - found mentions of Zigbee pro and ZLL and tried different firmwares on the stick but then zigbee2mqtt (or the shepherd) wouldn't start. So until now I have not been able to talk to the driver at all using zigbee2mqtt...

qosch, 2019-08-04:
Could you provide the name? Maybe I'll write them and ask for help...

nvoelzow, 2019-08-04:
http://www.szhoma.com/products_zigbee_HZM401.html is the only other reference I was able to find - the site lists the module, the driver and the wifi gateway (HGW401, also relabeled by renkforce)

closed 2019-10-10 by stale[bot]