Koenkk/zigbee2mqtt #1465: 2 coordinators on one server ?

pimseb, 2019-04-23:
Hello,
I'm setting up a zigbee network in my home and have a noob question :)
My coordinator is linked on my main raspberry running mosquitto as MQTT server and domoticz.
My house is quite big and has large walls, making routers (should be called repeaters imo) usage not easy. I would need netween 2 and 4 repeaters chained to reach my last place and I'm not sure it would work this way.
I wonder if it's possible to use 2 or 3 coordinators each plugged in a dedicated raspberry linked on my ethernet network.

I've read that a zigbee network can have only one coordinator. And it seems to be true because yesterday I ran 2 coordinators (one on my main raspberry and another one on another raspberry) all linked to the same mosquitto server and I got many errors in domoticz (unhandled messages).

So I'd like to know if there is a way to use 2 coordinators through  ethernet with a single mosquitto mqtt server.

Or should I use in my case 2 raspberry, 2 zigbee2mqtt coordinators, 2 mqtt and 2 domoticz python plugins ?

lolorc, 2019-04-25:
Hi, I'm using 2 coordinators connected to the same raspberry at the moment, both z2m instances use the same mqtt topic. I have a "production" network and a test one.

It is not very convenient as it is, it's working as I want for the devices topic but it's a hassle for all the bridge subtopics. At the moment I stop one of the z2m instance to use the network map feature for example.

I guess z2m could have 2 topics, one for bridge config, one for devices and the devices topic could be shared among z2m instances, that would make it easier to repair a device from a network to another.

danpowell88, 2019-04-26:
From my understanding and usage the routers work exactly as you want, or at least using powered bulbs (I'm using hue) will do this, looking at my network map I can see it shows multiple routes back to the co-ordinator some directly and other via routers.

The cc25xx device routers however dont show these routes on the network map, its highly likely this is just a display issue as I'm fairly sure I've observed them talking to other cc25xx devices in the debug output

closed 2019-07-02 by stale[bot]