Koenkk/zigbee2mqtt #1469: Add support to lanbon switch L7 series

m0oba, 2019-04-24:
Hi all,

I was trying to pair 3 gang switch from [Lambon,](http://www.lanbon.cn/en/product/73217058.html) but I couldn't pair it or see anything in zigbee2mqtt log. 

I start to sniff the network with Wireshark and I found out that the switch is on channel 15,
and record the communication between ZigBee coordinator and the lanbon switch.

I was also able to record one successful pairing,  with the lambon coordinator and the switch. 

I compare the 2 recordings, and the only difference I could find is the Zigbee security header when the coordinators send the Network key.  The lambon coordinator doesn't send it  (see below).

Lambon coordinator
[image]

Zigbee2mqtt coordinator
[image]

So from there, my idea was to find a way to disable the ZigBee security and see if the switch can pair with zigbee2mqtt. 
on this issue #1126 @Koenkk explains how to disable security by compiling new firmware and modifying the firmware .patch with  #define SECURE 0.

I try to compile the firmware,  but it was a total failure! :(
the patch didn't work (trailing white space), I try to manually correct all the files but at the end, I wasn't able to compile it.

I'm new to zigbee2mqtt and Zigbee protocol, I try to do it on my own, but I reach my limit.
if anybody can help me and give me some direction,  it would be much appreciated.

I've attached the 2 pcap files : 
- successful pairing with the lanbon coordinator
- unsuccessful paring with zigbee2mqtt 

[lanbon wireshark.zip](https://github.com/Koenkk/zigbee2mqtt/files/3113356/lanbon.wireshark.zip)

Koenkk, 2019-04-26:
To fix the whitespace errors, just using the following command `git apply firmware.patch --ignore-space-change`. I've updated the instructions here: https://github.com/Koenkk/Z-Stack-firmware/blob/dev/coordinator/Z-Stack_Home_1.2/COMPILE.md

m0oba, 2019-04-26:
Thank you @Koenkk the patch works.

I can compile the code, but I still have an issue:

- I have no problem to build without changing anything: Build OK, zigbee2mqtt OK

- When modifying preinclude.h with #define SECURE 0 or FALSE : Build OK with 79 warning, zigbee sheperd doesn't start properly.
 _warning from IAR workbench_
`Warning[Pe047]: incompatible redefinition of macro "SECURE" C:\Texas Instruments\Z-Stack Home 1.2.2a.44539\Projects\zstack\ZNP\CC253x\Source\preinclude.h 11 `

- When commenting the line // #define SECURE 1: BUILD OK, zigbee2mqtt start, but the security header is still there.

[build_with_SECURE_0.log](https://github.com/Koenkk/zigbee2mqtt/files/3121935/build_with_SECURE_0.log)

m0oba, 2019-04-26:
I fix the warning by changing -DSECURE=0 in f8wConfig.cfg,  but I'm still not able to start zigbee2mqtt.

zigbee2mqtt.log
```
> zigbee2mqtt@1.3.1 start /app

> node index.js

  zigbee2mqtt:info 4/26/2019, 5:30:09 PM Logging to directory: '/app/data/log/2019-04-26.17-30-09'
  zigbee2mqtt:debug 4/26/2019, 5:30:09 PM Removing old log directory '/app/data/log/2019-04-26.17-14-15'
  zigbee2mqtt:debug 4/26/2019, 5:30:11 PM Using zigbee-shepherd with settings: '{"net":{"panId":1174,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[15],"precfgkey":"HIDDEN"},"dbPath":"/app/data/database.db","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 4/26/2019, 5:30:11 PM Loaded state from file /app/data/state.json
  zigbee2mqtt:debug 4/26/2019, 5:30:11 PM Saving state to file /app/data/state.json
  zigbee2mqtt:info 4/26/2019, 5:30:12 PM Starting zigbee2mqtt version 1.3.1 (commit #unknown)
  zigbee2mqtt:info 4/26/2019, 5:30:12 PM Starting zigbee-shepherd
  zigbee2mqtt:info 4/26/2019, 5:30:14 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
  zigbee2mqtt:info 4/26/2019, 5:31:14 PM Starting zigbee-shepherd
  zigbee2mqtt:error 4/26/2019, 5:31:16 PM Error while starting zigbee-shepherd!
  zigbee2mqtt:error 4/26/2019, 5:31:16 PM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 4/26/2019, 5:31:16 PM Failed to start
	{"message":"rsp error: 1","stack":"Error: rsp error: 1\n    at rspHdlr (/app/node_modules/zigbee-shepherd/lib/components/controller.js:309:29)\n    at CcZnp.<anonymous> (/app/node_modules/cc-znp/lib/ccznp.js:264:13)\n    at Object.onceWrapper (events.js:273:13)\n    at CcZnp.emit (events.js:182:13)\n    at CcZnp._mtIncomingDataHdlr (/app/node_modules/cc-znp/lib/ccznp.js:366:14)\n    at Immediate._onImmediate (/app/node_modules/cc-znp/lib/ccznp.js:342:22)\n    at runCallback (timers.js:705:18)\n    at tryOnImmediate (timers.js:676:5)\n    at processImmediate (timers.js:658:5)"}
  zigbee2mqtt:error 4/26/2019, 5:31:16 PM Exiting...
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.3.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.3.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-04-26T17_31_16_786Z-debug.log
Using '/app/data' as data directory
```

[/root/.npm/_logs/2019-04-26T17_54_39_932Z-debug.log](https://github.com/Koenkk/zigbee2mqtt/files/3122498/2019-04-26T17_54_39_932Z-debug.log)

Koenkk, 2019-04-28:
Probably because the security key is written now. You need to comment the following lines:

https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/controller.js#L263
https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/controller.js#L285
https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/controller.js#L286

https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/controller.js#L588 till 590
https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/controller.js#L601 till 603

m0oba, 2019-05-11:
Now, zigbee-sheperd start normally  with the custom firmware. \o/

but now it still doesn't work because the coordinator doesn't send the transport key.
what I try to achieve is to send the transport key without the security key.

@Koenkk 
my issue is from zigbee-sheperd or the firmware?

Koenkk, 2019-05-13:
@m0oba I guess this requires a change in the firmware, as this part is not handled by zigbee2mqtt/zigbee-shepherd. The best place to ask firmware related questions is on the Texas Instruments forum.

m0oba, 2019-05-15:
i'm finally able to see the switch on zigbee2mqtt
the fix was quite easy, just comment #define TC_LINKKEY_JOIN in preinclude.h. 

Now i'm trying to find the modelID, but i cannot find it in the database.db
I delete the switch from the database.db and repair it but no success yet.

@Koenkk any ideas?

Log
```
> zigbee2mqtt@1.4.0 start /opt/zigbee
> node index.js

  zigbee2mqtt:info 5/15/2019, 2:35:56 PM Logging to directory: '/opt/zigbee/data/log/2019-05-15.14-35-56'
  zigbee2mqtt:debug 5/15/2019, 2:35:56 PM Removing old log directory '/opt/zigbee/data/log/2019-05-15.13-04-21'
  zigbee2mqtt:debug 5/15/2019, 2:35:58 PM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[15],"precfgkey":"HIDDEN"},"dbPath":"/opt/zigbee/data/database.db","coordBackupPath":"/opt/zigbee/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 5/15/2019, 2:35:58 PM Loaded state from file /opt/zigbee/data/state.json
  zigbee2mqtt:debug 5/15/2019, 2:35:58 PM Saving state to file /opt/zigbee/data/state.json
  zigbee2mqtt:info 5/15/2019, 2:35:58 PM Starting zigbee2mqtt version 1.4.0 (commit #6b75465)
  zigbee2mqtt:info 5/15/2019, 2:35:58 PM Starting zigbee-shepherd
  zigbee2mqtt:info 5/15/2019, 2:36:00 PM zigbee-shepherd started
  zigbee2mqtt:info 5/15/2019, 2:36:00 PM Coordinator firmware version: '20190425'
  zigbee2mqtt:debug 5/15/2019, 2:36:00 PM zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":"15","panId":65535,"extPanId":[0,0,18,25,56,75,89,165],"ieeeAddr":"0x00124b001938a559","nwkAddr":0},"firmware":{"transportrev":2,"product":2,"version":"2.7.2","revision":20190425},"startTime":1557905760,"joinTimeLeft":0}
  zigbee2mqtt:info 5/15/2019, 2:36:00 PM Currently 1 devices are joined:
  zigbee2mqtt:info 5/15/2019, 2:36:00 PM unknown (0x00124b0005b67ae3): unkown - undefined unknown (Router)
  zigbee2mqtt:warn 5/15/2019, 2:36:00 PM `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 5/15/2019, 2:36:00 PM Allowing new devices to join.
  zigbee2mqtt:warn 5/15/2019, 2:36:00 PM Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 5/15/2019, 2:36:00 PM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 5/15/2019, 2:36:00 PM Connecting to MQTT server at mqtt://localhost
  zigbee2mqtt:info 5/15/2019, 2:36:01 PM zigbee-shepherd ready
  zigbee2mqtt:info 5/15/2019, 2:36:01 PM Connected to MQTT server
  zigbee2mqtt:info 5/15/2019, 2:36:01 PM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 5/15/2019, 2:36:01 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.4.0","commit":"6b75465","coordinator":20190425,"log_level":"debug","permit_join":true}'
  zigbee2mqtt:debug 5/15/2019, 2:36:01 PM Mounted the cieApp (epId 11)
  zigbee2mqtt:debug 5/15/2019, 2:36:21 PM Received zigbee message of type 'endDeviceAnnce' with data '"0x00124b0005b67ae3"' of device 'undefined' (0x00124b0005b67ae3) of endpoint 12
  zigbee2mqtt:info 5/15/2019, 2:36:21 PM New device 'undefined' with address 0x00124b0005b67ae3 connected!
  zigbee2mqtt:debug 5/15/2019, 2:36:21 PM Saving state to file /opt/zigbee/data/state.json
  zigbee2mqtt:info 5/15/2019, 2:36:21 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00124b0005b67ae3","meta":{}}'
  zigbee2mqtt:warn 5/15/2019, 2:36:21 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 5/15/2019, 2:36:21 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:debug 5/15/2019, 2:37:22 PM Received zigbee message of type 'devInterview' with data '"0x00124b0005b67ae3"'
  zigbee2mqtt:debug 5/15/2019, 2:37:52 PM Received zigbee message of type 'devInterview' with data '"0x00124b0005b67ae3"'
  zigbee2mqtt:debug 5/15/2019, 2:38:22 PM Received zigbee message of type 'devInterview' with data '"0x00124b0005b67ae3"'
  zigbee2mqtt:debug 5/15/2019, 2:38:22 PM Accept device incoming with ieeeAddr '0x00124b0005b67ae3' permit join is 'true'
  zigbee2mqtt:debug 5/15/2019, 2:38:22 PM Allowing device '0x00124b0005b67ae3' to join
  zigbee2mqtt:debug 5/15/2019, 2:38:22 PM Received zigbee message of type 'devIncoming' with data '"0x00124b0005b67ae3"' of device 'undefined' (0x00124b0005b67ae3) of endpoint 12
  zigbee2mqtt:info 5/15/2019, 2:38:22 PM Device incoming...
  zigbee2mqtt:info 5/15/2019, 2:38:22 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:warn 5/15/2019, 2:38:22 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 5/15/2019, 2:38:22 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:debug 5/15/2019, 2:38:22 PM Received zigbee message of type 'devStatus' with data '"online"' of device 'undefined' (0x00124b0005b67ae3) of endpoint 12
  zigbee2mqtt:warn 5/15/2019, 2:38:22 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 5/15/2019, 2:38:22 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:info 5/15/2019, 2:39:20 PM Succesfully reenabled joining
  zigbee2mqtt:debug 5/15/2019, 2:40:58 PM Saving state to file /opt/zigbee/data/state.json
  zigbee2mqtt:info 5/15/2019, 2:42:40 PM Succesfully reenabled joining
```
database.db
```
[... cut ...] Id":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}}}}},"_id":"07Jse11nA1QEDkdF"}
```

Koenkk, 2019-05-15:
This devices probably doesn't send it's modelId, you can add one manually: http://www.zigbee2mqtt.io/getting_started/pairing_devices.html#gledopto

m0oba, 2019-05-16:
Thank you @Koenkk,  I really appreciate it your help  :)

I don't really know what I'm doing!
so, please find below my first two tries :)

device.js #1
```
zigbeeModel: ['L7-HS3'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
        model: 'L7-HS3', // Vendor model number, look on the device for a model number
        vendor: 'Lanbon', // Vendor of the device (only used for documentation and startup logging)
        description: 'Switch', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
        supports: 'on/off', // Actions this device supports (only used for documentation)
        fromZigbee: [fz.ignore_onoff_change, fz.state],
        toZigbee: [tz.on_off],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 14);
            const cfg = {direction: 0, attrId: 0, dataType: 32, minRepIntval: 0, maxRepIntval : 1000, repChange: 0};
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
            ];

            execute(device, actions, callback);
        },
```
Log #1 
```
zigbee2mqtt:info 5/17/2019, 12:35:19 AM Logging to directory: '/opt/zigbee/data/log/2019-05-17.00-35-18'
  zigbee2mqtt:debug 5/17/2019, 12:35:19 AM Removing old log directory '/opt/zigbee/data/log/2019-05-15.21-23-38'
  zigbee2mqtt:debug 5/17/2019, 12:35:20 AM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[15],"precfgkey":"HIDDEN"},"dbPath":"/opt/zigbee/data/database.db","coordBackupPath":"/opt/zigbee/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 5/17/2019, 12:35:20 AM Loaded state from file /opt/zigbee/data/state.json
  zigbee2mqtt:debug 5/17/2019, 12:35:20 AM Saving state to file /opt/zigbee/data/state.json
  zigbee2mqtt:info 5/17/2019, 12:35:20 AM Starting zigbee2mqtt version 1.4.0 (commit #6b75465)
  zigbee2mqtt:info 5/17/2019, 12:35:20 AM Starting zigbee-shepherd
  zigbee2mqtt:info 5/17/2019, 12:35:22 AM zigbee-shepherd started
  zigbee2mqtt:info 5/17/2019, 12:35:22 AM Coordinator firmware version: '20190315'
  zigbee2mqtt:debug 5/17/2019, 12:35:22 AM zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":"15","panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"ieeeAddr":"0x00124b001938a559","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20190315},"startTime":1558028122,"joinTimeLeft":0}
  zigbee2mqtt:info 5/17/2019, 12:35:22 AM Currently 1 devices are joined:
  zigbee2mqtt:info 5/17/2019, 12:35:22 AM 0x00124b0005b67ae3 (0x00124b0005b67ae3): L7-HS3 - Lanbon Switch (Router)
  zigbee2mqtt:warn 5/17/2019, 12:35:22 AM `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 5/17/2019, 12:35:22 AM Allowing new devices to join.
  zigbee2mqtt:warn 5/17/2019, 12:35:22 AM Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 5/17/2019, 12:35:22 AM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 5/17/2019, 12:35:22 AM Connecting to MQTT server at mqtt://localhost
  zigbee2mqtt:debug 5/17/2019, 12:35:23 AM Configuring 0x00124b0005b67ae3 (0x00124b0005b67ae3) ...
  zigbee2mqtt:info 5/17/2019, 12:35:23 AM zigbee-shepherd ready
  zigbee2mqtt:info 5/17/2019, 12:35:23 AM Connected to MQTT server
  zigbee2mqtt:info 5/17/2019, 12:35:23 AM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 5/17/2019, 12:35:23 AM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.4.0","commit":"6b75465","coordinator":20190315,"log_level":"debug","permit_join":true}'
  zigbee2mqtt:debug 5/17/2019, 12:35:23 AM Mounted the cieApp (epId 11)
  zigbee2mqtt:warn 5/17/2019, 12:35:53 AM Failed to configure 0x00124b0005b67ae3 (0x00124b0005b67ae3) ('Error: Timed out after 30000 ms') (attempt #1)
  zigbee2mqtt:warn 5/17/2019, 12:35:53 AM This can be ignored if the device is working properly
```
---------------------------------------------------------
device.js #2 
```
{
        zigbeeModel: ['L7-HS3'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
        model: 'L7-HS3', // Vendor model number, look on the device for a model number
        vendor: 'Lanbon', // Vendor of the device (only used for documentation and startup logging)
        description: 'Switch', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
        supports: 'on/off', // Actions this device supports (only used for documentation)
        fromZigbee: [fz.ignore_onoff_change, fz.state],
        fromZigbee: [],
        toZigbee: [tz.on_off],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
[... 14 lines cut ...]
  zigbee2mqtt:debug 5/17/2019, 12:44:19 AM Removing old log directory '/opt/zigbee/data/log/2019-05-15.21-25-45'
[... cut ...]
  zigbee2mqtt:info 5/17/2019, 12:44:23 AM zigbee-shepherd ready
  zigbee2mqtt:info 5/17/2019, 12:44:23 AM Connected to MQTT server
  zigbee2mqtt:info 5/17/2019, 12:44:23 AM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 5/17/2019, 12:44:24 AM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.4.0","commit":"6b75465","coordinator":20190315,"log_level":"debug","permit_join":true}'
  zigbee2mqtt:debug 5/17/2019, 12:44:24 AM Mounted the cieApp (epId 11)
```

Koenkk, 2019-05-17:
Are you able to turn the device on off by sending to `zigbee2mqtt/0x00124b0005b67ae3/set` payload `ON` or `OFF`?

closed 2019-07-23 by stale[bot]

digit-ill, 2020-02-11:
@m0oba, I’ve just started playing with Lanbon switches and wanting to know if you had any success in the end? I find the Lanbon app pretty limited and would like to be able to integrate these switches into a wider environment and not have them connected to the internet. I’m completely new to the world of zigbee2mqtt so wondering whether it’s worth getting the hardware or just accepting that the Lanbon switches are what they are. Thanks in advance.