Koenkk/zigbee2mqtt #1471: Is there a way to clear the previous device memory on CC253x

flexage, 2019-04-25:
Latest Z2M and latest max_stability coorfinator firmware.

I tried to remove all my paired devices using the `/zigbee2mqtt/config/remove`  topic, for each device the logs stated that devices couldn't be removed, and that it was trying force remove.

The force remove worked, but from some strange behaviours I'm seeing with only previously paired devices being willing to reconnect, but now connecting as an unrecognised devices - I'm pretty sure they weren't removed from the memory on the CC253x stick I'm using.

I've tried wiping, and re-flashing but that doesn't seem to have fixed the problem... I've tried searching the net for any docs related to this, but not managed to turn anything up. I don't really want to have to take my chances on waiting another month for a new CC stick to arrive from China (the last one I ordered never did). So is there a way I can COMPLETELY wipe/factory reset the CC253x stick... firmware, memory, kitchen sink and all, so that I can reflash the firmware and be back at square one?

andreasbrett, 2019-04-25:
Have you tried removing all the necessary lines from the database.db file?

flexage, 2019-04-25:
I have, I've tried uninstalling the Z2M hass.io addon and deleting the directory with the database.db, configuration.yaml, etc...

Koenkk, 2019-04-26:
Removing everything under `data` and reflashing the stick completely clears the memory. But note that the zigbee network is still running (the devices that were in your previous network still form a valid zigbee network). Meaning that when you power on the CC253x again, everything device will re-join your network. 

In case you don't want this, you could either change the channel or network key (http://www.zigbee2mqtt.io/information/what_does_and_doesnt_require_repairing.html)

flexage, 2019-04-28:
Closing, I eventually got them to pair by changing the channel, switching off all mains powered zigbee devices and hubs (including all my hue devices on a separate hue hub), and getting the bulbs closer to the CC2531. (I had to physically move the server tower, since the GU10's I was trying to mount were in the ceiling, and I had no other way of powering them 😂 )

closed 2019-04-28 by flexage