Koenkk/zigbee2mqtt #1472: Unable to decode ZCL data after re-pairing

he902, 2019-04-26:
After changed the MAC address of my CC2531, I did the re-pairing.
But is is unable to decode ZCL data after re-pairing:
```
  serialport/stream binding.read finished +5s
  cc-znp { sof: 254,
  cc-znp   len: 27,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'AF',
  cc-znp   cmd: 'incomingMsg',
  cc-znp   payload:
  cc-znp    { groupid: 0,
  cc-znp      clusterid: 32,
  cc-znp      srcaddr: 4946,
  cc-znp      srcendpoint: 1,
  cc-znp      dstendpoint: 1,
  cc-znp      wasbroadcast: 0,
  cc-znp      linkquality: 76,
  cc-znp      securityuse: 0,
  cc-znp      timestamp: 9420037,
  cc-znp      transseqnumber: 0,
  cc-znp      len: 7,
  cc-znp      data: <Buffer 08 b6 0a 00 50 20 00> },
  cc-znp   fcs: 91,
  cc-znp   csum: 91 } +5s
  serialport/stream _read reading +10ms
  serialport/binding-abstract read +11ms
  serialport/bindings/unixRead Starting read +10ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 32, srcaddr: 4946, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 76, securityuse: 0, timestamp: 9420037, transseqnumber: 0, len: 7, data: <Buffer 08 b6 0a 00 50 20 00> } +5s
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":32,"srcaddr":4946,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":76,"securityuse":0,"timestamp":9420037,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[8,182,10,0,80,32,0]}} +5s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +5s
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +3ms
```
In the past, it was OK to decode the ZCL data:
```
4/24/2019, 11:51:14 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"genPollCtrl","data":{"20480":0}}' of device 'FNB56-DOR23FB1.3' (0x00158d0002acba74) of endpoint 1
```
The ZCL data `<Buffer 08 b6 0a 00 50 20 00>` is the same as before.
Here is my [database.db](https://github.com/Koenkk/zigbee2mqtt/files/3119398/database.db.txt)

he902, 2019-04-26:
After checked the log, I found the pairing was not successful.
Tried several time of re-pairing, no luck.

I noticed, there was no output in the log during re-pairing, seems zigbee2mqtt or CC2531 was dead!
So I decided to plug/re-plug CC2531 from the USB port. After that, everything goes OK!

Koenkk, 2019-04-26:
Great that everything is OK now!

closed 2019-04-26 by Koenkk