Koenkk/zigbee2mqtt #1474: Enddevices unreachable because rejoin=false

allofmex, 2019-04-26:
# Bug Report
Not a zigbee2mqtt bug but effects all zigbee(-shepherd) user
See also [Enddevices unreachable because route not updating](https://github.com/Koenkk/zigbee2mqtt/issues/1408) (similar effects but different reason)

## What happened
Enddevices (mainly hue motion sensor) get disconnected from network and are not working anymore

## How to reproduce it (minimal and precise)
Motion sensor stops working after some hours.
Those with previous firmware (6.1.0.18912) rejoin after some time (up to 30 minutes)
Those with latest firmware (6.1.1.27575) shut down completely, setup button or battery reset needed

## Debug Info
The problem seems to be:
Some of my routers are kicking out devices be sending `LEAVE` with **REJOIN=FALSE**

[image]
Motion sensor (src64=...62:77) sends some last messages incl. own `LEAVE` anouncement and disconnects **permanently** from network. Next is coordinators `route request` that will never be answered anymore.

Have seen this REJOIN=FALSE for following parent router
- MüllerLicht, ZBT-ColorTemperature Bulb (Jennic 00:15:8d...)
- Osram Plug (7c:b0:3e...)

[image]

latest ioBroker dev version
CC253X firmware version: custom firmware close to 20190425

Expected behavior would be REJOIN=TRUE, as it happens on most of the parent changes:
[image]

I don't expect that we can fix this ourselfs :-(
First step should be to gather some more details:
- Who has similar experiences?
- Who uses Hue motion sensors? Are they stable? What router do you have within range?
(This may effect mainly (but not only) hue motion sensor owners since these sensor show a very annoying "parent hopping" behavior. Thats why they are much more in risk to be kicked out by one if it's many parents.)
- Does anyone know what would be the correct behavior on rejoin=false? "Don't rejoin the [network/parent/parent for next x minutes/....?]"

allofmex, 2019-04-26:
Very wired.
Sometimes even new firmware recovers from rejoin=false

[image]

20 minutes later (15:27), sensor does a rejoin request to same parent that kicked it out, and the parent confirms. Again working.

Two hours later it was kicked again, it did not recover as of now (50 minutes)

allofmex, 2019-04-28:
Removed all Osram plugs and ZBT bulbs from network. All 4 motion sensor stable for 24 hours :-) (then cc2531 crashed, but different story :-/ )

**No one out there with hue motion sensors?**

sti0, 2019-05-14:
I'm running HASSIO with Zigbee2MQTT plugin and an cc2531 as an coordinator. Got some IKEA and HUE bulbs, HUE motion sensors, dimmer switches, xiaomi devices and these OSRAM plugs. Every day I got another issue, first I thought it was the signal strenght of the cc2531 so I added an cc2530+cc2591 to my network but it got even worse. Motion sensor and the xiaomi devices stopped working randomly. 
After googleing around I found this thread and started to remove the OSRAM plugs. Seems that the network got even stable.

pergolafabio, 2019-05-15:
i also have only the conbee stick, with ONLY 4 x aqara motion sensors, all is stable for about 2-3 weeks, then suddenly 1 aqara motion is unavailable, i need to repair

is this the same issue?

allofmex, 2019-05-15:
> started to remove the OSRAM plugs. Seems that the network got even stable.

@sti0 Thanks for the report. I hope we find a third one that can confirm, than it will be worth a try to contact the manufacturer

> only the conbee stick, with ONLY 4 x aqara motion sensors

@pergolafabio Sorry, seems not to be related. This issue is that router devices kick the sensor. But you don't have any router in your network.

pergolafabio, 2019-05-15:
well @allofmex  , i also had Osram Smart plugs, then my issue with end devices was much more frequent, like 5-8 days, then my end devices were kiced
i didnt use the osrams, just plugged one in on every floor , now with the removal of osrams, its much stabeler, now end devices are lost like every 3-4 weeks

see screenshot, ignore the red circle , thats something else :)

https://www.dropbox.com/s/l6ant656clm7z5l/nodeinfo.JPG?dl=0

peexl, 2019-06-18:
Hello,
Thank to you now I know why my hue motion does not work as expected .
I've tried everything : repair hundreds times, change network key, change batteries , but the outcome is still the same : the sensor work for some time( generally 5-10 mins) then stop working (could be 1h, 2h, 3h, not always the same time) and  work again  etc
In the end i gave up the idea of using my hue motion sensor.

My setup is : 
Routers : 
4 Xiaomi aqara led bulbs 
3 Philips Hue white and color
1 Ikea outlet
1 ikea Bulb
1 Innr outlet 

Enddevices : 
2 philips hue dimmers
1 hue motion sensor
5 xiaomi body sensors
4 xiaomi wireless switches 
3 xiaomi temperature sensors 

Only the hue motion sensor act like this all the others devices are working just fine

Unfortunately I don't have the tools to sniff zigbee packets but here my logs 

when it's working i have :
`6/18/2019, 4:06:30 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"msOccupancySensing","data":{"occupancy":1}}' of device 'SML001' (0x001788010213d449) of endpoint 2
6/18/2019, 4:06:30 PM - info: MQTT publish: topic 'zigbee2mqtt/0x001788010213d449', payload '{"battery":100,"linkquality":60,"illuminance":17022,"temperature":24.28,"occupancy":true}'
6/18/2019, 4:06:32 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":16828}}' of device 'SML001' (0x001788010213d449) of endpoint 2
6/18/2019, 4:06:32 PM - info: MQTT publish: topic 'zigbee2mqtt/0x001788010213d449', payload '{"battery":100,"linkquality":60,"illuminance":16828,"temperature":24.28,"occupancy":true}'
6/18/2019, 4:06:32 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":16828}}' of device 'SML001' (0x001788010213d449) of endpoint 2
6/18/2019, 4:06:35 PM - debug: Received zigbee message of type 'endDeviceAnnce' with data '"0x001788010213d449"' of device 'SML001' (0x001788010213d449) of endpoint 1
6/18/2019, 4:06:37 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":16583}}' of device 'SML001' (0x001788010213d449) of endpoint 2
6/18/2019, 4:06:37 PM - info: MQTT publish: topic 'zigbee2mqtt/0x001788010213d449', payload '{"battery":100,"linkquality":60,"illuminance":16583,"temperature":24.28,"occupancy":true}'`

and when it stopped working I have only :
`6/18/2019, 4:12:26 PM - debug: Received zigbee message of type 'endDeviceAnnce' with data '"0x001788010213d449"' of device 'SML001' (0x001788010213d449) of endpoint 1`

I hope this can help you

closed 2019-08-24 by stale[bot]

rm2kdev, 2019-09-26:
@allofmex not that it helps, but its worth noting i have hue motion sensors connected to the same network that my hue dimmer is dropping on, and those are very stable reporting on motion consistently and never dropping out for many weeks now.

allofmex, 2019-09-26:
> i have hue motion sensors connected to the same network that my hue dimmer is dropping 

Is the sensor close to one of the router? It chooses the parent out of the strongest neighbors. But if it is stable, even if frequently connecting the suspicious router, than this may not be the reason for your drops.

rm2kdev, 2019-09-27:
its about 5 cm away from the hue remote and in the same room.

Both devices are about 1 meter away from the same router, another router at 3 meters, another router at 6 meters and then the hub is in total about 6-7 meters away from this room

allofmex, 2019-12-13:
Found [http://dev.ti.com/tirex/content/simplelink_zigbee_sdk_plugin_1_60_00_14/docs/zigbee_user_guide/html/zigbee/developing_zigbee_applications/z_stack_developers_guide/z-stack-overview.html](url)

Section **9.4 Leave Network**:
> In the base device behavior specification is also stated that if any device receives a valid leave request with rejoin set to FALSE (meaning that this device shall not rejoin the network), then that device is forced to perform a Factory New reset.

So the Hue enddevices are working correct. The problem is with the OSRAM (and others) router. It sends an **incorrect leave request.**

Did anyone found info's if there is new firmware available for OSRAM/Lightify Plugs?
Mine are on V1.04.12, read about a V1.05.09 (V01020509), but cannot find changelog or firmware files.