Koenkk/zigbee2mqtt #1476: My new device don't have model id

GuGu927, 2019-04-28:
# Bug Report
## What happened
I'm using Hass.io + zigbee2mqtt add-on.
I added my device.

log show.
  zigbee2mqtt:warn 4/28/2019, 11:03:46 PM Device with modelID '' is not supported.
  zigbee2mqtt:warn 4/28/2019, 11:03:46 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

I got device information from database.db

{"id":17,"type":"Router","ieeeAddr":"0x00158d0002125512","nwkAddr":3147,"manufId":4404,"manufName":"","powerSource":"Mains (single phase)","modelId":"","epList":[1],"status":"online","joinTime":1556459316,"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,6],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"manufacturerName":"","modelId":"","dateCode":"","powerSource":1}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0}},"genOta":{"dir":{"value":2},"attrs":{}}}}},"_id":"q4eGmz2REaAMDOGs"}

{"id":17,"type":"Router","ieeeAddr":"0x00158d0002125512","nwkAddr":3147,"manufId":4404,"manufName":"","powerSource":"Mains (single phase)","modelId":"","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,6],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"manufacturerName":"","modelId":"","dateCode":"","powerSource":1}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0}},"genOta":{"dir":{"value":2},"attrs":{}}}}},"_id":"q4eGmz2REaAMDOGs"}

As you see, modelId is blank.
Can you help me?

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:1.3.1
CC253X firmware version:CC2531_MAX_STABILITY_20190315

Koenkk, 2019-04-28:
- What device is this?
- As a workaround, you can manually fill in the modelId (doesn't matter what you fill in here, e.g. `mymodel123` is OK)

GuGu927, 2019-04-28:
@Koenkk

Oh! Thank you so much

My device is switch.
Made in South Korea, name "Goqual" and there are 6 type of switches. 1-6gang.

I have 1, 3 and 4 gang switches.
If I having trouble with pairing, I will contact to you again.

Thank yoy again!

GuGu927, 2019-05-03:
@Koenkk 

I added all of my switches, but I can not control it.
When I turn on/off switch,  logs throw error.
Can you help me?

Here is my log.
  zigbee2mqtt:error 5/3/2019, 4:17:03 PM Failed to find device with ieeAddr: 'goqual_bedroom_toilet/1'
  zigbee2mqtt:info 5/3/2019, 4:17:03 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"entity_not_found","message":"goqual_bedroom_toilet/1"}'
  zigbee2mqtt:error 5/3/2019, 4:17:04 PM Failed to find device with ieeAddr: 'goqual_bedroom_toilet/2'
  zigbee2mqtt:info 5/3/2019, 4:17:04 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"entity_not_found","message":"goqual_bedroom_toilet/2"}'
  zigbee2mqtt:error 5/3/2019, 4:17:04 PM Failed to find device with ieeAddr: 'goqual_bedroom_toilet/3'
  zigbee2mqtt:info 5/3/2019, 4:17:04 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"entity_not_found","message":"goqual_bedroom_toilet/3"}'
  zigbee2mqtt:error 5/3/2019, 4:17:04 PM Failed to find device with ieeAddr: 'goqual_bedroom_toilet/4'
  zigbee2mqtt:info 5/3/2019, 4:17:04 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"entity_not_found","message":"goqual_bedroom_toilet/4"}'

Here is my devices.js
zigbeeModel: ['gq.switch_4gang'],
model: 'GQSWITCH_4',
vendor: 'Goqual',
description: 'Goqual Switch',
supports: 'on/off',
fromZigbee: [fz.generic_state_multi_ep, fz.ignore_onoff_change],
toZigbee: [tz.on_off],
ep: (device) => {
    return {'1': 1, '2': 2, '3': 3, '4': 4};
},
configure: (ieeeAddr, shepherd, coordinator, callback) => {
    const ep1 = shepherd.find(ieeeAddr, 1);
    execute(ep1, [(cb) => ep1.bind('genOnOff', coordinator, cb)], () => {
        const ep2 = shepherd.find(ieeeAddr, 2);
        execute(ep2, [(cb) => ep2.bind('genOnOff', coordinator, cb)], () => {
            const ep3 = shepherd.find(ieeeAddr, 3);
            execute(ep3, [(cb) => ep3.bind('genOnOff', coordinator, cb)], () => {
                const ep4 = shepherd.find(ieeeAddr, 4);
                execute(ep4, [(cb) => ep4.bind('genOnOff', coordinator, cb)], callback);
            });
        });
    });
},

Here is my mapping code
'GQSWITCH_4': [switchWithPostfix('1'),switchWithPostfix('2'),switchWithPostfix('3'),switchWithPostfix('4')],

Here is my database.db
{"id":3,"type":"Router","ieeeAddr":"0x00158d0001987f60","nwkAddr":44107,"manufId":4404,"manufName":"","powerSource":"Mains (single phase)","modelId":"gq.switch_4gang","epList":[1,2,3,4],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,6],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"manufacturerName":"","modelId":"gq.switch_4gang","dateCode":"","powerSource":1}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0}},"genOta":{"dir":{"value":2},"attrs":{}}}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[0,3,6],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"manufacturerName":"","modelId":"gq.switch_4gang","dateCode":"","powerSource":1}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0}},"genOta":{"dir":{"value":2},"attrs":{}}}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[0,3,6],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"manufacturerName":"","modelId":"gq.switch_4gang","dateCode":"","powerSource":1}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0}},"genOta":{"dir":{"value":2},"attrs":{}}}},"4":{"profId":260,"epId":4,"devId":256,"inClusterList":[0,3,6],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"manufacturerName":"","modelId":"gq.switch_4gang","dateCode":"","powerSource":1}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0}},"genOta":{"dir":{"value":2},"attrs":{}}}}},"_id":"lrtSWdtqlqvLVKzY"}

{"id":3,"type":"Router","ieeeAddr":"0x00158d0001987f60","nwkAddr":44107,"manufId":4404,"manufName":"","powerSource":"Mains (single phase)","modelId":"gq.switch_4gang","epList":[1,2,3,4],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,6],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"manufacturerName":"","modelId":"gq.switch_4gang","dateCode":"","powerSource":1}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0}},"genOta":{"dir":{"value":2},"attrs":{}}}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[0,3,6],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"manufacturerName":"","modelId":"gq.switch_4gang","dateCode":"","powerSource":1}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0}},"genOta":{"dir":{"value":2},"attrs":{}}}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[0,3,6],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"at [... cut ...] genOta":{"dir":{"value":2},"attrs":{}}}},"4":{"profId":260,"epId":4,"devId":256,"inClusterList":[0,3,6],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"manufacturerName":"","modelId":"gq.switch_4gang","dateCode":"","powerSource":1}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0}},"genOta":{"dir":{"value":2},"attrs":{}}}}},"_id":"lrtSWdtqlqvLVKzY"}

closed 2019-05-04 by GuGu927