Koenkk/zigbee2mqtt #1477: Supporting New Device - Getting Error

flexage, 2019-04-28:
After managing to solve my previous errors regarding pairing new devices, I am trying to add support for them.

They are just another variation of some  Innr GU10 White Bulbs with adjustable White Temperature that are already supported.

My particular model is "RS 228 T".

In the zigbee-shepherd `devices.js` there is already an entry for "RS 128 T" so I have duplicated that entry out and updated the model:

```
{
        zigbeeModel: ['RS 228 T'],
        model: 'RS 228 T',
        vendor: 'Innr',
        description: 'GU10 Spot 350 lm, dimmable, white spectrum',
        extend: generic.light_onoff_brightness_colortemp,
    },
```

I've also updated the `extensions/homeassistant.js` and the bulbs are showing in home assistant:

```
'RS 228 T': [configurations.light_brightness_colortemp],
```

The problem is whenever I try and toggle any of the 8 of these bulbs I added, the HomeAssistant State for them goes to `unknown` and I can see in the Hass.io Zigbee2Mqtt addon logs that we are getting an error:

```
zigbee2mqtt:info 4/28/2019, 4:08:03 PM Zigbee publish to device '0x00158d0002ffa437', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 4/28/2019, 4:08:03 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002ffa437', payload '{"state":"ON"}'
/zigbee2mqtt-1.3.1/lib/extension/deviceReceive.js:109
                if (c.cid === cid) {
                      ^
TypeError: Cannot read property 'cid' of undefined
    at mappedDevice.fromZigbee.filter (/zigbee2mqtt-1.3.1/lib/extension/deviceReceive.js:109:23)
    at Array.filter (<anonymous>)
    at DeviceReceive.onZigbeeMessage (/zigbee2mqtt-1.3.1/lib/extension/deviceReceive.js:105:52)
    at extensions.filter.forEach (/zigbee2mqtt-1.3.1/lib/controller.js:145:31)
    at Array.forEach (<anonymous>)
    at Controller.onZigbeeMessage (/zigbee2mqtt-1.3.1/lib/controller.js:145:14)
    at Zigbee.onMessage (/zigbee2mqtt-1.3.1/lib/zigbee.js:224:18)
    at ZShepherd.emit (events.js:182:13)
    at ZShepherd.<anonymous> (/zigbee2mqtt-1.3.1/node_modules/zigbee-shepherd/lib/shepherd.js:192:14)
    at ZShepherd.emit (events.js:182:13)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.3.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.3.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-04-28T15_08_03_772Z-debug.log
```

Any ideas what I might have done wrong here?

flexage, 2019-04-28:
Looks like my addition to `homeassistant.js` keeps getting wiped out (I'm running Hass.io on HASSOS, so I guess everytime the docker container even restarts it's gone), I can't seem to find a way to make it persist long enough to test and submit a PR. Kinda stuck in limbo here

flexage, 2019-04-28:
Ok, so if i would have read the z2m documentation well enough I would have seen it says to restart the docker container via the host cli, that way it retains the container changes, restarting z2m via the hass.io addons panel replaces with a new container.

The original issue with the `cid` error persists with the bulbs though... I can control them in HA, but not getting state properly and z2m container crashes upon controlling the bulbs.

flexage, 2019-04-28:
I did a `console.log({c})` on `lib/extension/deviceReceive.js:109` and it looks like one of the expected objects is undefined:

```
{ c:
   { cid: 'genLevelCtrl',
     type: 'devChange',
     convert: [Function: convert] } }
{ c:
   { cid: 'lightingColorCtrl',
     type: 'devChange',
     convert: [Function: convert] } }
{ c:
   { cid: 'genOnOff',
     type: 'devChange',
     convert: [Function: convert] } }
{ c:
   { cid: 'genOnOff',
     type: [ 'attReport', 'readRsp' ],
     convert: [Function: convert] } }
{ c:
   { cid: 'genLevelCtrl',
     type: [ 'attReport', 'readRsp' ],
     convert: [Function: convert] } }
{ c:
   { cid: 'lightingColorCtrl',
     type: [ 'attReport', 'readRsp' ],
     convert: [Function: convert] } }
{ c:
   { cid: 'genGroups',
     type: 'devChange',
     convert: [Function: convert] } }
{ c:
   { cid: 'genIdentify',
     type: 'devChange',
     convert: [Function: convert] } }
{ c: undefined }
```

So I guess theres something coming back from the device that is unsupported... have I chosen the wrong base settings to extend in `devices.js`.

My database.db entries for this bulb are:

```
{"id":4,"type":"Router","ieeeAddr":"0x00158d0002f6684c","nwkAddr":26308,"manufId":4454,"manufName":"innr","powerSource":"Mains (single phase)","modelId":"RS 228 T","epList":[1,242],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"9":5,"10":{},"11":"www.innr.com","65533":1,"zclVersion":2,"appVersion":16,"stackVersion":21,"hwVersion":1,"manufacturerName":"innr","modelId":"RS 228 T","dateCode":"20180810-1","powerSource":1,"appProfileVersion":0,"swBuildId":"2.0"}},"genIdentify":{"dir":{"value":1},"attrs":{"65533":1,"identifyTime":115,"identifyCommissionState":0}},"genGroups":{"dir":{"value":1},"attrs":{"65533":1,"nameSupport":0}},"genScenes":{"dir":{"value":1},"attrs":{"65533":1,"count":1,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0,"lastCfgBy":"0x0000000000000000"}},"genOnOff":{"dir":{"value":1},"attrs":{"16387":1,"65533":1,"onOff":1,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"dir":{"value":1},"attrs":{"15":0,"16384":255,"65533":1,"currentLevel":254,"remainingTime":0}},"genOta":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{"15":0,"16397":200,"16400":65535,"65533":1,"remainingTime":0,"currentX":29991,"currentY":26872,"colorTemperature":370,"colorMode":2,"numPrimaries":0,"enhancedColorMode":1,"colorCapabilities":16,"colorTempPhysicalMin":200,"colorTempPhysicalMax":454}},"lightLink":{"dir":{"value":1},"attrs":{"65533":1}}}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{"genGreenPowerProxy":{"dir":{"value":2},"attrs":{}}}}},"_id":"xEEL4YFBGmZ8ySyq"}
```

and

```
{"id":4,"type":"Router","ieeeAddr":"0x00158d0002f6684c","nwkAddr":26308,"manufId":4454,"manufName":"innr","powerSource":"Mains (single phase)","modelId":"RS 228 T","epList":[1,242],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"9":5,"10":{},"11":"www.innr.com","65533":1,"zclVersion":2,"appVersion":16,"stackVersion":21,"hwVersion":1,"manufacturerName":"innr","modelId":"RS 228 T","dateCode":"20180810-1","powerSource":1,"appProfileVersion":0,"swBuildId":"2.0"}},"genIdentify":{"dir":{"value":1},"attrs":{"65533":1,"identifyTime":115,"identifyCommissionState":0}},"genGroups":{"dir":{"value":1},"attrs":{"65533":1,"nameSupport":0}},"genScenes":{"dir":{"value":1},"attrs":{"65533":1,"count":1,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0,"lastCfgBy":"0x0000000000000000"}},"genOnOff":{"dir":{"value":1},"attrs":{"16387":1,"65533":1,"onOff":1,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"dir":{"value":1},"attrs":{"15":0,"16384":255,"65533":1,"currentLevel":254,"remainingTime":0}},"genOta":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{"15":0,"16397":200,"16400":65535,"65533":1,"remainingTime":0,"currentX":29991,"currentY":26872,"colorTemperature":370,"colorMode":2,"numPrimaries":0,"enhancedColorMode":1,"colorCapabilities":16,"colorTempPhysicalMin":200,"colorTempPhysicalMax":454}},"lightLink":{"dir":{"value":1},"attrs":{"65533":1}}}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{"genGreenPowerProxy":{"dir":{"value":2},"attrs":{}}}}},"_id":"xEEL4YFBGmZ8ySyq"}

```

Koenkk, 2019-04-28:
Hi!

I've added this device, should be support in the hassio **edge** addon in 2 hours.

flexage, 2019-04-28:
Amazing, thank you very much! Just so I can learn from this, what did I do wrong?

Koenkk, 2019-04-28:
Can you share your complete `devices.js` file? (on https://pastebin.com/)

flexage, 2019-04-28:
Sure! I only added the lines I mentioned in my first post: https://pastebin.com/aiAbn3C3

Koenkk, 2019-04-29:
That looks ok, now sure where it comes from. I've seen this error multiple times, most of the times by using an outdated zigbee2mqtt version.

closed 2019-04-29 by Koenkk